espressif/esp-zigbee-sdk #80: Understanding about Electrical Measurement Cluster

Tropaion, 2023-08-24:
I created my own [zigbee device](https://github.com/Tropaion/ZigBee_SmartMeter_Reader/blob/main/software/zigbee/components/zigbee_electricity_meter/src/zb_electricity_meter.c#L198) which reads phase a, b, and c data from an smartmeter.
In detail, these attributes are sent:
```
Attribute Set 0x03: AC (Non-phase Specific) Measurements (S. 303)
electrical power P+ (W): 0x0304 TotalActivePower

Attribute Set 0x05: AC (Single Phase or Phase A) Measurements (S. 306)
voltage L1: 0x0505 RMSVoltage
current L1: 0x0508 RMSCurrent

Attribute Set 0x09: AC Phase B Measurements (S. 313)
voltage L2: 0x0905 RMSVoltagePhB
current L2: 0x0908 RMSCurrentPhB

Attribute Set 0x0A: AC Phase C Measurements
voltage L3: 0x0A05 RMSVoltagePhB
current L3: 0x0A08 RMSCurrentPhB
```
But ZHA only creates the three entities power factor, phase A current and phase A voltage even though I don't even report power factor. Any idea what I have done wrong, maybe with configuring the cluster?
[image]

And, is it still needed to use static variables to create an attribute or are they now stored internally?
Thanks for your help.

aquaticus, 2023-09-11:
This may be specific for ZHA implementation. Did you try to use `zha-toolkit` or `zigbee2mqyy`?

Tropaion, 2023-09-12:
No, I didn't try this, but looks like I will have to.

xieqinan, 2023-10-31:
@Tropaion ,

Have you tried other ZHA implementations? We can discuss any updates here.

Tropaion, 2023-10-31:
@xieqinan Thanks for asking. I didn't have time until now, but I will probably try in the next few days.

chshu, 2024-01-08:
@Tropaion Seems not the Zigbee SDK issue, please reopen if you have any further quesions.

closed 2024-01-08 by chshu