espressif/esp-zigbee-sdk #81: esp_zb_set_extended_pan_id has no effect.

petergjoel, 2023-08-24:
The function `esp_zb_set_extended_pan_id` has no effect, the following snippet of code demonstrates it:

```c
err = nvs_get_blob(nvs_h, NVS_STORAGE_EXT_PAN_ID_KEY, epan, &dummy);
if(ESP_OK == err)
{
    esp_zb_set_extended_pan_id(epan);
} else {
    printf("Failed to load NVS_STORAGE_EXT_PAN_ID_KEY (status: %s) \n", esp_err_to_name(err));
    paired = false;
}
printf("Loaded config (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx)\n",
    epan[7], epan[6], epan[5], epan[4],
    epan[3], epan[2], epan[1], epan[0],
    pan_id);
esp_zb_get_extended_pan_id(epan);
printf("Actual config (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx)\n",
    epan[7], epan[6], epan[5], epan[4],
    epan[3], epan[2], epan[1], epan[0],
    esp_zb_get_pan_id());
```

Output
```
Loaded config (Extended PAN ID: 4x:97:210:162:60:163:74:19, PAN ID: 0xdd73)
Actual config (Extended PAN ID: 00:00:00:00:00:00:00:00, PAN ID: 0xdd73)
Loaded Panid
```

Despite the documentation (and conventional C-semantics), `esp_zb_get_extended_pan_id` works fine with the direct output parameter, even though it *ought to be* a pointer to a struct instead, otherwise C should pass the entire struct by value.

Setting panid, network-key and channel mask seems to work fine.

petergjoel, 2023-08-24:
Maybe some context here: I'm running this on an end-device. We are experiencing that the devices crash and permanently drop out of the network.
We are thus trying to re-establish the network connection.

xieqinan, 2023-08-25:
Hi,
For Zigbee end devices, the extended PAN ID is distributed by the coordinator, rendering the `esp_zb_set_extended_pan_id()` function ineffective. However, this function provides useful for an end device that intends to join a specific network with the extended PAN ID. In simpler terms, if an end device employs the `esp_zb_set_extended_pan_id()` function to set the extended PAN ID, it will attempt to join the network associated with this setting extended PAN ID.

petergjoel, 2023-08-28:
> if an end device employs the esp_zb_set_extended_pan_id() function to set the extended PAN ID, it will attempt to join the network associated with this setting extended PAN ID.

@xieqinan, yes, this is what I expect. However, I am *not* observing that behavior. Or is this assuming that the coordinator is in the 'join'-mode at the same time?

Essentially my edge device already connected to the network once, and hence I have both panid, extended panid and network key. For some (unknown to me yet) reason, I occasionally get a panic, the device reboots and has all zigbee information lost.
I am thus trying to hot-fix the situation by manually rejoining.

(Side-note: When manually rebooting the device when it is in the network it rejoins the network fine, it is only after a specific panic type I need to re-establish the connection).

xieqinan, 2023-08-30:
Hello,
> I am _not_ observing that behavior. Or is this assuming that the coordinator is in the 'join'-mode at the same time?

I'm a bit confused about the meaning of the 'join' mode. If the network has already been established by the coordinator, and the end device's extended PANID has been configured (keeping it on the same channel as the coordinator) using the `esp_zb_set_extended_pan_id()` function, the end device will join the network of the specified coordinator.

> For some (unknown to me yet) reason, I occasionally get a panic,

As I'm unaware of any additional logics added by you that might have been incorporated into the example, potentially leading to the panic, kindly furnish me with a detailed, step-by-step procedure to reproduce the scenario. This would significantly accelerate the process of diagnosing and addressing the problem.

chshu, 2023-10-12:
@petergjoel Do you have any update on this issue?

If you still see the panic with the latest SDK, could you share us some detailed logs with [trace logging ](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#enable-trace-logging)enabled?

chshu, 2024-01-08:
@petergjoel closing the issue, please reopen if any further updates on this issue.

closed 2024-01-08 by chshu

rsibille, 2024-06-28:
Hello
I'm facing the same issue : setting the extended PANID on the ED (end device) seems to have no effect.
The situation is : 
- I'm using two ESP32C6, respectively as coordinator and ED, both running ESP-IDF v5.2.2
- the coordinator is started, the signal `ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS` is sent, with an open time of 180s. So I understand it is ready to accept joining from devices
- when the ED starts, I set the ext. PANID to the value read on the coordinator
- but the ED does not join : `esp_zb_get_extended_pan_id()` still returns the ED mac address and I'm getting a signal '_NWK Permit Join_', as if the ED were waiting for another to join ...

So my questions are :
1. is this call correct ?
```
    esp_zb_ieee_addr_t ext_pan_id = {0xF4, 0x82, 0x44, 0xFE, 0xFF, 0xCA, 0x4C, 0x40}; // 40:4c:ca:ff:fe:44:82:f4
    esp_zb_set_extended_pan_id(ext_pan_id);
```
2. should I also call `set_pan_id()`
3. when should I do these calls ? They are currently inside `case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:`

Thanks in advance for any clue.