espressif/esp-zigbee-sdk #82: Can't connect after setting MAC address with manufacturing tool

nomis, 2023-08-26:
Using 0.9.1:
```
tools/mfg_tool/esp_zb_mfg_tool.py -c 0x7FFFFFFF -m 404CCAFFFE430BD0
cd examples/esp_zigbee_HA_sample/HA_on_off_light
idf.py set-target esp32c6
idf.py build
idf.py erase-flash
esptool.py -p $ESPPORT write_flash 0xb7000 ../../../tools/mfg_tool/bin/404CCAFFFE423078.bin
idf.py flash
```

I only get as far as "Starting Interview" trying to add the device. The right MAC address is displayed by the co-ordinator but the end device won't connect:
```
I (442) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_OK
I (442) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (442) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (9222) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (9222) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (27172) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (27172) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (36962) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (36962) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (79322) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (79322) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```

It works if I erase the `zb_fct` partition:
```
I (442) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (442) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (442) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (42242) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: *, PAN ID: *, Channel:26)
I (60572) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(0x64), cluster(0x6), attribute(0x0), data size(1)
I (60572) ESP_ZB_ON_OFF_LIGHT: Light sets to On
I (65722) ESP_ZB_ON_OFF_LIGHT: Received message: endpoint(0x64), cluster(0x6), attribute(0x0), data size(1)
I (65722) ESP_ZB_ON_OFF_LIGHT: Light sets to Off
```

xieqinan, 2023-08-28:
Hello,
The `esp_zb_mfg_tool.py` provides a method for generating and programming the `install code`, `MAC address`, and manufacturer code for Zigbee devices. When the device's `zb_fct` (functionality) has been configured, it will activate the `install code` mode to facilitate network joining.

Based on your explanation, the light example only programs the `channel` and `MAC address`, omitting the `install code`, which would still trigger the `install code` mode. In such a scenario, the coordinator must also activate the `install code` option. Furthermore, the coordinator needs to utilize the `esp_zb_secur_ic_str_add()` function to incorporate the joining device's `install code` after the network has been established. Failing to do so might result in the coordinator excluding the device from its network.

To enhance user-friendliness, modifications will be made to the `esp_zb_mfg_tool.py` to offer a more user-oriented API.

The correct sequence of steps is as follows:
- on light side
```
cd ~/esp/esp-zigbee-sdk/tools/mfg_tool
python esp_zb_mfg_tool.py -i 83FED3407A939723A5C639B26916D505C3B5 -m CAFEBEEF50C0FFED -c 0x07FFF800 -mn Espressif -mc 0x131B
cd ~/esp/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_light
idf.py -p [PORT] erase-flash
esptool.py -p [PORT] write_flash 0xb7000 ~/esp/esp-zigbee-sdk/tools/mfg_tool/bin/CAFEBEEF50C0FFED.bin
idf.py -p [PORT] flash monitor
```
- on switch side
```
// in esp_on_off_switch.h
#define INSTALLCODE_POLICY_ENABLE true

// add the following code after the network formation to esp_on_off_switch.c 
uint8_t address[] = {0xed, 0xff, 0xc0, 0x50, 0xef, 0xbe, 0xfe, 0xca};
char ic_code[] = "83FED3407A939723A5C639B26916D505C3B5";
esp_zb_secur_ic_str_add(address, ic_code);
```

nomis, 2023-08-28:
> To enhance user-friendliness, modifications will be made to the `esp_zb_mfg_tool.py` to offer a more user-oriented API.

The install code needs to be optional on the end device.

That's not the only parameter that has an undesirable default. The channel mask defaults to only allowing one channel when it should default to all channels.

xieqinan, 2023-08-28:
Thank you for your suggestion.

mallaprashant, 2023-09-20:
> The correct sequence of steps is as follows:
>
> * on light side
>
> ```
> cd ~/esp/esp-zigbee-sdk/tools/mfg_tool
> python esp_zb_mfg_tool.py -i 83FED3407A939723A5C639B26916D505C3B5 -m CAFEBEEF50C0FFED -c [...]
@xieqinan, I have diligently followed the specified steps using "HA_on_off_light" and "HA_on_off_switch" examples, successfully establishing connectivity between the end device and the coordinator through the implementation of the install code policy. However, when I manually restart the coordinator, it effortlessly reestablishes its connection with the end device. Unfortunately, the same cannot be said when I attempt to manually restart the end device. It fails to reconnect to the coordinator, and I encounter the following error message in the output log: "ESP_HA_ON_OFF_LIGHT: failed to initialize zigbee stack."

Additionally, I've encountered a similar issue when experimenting with the "esp_zigbee_customized_devices" examples, which results in the following log output: "ESP_HA_ON_OFF_SWITCH: stack BDB Device Reboot failure with ESP_FAIL status, steering."

I'm eager to gain insights into the root causes of these occurrences and would appreciate your guidance on this matter.

nomis, 2023-09-20:
@mallaprashant please create your own issue instead of hijacking this one

closed 2024-01-12 by espressif-bot