espressif/esp-zigbee-sdk #83: Group membership incorrectly reported for all endpoints

nomis, 2023-08-28:
I have an end device with several endpoints that each have their own on/off and group clusters.

https://github.com/nomis/candle-dribbler (with `NUTT_MAX_LIGHTS=4` aka "Component config" -> "Candle Dribbler" -> "Number of lights supported")

The cluster list for the endpoints is configured in `src/light.cpp`, `OnOffEndpoint::configure_light_cluster_list(esp_zb_cluster_list_t&)` and `OnOffEndpoint::configure_switch_cluster_list(esp_zb_cluster_list_t&)`  which is called from `src/zigbee.cpp`, `ZigbeeDevice::add(ZigbeeEndpoint&)`.

When I use the co-ordinator to add specific endpoints to a group they work as expected, with the on/off attribute being turned on/off using the group.

The problem arises when the co-ordinator queries the device to establish the group membership of each of its endpoints. I've added endpoint 12 to group 0x0005 and endpoints 13 and 14 to group 0x0002, but when the co-ordinator refreshes group membership information it displays all 16 on/off endpoints as members of both groups.

The co-ordinator log when I remove all the endpoints that appear to be group members but shouldn't be, showing the device returning an error because it agrees those endpoints are not members of the groups:
```
2023-08-27 15:03:51.682 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.703 DEBUG (MainThread) [zigpy.zcl] [0x0E42:21:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.730 DEBUG (MainThread) [zigpy.zcl] [0x0E42:31:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.804 DEBUG (MainThread) [zigpy.zcl] [0x0E42:71:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.828 DEBUG (MainThread) [zigpy.zcl] [0x0E42:22:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.851 DEBUG (MainThread) [zigpy.zcl] [0x0E42:32:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.893 DEBUG (MainThread) [zigpy.zcl] [0x0E42:72:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.924 DEBUG (MainThread) [zigpy.zcl] [0x0E42:13:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:51.979 DEBUG (MainThread) [zigpy.zcl] [0x0E42:23:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.040 DEBUG (MainThread) [zigpy.zcl] [0x0E42:33:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.063 DEBUG (MainThread) [zigpy.zcl] [0x0E42:73:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.085 DEBUG (MainThread) [zigpy.zcl] [0x0E42:14:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.117 DEBUG (MainThread) [zigpy.zcl] [0x0E42:24:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.143 DEBUG (MainThread) [zigpy.zcl] [0x0E42:34:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:03:52.171 DEBUG (MainThread) [zigpy.zcl] [0x0E42:74:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0005)
2023-08-27 15:04:00.712 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.733 DEBUG (MainThread) [zigpy.zcl] [0x0E42:21:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.795 DEBUG (MainThread) [zigpy.zcl] [0x0E42:31:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.821 DEBUG (MainThread) [zigpy.zcl] [0x0E42:71:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.867 DEBUG (MainThread) [zigpy.zcl] [0x0E42:12:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.910 DEBUG (MainThread) [zigpy.zcl] [0x0E42:22:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.934 DEBUG (MainThread) [zigpy.zcl] [0x0E42:32:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:00.988 DEBUG (MainThread) [zigpy.zcl] [0x0E42:72:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.038 DEBUG (MainThread) [zigpy.zcl] [0x0E42:23:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.064 DEBUG (MainThread) [zigpy.zcl] [0x0E42:33:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.110 DEBUG (MainThread) [zigpy.zcl] [0x0E42:73:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.187 DEBUG (MainThread) [zigpy.zcl] [0x0E42:24:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.198 DEBUG (MainThread) [zigpy.zcl] [0x0E42:34:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
2023-08-27 15:04:01.236 DEBUG (MainThread) [zigpy.zcl] [0x0E42:74:0x0004] Decoded ZCL frame: Groups:remove_response(status=<Status.NOT_FOUND: 139>, group_id=0x0002)
```

The log when the co-ordinator next refreshes the group membership information, where all the endpoints claim to be members of both groups:
```
2023-08-28 07:53:37.520 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.282 DEBUG (MainThread) [zigpy.zcl] [0x0E42:21:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.412 DEBUG (MainThread) [zigpy.zcl] [0x0E42:31:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.491 DEBUG (MainThread) [zigpy.zcl] [0x0E42:71:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.590 DEBUG (MainThread) [zigpy.zcl] [0x0E42:12:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.660 DEBUG (MainThread) [zigpy.zcl] [0x0E42:22:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.750 DEBUG (MainThread) [zigpy.zcl] [0x0E42:32:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.831 DEBUG (MainThread) [zigpy.zcl] [0x0E42:72:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.911 DEBUG (MainThread) [zigpy.zcl] [0x0E42:13:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:38.979 DEBUG (MainThread) [zigpy.zcl] [0x0E42:23:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:39.069 DEBUG (MainThread) [zigpy.zcl] [0x0E42:33:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:39.171 DEBUG (MainThread) [zigpy.zcl] [0x0E42:73:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:39.259 DEBUG (MainThread) [zigpy.zcl] [0x0E42:14:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:39.358 DEBUG (MainThread) [zigpy.zcl] [0x0E42:24:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
[... cut ...]
2023-08-28 07:53:37.520 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:37.520 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] Received command 0x02 (TSN 161): get_membership_response(capacity=6, groups=[0x0005, 0x0002])
2023-08-28 07:53:37.520 DEBUG (MainThread) [zigpy.zcl] [0x0E42:11:0x0004] No explicit handler for cluster command 0x02: get_membership_response(capacity=6, groups=[0x0005, 0x0002])
```

nomis, 2023-10-14:
This bug is still present in 1.0.0

xieqinan, 2023-10-18:
Hello @nomis ,

I attempted to reproduce the issue based on the description you provided; however, I was unsuccessful in doing so.

Here are the steps I followed:

1. I created 16 endpoints with group cluster and on/off cluster on a device node in an end device example.
2. I created another coordinator example to assist the endpoints in different groups.
3. Initially, I used `esp_zb_zcl_groups_add_group_cmd_req()` to add endpoint x to group x. For instance, I added endpoint 1 to group 1, endpoint 2 to group 2, and so on, up to endpoint 16 in group 16.
4. Subsequently, I added endpoint 12 to group 0x0005, and endpoints 13 and 14 to group 0x0002.
5. Finally, I used `esp_zb_zcl_groups_get_group_membership_cmd_req()` to retrieve the group memberships.
6. The results matched the expected outcome.

Could you please confirm if there are any errors in your provided description?

nomis, 2023-10-18:
>     5. Finally, I used `esp_zb_zcl_groups_get_group_membership_cmd_req()` to retrieve the group memberships.

That function expects you to provide a list of groups to look for. If you're only requesting information on the groups you expect, you won't get the additional groups that are not really present.

On my device:
* Group 0x0005 has endpoint 12
* Group 0x0002 has endpoints 13 and 14
* Group 0x0006 has endpoints 12, 13 and 14

If I make a request using Home Assistant and ZHA Toolkit for an endpoint that is not a member of any groups:
```
service: zha_toolkit.get_groups
data:
  ieee: "40:4c:ca:ff:fe:43:0b:d0"
  endpoint: 11
```

It returns all of the groups that any of the endpoints are members of:
```
2023-10-18 12:25:02.509 DEBUG (MainThread) [zigpy.zcl] [0x3828:11:0x0004] Sending request: get_membership(groups=[])
2023-10-18 12:25:02.578 DEBUG (MainThread) [zigpy.zcl] [0x3828:11:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=5, groups=[0x0006, 0x0005, 0x0002])
```

I only need one endpoint to be a member of one group to reproduce this. All of the endpoints that implement the groups cluster will claim to be members of it when queried.

xieqinan, 2023-10-19:
Hello @nomis ,

I understand your point.

There might be some ambiguity in understanding the ZCL group specification, as follows:

[image]

The Zboss stack believes that if the group count field of the command frame has a value of 0, it should respond with all group identifiers that trigger this issue. However, I agree with your suggestion. I will attempt to provide feedback on the issue and will inform you of the final result as soon as possible.

nomis, 2023-10-19:
That's inconsistent with its other behaviour because it maintains per-endpoint group membership. Only the expected endpoints are affected by commands on the groups, and the "remove group" command is aware of the per-endpoint membership.

The specification for this is poor because it keeps mixing "entity" and device" but the "add group" command is clear: "The Add Group command allows the sending device to add group membership in a particular group for **one or more endpoints** on the receiving device."

The overview makes a similar statement: "any endpoint on any device MAY be assigned to one or more groups".

The server section defines what "entity" means in this context: "the application **entity** resident at that **endpoint** will process the command".

xieqinan, 2023-10-19:
However, I found another phrase describing the group list as follows, which supports the usage of the ZBOSS stack
[image]

nomis, 2023-10-19:
> However, I found another phrase describing the group list as follows, which supports the usage of the ZBOSS stack

Then the specification is just missing detail in more places. It needs to filter the group table by the endpoint, just like it uses the endpoint when it processes the "add group" and "remove group" commands.

"The get group membership command allows the sending device to inquire about the group membership of the **receiving device and endpoint** in a number of ways."

xieqinan, 2023-10-20:
@nomis ,

There is some ambiguity in this question. I have already submitted an issue to the ZBOSS stack regarding this matter and am waiting for their response. We are currently not making any modifications. As a workaround, you can use `esp_zb_zcl_groups_get_group_membership_cmd_req()` with an empty group list to retrieve all groups from the group_table where the endpoint is resident. Then, call it again with the complete group list to obtain the groups associated with the endpoint.

nomis, 2024-01-20:
>  As a workaround, you can use `esp_zb_zcl_groups_get_group_membership_cmd_req()` with an empty group list to retrieve all groups from the group_table where the endpoint is resident. Then, call it again with the complete group list to obtain the groups associated with the endpoint.

That's not consistent with your own selective interpretation of the specification:
> "The Group list field SHALL contain the identifiers either of all the groups in the group table (in the case where the group list field of the received get group membership command was empty) or all the groups from the group list field of the received get group membership command which are in the group table."
>
> * all the groups in the **group table** (in the case where the group list field of the received get group membership command was empty)
> * all the groups from the group list field of the **received get group membership command** which are in the **group table**

You can't have it both ways: either the group table is all groups on all endpoints or only groups on the current endpoint. Currently as implemented it varies based on whether the received command has a list of groups.

When is this going to be fixed?

closed 2024-03-08 by espressif-bot

xieqinan, 2024-03-08:
Hello @nomis ,

This issue has been addressed in version 1.2.0. Could you please update and test it?

nomis, 2024-03-09:
This is working as expected now:
```
11:24:42.784 [0x78C0:11:0x0004] Sending request: get_membership(groups=[])
11:24:42.888 [0x78C0:11:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:42.995 [0x78C0:21:0x0004] Sending request: get_membership(groups=[])
11:24:43.094 [0x78C0:21:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:43.199 [0x78C0:31:0x0004] Sending request: get_membership(groups=[])
11:24:43.298 [0x78C0:31:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:43.402 [0x78C0:12:0x0004] Sending request: get_membership(groups=[])
11:24:43.500 [0x78C0:12:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[0x0005, 0x0006])
11:24:43.604 [0x78C0:22:0x0004] Sending request: get_membership(groups=[])
11:24:43.702 [0x78C0:22:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:43.805 [0x78C0:32:0x0004] Sending request: get_membership(groups=[])
11:24:43.904 [0x78C0:32:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:44.009 [0x78C0:13:0x0004] Sending request: get_membership(groups=[])
11:24:44.111 [0x78C0:13:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[0x0002, 0x0006])
11:24:44.216 [0x78C0:23:0x0004] Sending request: get_membership(groups=[])
11:24:44.317 [0x78C0:23:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:44.421 [0x78C0:33:0x0004] Sending request: get_membership(groups=[])
11:24:44.523 [0x78C0:33:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:44.628 [0x78C0:14:0x0004] Sending request: get_membership(groups=[])
11:24:44.726 [0x78C0:14:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[0x0002, 0x0006])
11:24:44.831 [0x78C0:24:0x0004] Sending request: get_membership(groups=[])
11:24:44.930 [0x78C0:24:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
11:24:45.035 [0x78C0:34:0x0004] Sending request: get_membership(groups=[])
11:24:45.137 [0x78C0:34:0x0004] Decoded ZCL frame: Groups:get_membership_response(capacity=29, groups=[])
```