espressif/esp-zigbee-sdk #84: customized_client Sample program An warning is reported that stack initialization failed(ESP32C6)

321759590, 2023-08-29:
When I run the program,the warning is:
W(5891) ESP_HA_ON_SWITCH:Faild to initialize Zigbee Stack(status:ESP_FAIL)
And then my coordinator device doesn't hear from this device

xieqinan, 2023-08-29:
Hello,

The log message "`W(5891) ESP_HA_ON_SWITCH: Failed to initialize Zigbee Stack (status: ESP_FAIL)`" could be prompted by incorrect network information. Here are two potential scenarios:

1. If the end device lacks network information in its memory, there's a possibility that the device possesses an incorrect pre-configured network setting. Consequently, it struggles to locate a suitable network.

2. In cases where the end device had previously connected to a network, the issue could arise from the absence of the old network. This absence might lead to the end device's rejoining process encountering failures.

You can utilize the command `idf.py -p [port] erase-flash && idf.py -p [port] flash monitor` to completely erase all cached memory on the end device, prompting it to initiate a new network search. Alternatively, you can invoke the function `esp_zb_nvram_erase_at_start(true)` before ` esp_zb_start()`  on the end device to eliminate the stored NVRAM information.

321759590, 2023-08-29:
> Hello,
>
> The log message "`W(5891) ESP_HA_ON_SWITCH: Failed to initialize Zigbee Stack (status: ESP_FAIL)`" could be prompted by incorrect network information. Here are two potential scenarios:
>
> 1. [...]

I did what you said, but it was the same thing. He had a warning ahead of him:W (239) clk: esp_perip_clk_init() has not been implemented yet

xieqinan, 2023-08-29:
I also attempted to use the latest version of the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) to run the example, but I was unable to reproduce the issue you mentioned. Could you please try erasing the flash memory on both the coordinator and the end device, and then attempt the process once more?

Additionally, do you have any other modifications in mind for this example?

321759590, 2023-08-29:
> I also attempted to use the latest version of the [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) to run the example, but I was unable to reproduce the issue you mentioned. Could you [...]

Yes, I'm sure I cleared the flash successfully, but I still had the two problems mentioned above. The coordinator worked, but the routine failed to initialize the stack, even though I switched to esp32h2

xieqinan, 2023-08-29:
OK, Please  try to add the following code:
```
ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
```

321759590, 2023-08-29:
> 

I do as you say, but he is just one more warning, now has three warning:
W (239) clk: esp_perip_clk_init() has not been implemented yet 
W (365)ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_OK)
 W (5905) ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_FAIL)

321759590, 2023-08-29:
> I do as you say, but he is just one more warning, now has three warning:
> W (239) clk: esp_perip_clk_init() has not been implemented yet
> W (365)ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_OK)
> W (5905) ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_FAIL)

I do as you say, but he is just one more warning, now has three warning:
W (239) clk: esp_perip_clk_init() has not been implemented yet
W (365)ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_OK)
W (5905) ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_FAIL)

xieqinan, 2023-08-29:
> _/_ commissioning failed _/ ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));_

```
/ commissioning failed /
ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
```
Please try to use the following code to replace it
```
ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING); // try to steering, when initialization failure
```

xieqinan, 2023-08-30:
Hello,

If the issue has been resolved, please close it, thanks.

huangdiajd, 2023-08-30:
> Hello,
> 
> If the issue has been resolved, please close it, thanks.
Hello, I have this problem too. 
W (5917) ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_FAIL) and then I look at the method you gave me, 
first, I will esp_zb_nvram_erase_at_start(true); Put in ESP_ERROR_CHECK(esp_zb_start(true)); 
second, I will also /* commissioning failed */ // ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status)); ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status)); esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING); 
// try to steering, when initialization failure. 
In the end, the problem was not solved. 
The terminal will print the following message: W (5917) ESP_HA_ON_OFF_SWITCH: Failed to initialize Zigbee stack (status: ESP_FAIL) 
I (11237) ESP_HA_ON_OFF_SWITCH: Network steering was not successful (status: ESP_FAIL) 
I (17557) ESP_HA_ON_OFF_SWITCH: Network steering was not successful (status: ESP_FAIL) 
I (23887) ESP_HA_ON_OFF_SWITCH: Network steering was not successful (status: ESP_FAIL) 
I (30217) ESP_HA_ON_OFF_SWITCH: Network steering was not successful (status: ESP_FAIL) 
I (36547) ESP_HA_ON_OFF_SWITCH: Network steering was not successful (status: ESP_FAIL) 
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked 
the correct console for serial communication. 
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked 
the correct console for serial communication.

xieqinan, 2023-08-30:
Hi @huangdiajd ,
This phenomenon might have been caused by the expiration of the coordinator commissioning time, which is set to a default of 180 seconds. Would you be able to attempt a reset on the coordinator side for reset the commissioning time?

xieqinan, 2023-09-22:
Hello @huangdiajd ,

Your issue whether has been resolved, please feel free to discussion any problem for esp-zigbee-sdk.

closed 2023-09-23 by 321759590