espressif/esp-zigbee-sdk #85: Receive and Handle signal from standard zigbee switch

muc-martin, 2023-08-30:
I am using a standard zigbee switch (e.g.  [Paulmann Smart Switch](https://de.paulmann.com/smart-home/zigbee/smart-switch/)) and want to receive and handle button presses on a esp32h2 running the script “[customized_server](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices/customized_server)”. The script is working fine and the switch joins the network. Now, however, I would like to be able to receive and handle button presses of the zigbee switch on the esp32h2 (for the first test a print to the terminal would be fine). Could you give me hints how to achieve this?

xieqinan, 2023-08-31:
Hello,
In a standard scenario, when the customized_server receives an ON/OFF command from the connected Zigbee switch, users are able to handle the received command in a manner similar to this example: [customized_light](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c#L116). However, I'm uncertain about the specific command that the Zigbee switch will transmit. If you could provide a Wireshark packet containing the command dispatched by the Zigbee switch, it's likely that resolving the issue could be accelerated.

muc-martin, 2023-09-11:
@xieqinan I attached the Wireshark log that contains the join of the of the switch as well as the button press.

[wireshark_zigbee.txt](https://github.com/espressif/esp-zigbee-sdk/files/12575323/wireshark_zigbee.txt)

xieqinan, 2023-09-11:
Hi @muc-martin ,

The Wireshark log is useful; however, I cannot find any commands for the on/off cluster in the log. Could you confirm whether the on/off control command is sent from the [Paulmann Smart Switch](https://de.paulmann.com/smart-home/zigbee/smart-switch/)?

muc-martin, 2023-09-11:
Hi @xieqinan,

I rechecked the wireshark output and I see the following output the moment I press a button on the smart switch. The output is only sent after it joined the network.

```
No.     Time           Source                Destination           Protocol Length Info
    245 336.556052                                                 IEEE 802.15.4 5      Ack

Frame 245: 5 bytes on wire (40 bits), 5 bytes captured (40 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Sep 11, 2023 15:28:35.556052000 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1694438915.556052000 seconds
    [Time delta from previous captured frame: 0.001959000 seconds]
    [Time delta from previous displayed frame: 0.001959000 seconds]
    [Time since reference or first frame: 336.556052000 seconds]
    Frame Number: 245
    Frame Length: 5 bytes (40 bits)
    Capture Length: 5 bytes (40 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan]
IEEE 802.15.4 Ack, Sequence Number: 58
    Frame Control Field: 0x0002, Frame Type: Ack, Destination Addressing Mode: None, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: None
        .... .... .... .010 = Frame Type: Ack (0x2)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .0.. .... = PAN ID Compression: False
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 00.. .... .... = Destination Addressing Mode: None (0x0)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        00.. .... .... .... = Source Addressing Mode: None (0x0)
    Sequence Number: 58
    FCS: 0x2b61 (Correct)

0000  02 00 3a 61 2b                                    ..:a+

No.     Time           Source                Destination           Protocol Length Info
    246 336.596094     0xba21                Broadcast             ZigBee   49     Data, Dst: Broadcast, Src: 0xba21

Frame 246: 49 bytes on wire (392 bits), 49 bytes captured (392 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Sep 11, 2023 15:28:35.596094000 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1694438915.596094000 seconds
    [Time delta from previous captured frame: 0.040042000 seconds]
    [Time delta from previous displayed frame: 0.040042000 seconds]
    [Time since reference or first frame: 336.596094000 seconds]
    Frame Number: 246
    Frame Length: 49 bytes (392 bits)
    Capture Length: 49 bytes (392 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:data]
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x0000
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
[... 68 lines cut ...]
IEEE 802.15.4 Command, Dst: 0x0000, Src: 0xba21
    Frame Control Field: 0x8863, Frame Type: Command, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .011 = Frame Type: Command (0x3)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..1. .... = Acknowledge Request: True
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        10.. .... .... .... = Source Addressing Mode: Short/16-bit (0x2)
    Sequence Number: 59
    Destination PAN: 0xb588
    Destination: 0x0000
    Source: 0xba21
    [Extended Source: SiliconL_ff:fe:14:10:08 (b4:e3:f9:ff:fe:14:10:08)]
    [Origin: 13]
    Command Identifier: Data Request (0x04)
    FCS: 0x2e87 (Correct)

0000  63 88 3b 88 b5 00 00 21 ba 04 87 2e               c.;....!....

```

xieqinan, 2023-09-22:
Hello @muc-martin ,

Could you please attempt to use the following code at this [line](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c#L164) on the [customized_server](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices/customized_server)? and then try it again.

```
 esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID);
```

muc-martin, 2023-10-11:
Hi @xieqinan

After trying around for a while, I got it to work using the [touchlink_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink/touchlink_light). Now however I am facing another problem. I want to output the four different button-presses so I can configure each button-presses individually. Later I want to change the state of the pins (high / low output) of the esp32h2 for interrupts on another MC. Do you have a clue how to implement it?

ArturProchniak, 2023-10-11:
Hello @xieqinan,

I have tried getting the [touchlink_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink/touchlink_light) to run on my esp32h2. Unfortunately it often does not work and I get this log, which is quite different from the “Example Output” in the [README.md](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_touchlink/touchlink_light/README.md). Do know how to fix it?

```
I (23) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (24) boot: compile time Oct 11 2023 17:32:19
I (25) boot: chip revision: v0.1
I (26) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000a2800
I (79) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=42078020 size=11250h ( 70224) map
I (124) esp_image: segment 1: paddr=00021278 vaddr=40800000 size=06da0h ( 28064) load
I (133) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=77f70h (491376) map
I (255) esp_image: segment 3: paddr=0009ff98 vaddr=40806da0 size=05c7ch ( 23676) load
I (262) esp_image: segment 4: paddr=000a5c1c vaddr=4080ca20 size=013c8h (  5064) load
I (266) boot: Loaded app from partition at offset 0x10000
I (267) boot: Disabling RNG early entropy source...
I (283) cpu_start: Unicore app
I (284) cpu_start: Pro cpu up.
W (292) clk: esp_perip_clk_init() has not been implemented yet
I (299) cpu_start: Pro cpu start user code
I (299) cpu_start: cpu freq: 96000000 Hz
I (300) cpu_start: Application information:
I (302) cpu_start: Project name:     esp_touchlink_light
I (308) cpu_start: App version:      1
I (312) cpu_start: Compile time:     Oct 11 2023 17:32:02
I (318) cpu_start: ELF file SHA256:  d7d89b6108d18ce9...
I (324) cpu_start: ESP-IDF:          v5.1.1
I (329) cpu_start: Min chip rev:     v0.0
I (334) cpu_start: Max chip rev:     v0.99
I (339) cpu_start: Chip rev:         v0.1
I (344) heap_init: Initializing. RAM available for dynamic allocation:
I (351) heap_init: At 40814890 len 00038AF0 (226 KiB): D/IRAM
I (357) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (365) spi_flash: detected chip: generic
I (369) spi_flash: flash io: dio
W (372) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (386) sleep: Configure to isolate all GPIO pins in sleep state
I (393) sleep: Enable automatic switching of GPIO sleep configuration
I (400) app_start: Starting scheduler on CPU0
I (405) main_task: Started on CPU0
I (405) main_task: Calling app_main()
W (415) rmt: channel resolution loss, real=10666666
I (415) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (435) phy: phy_version: 200,0, 8a31b65, Jun  6 2023, 20:13:10
I (445) phy: libbtbb version: 98b760f, Jun  6 2023, 20:13:22
I (445) main_task: Returned from app_main()
I (575) ESP_TL_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (575) ESP_TL_ON_OFF_LIGHT: Touchlink target started
I (15205) ESP_TL_ON_OFF_LIGHT: Device is on permit join status
I (15205) ESP_TL_ON_OFF_LIGHT: Touchlink target : network started
I (15205) ESP_TL_ON_OFF_LIGHT: Touchlink target finished: success
I (19425) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (20385) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (21345) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (22295) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (23255) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (28045) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (29005) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (41785) ESP_TL_ON_OFF_LIGHT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```

xieqinan, 2023-10-24:
Hello @muc-martin 

> Later I want to change the state of the pins (high / low output) of the esp32h2 for interrupts on another MC. Do you have a clue how to implement it?

What is the primary challenge you are facing? Is it that the hardware of esp32h2 doesn't support four buttons? Or, do you need assistance with driving the four buttons? Alternatively, are you unsure about how to send commands to interrupt another microcontroller?

xieqinan, 2023-10-24:
@ArturProchniak ，

The [touchlink_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink/touchlink_light) serves as the touchlink target, opening its network and waiting for the [touchlink initiator](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_touchlink/touchlink_switch) to join. The network remains open for approximately 1 minute. Therefore, in the example, you SHOULD ensure the touchlink initiator joins the touchlink within this 1-minute window.

chshu, 2023-11-22:
> t is the primary challenge you are facing? Is it that the hardware of esp32h2 doesn't support four buttons? Or, do you need assistance with driving the four buttons? Alternatively, are you unsure about how to send commands to interrupt another microcontroller?

@muc-martin Any feedback on these questions? Please close the issue if you have made it work.

closed 2024-01-08 by chshu