espressif/esp-zigbee-sdk #86: Crash in zb_zcl_send_report_attr_command

nomis, 2023-08-30:
Running https://github.com/nomis/candle-dribbler/tree/d75972b860a44723541fe4816a9c7f64ce870ed6 (with `NUTT_MAX_LIGHTS=4` aka "Component config" -> "Candle Dribbler" -> "Number of lights supported") on an ESP32-C6:

I enabled on/off endpoints 71 to 74 (which will cause on/off endpoints 81 to 84 to be enabled using `esp_zb_zcl_set_attribute_val()`) and it crashed after updating the value of the last one. It doesn't happen every time. All changes are made from the Zigbee task outside of any callbacks.

```
D (1995203) nutt.Light: Light 1 set persistent enable 0 -> 1
D (1995213) nutt.Light: Light 1 set temporary enable 0 -> 1 (auto)
I (1995213) nutt.Light: Light 1 update state 0 -> 0
D (1995213) nutt.Device: Refresh
D (1995213) nutt.Light: Light 1 report temporary enable 1
D (1996723) nutt.Light: Light 2 set persistent enable 0 -> 1
D (1996733) nutt.Light: Light 2 set temporary enable 0 -> 1 (auto)
I (1996733) nutt.Light: Light 2 update state 0 -> 0
D (1996733) nutt.Device: Refresh
D (1996733) nutt.Light: Light 2 report temporary enable 1
D (1997933) nutt.Light: Light 3 set persistent enable 0 -> 1
D (1997943) nutt.Light: Light 3 set temporary enable 0 -> 1 (auto)
I (1997943) nutt.Light: Light 3 update state 0 -> 0
D (1997943) nutt.Device: Refresh
D (1997943) nutt.Light: Light 3 report temporary enable 1
D (1998823) nutt.Light: Light 4 set persistent enable 0 -> 1
D (1998833) nutt.Light: Light 4 set temporary enable 0 -> 1 (auto)
I (1998833) nutt.Light: Light 4 update state 0 -> 0
D (1998833) nutt.Device: Refresh
D (1998833) nutt.Light: Light 4 report temporary enable 1
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42033eba  RA      : 0x42033eba  SP      : 0x408249f0  GP      : 0x4080f9d0
TP      : 0x40811ed0  T0      : 0x40022494  T1      : 0x4081e92c  T2      : 0x0000003f
S0/FP   : 0x4081e984  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000000
A2      : 0x00000000  A3      : 0x0000ffff  A4      : 0x00000003  A5      : 0x00000003
A6      : 0x00000000  A7      : 0x00000003  S2      : 0x00000020  S3      : 0x00000008
S4      : 0x4081e984  S5      : 0x00000017  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x0000008b  T4      : 0x00000086  T5      : 0x40824de4  T6      : 0x00000001
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000003
MHARTID : 0x00000000

Stack memory:
408249f0: 0x00000000 0x00000001 0x00000002 0x0000001b 0x00000068 0x00000001 0x0002ddcc 0x00000000
40824a10: 0x00000000 0x00000000 0x00000032 0x00000008 0x00000020 0x00000000 0x4081e984 0x42038226
40824a30: 0x00000001 0x4081435e 0x00000001 0x0002ddcc 0x00000000 0x00000001 0x00000000 0x42021674
40824a50: 0x000000fa 0x420381ae 0x00000008 0x00353600 0x00000000 0x00000000 0x00000000 0x00000000
40824a70: 0x00000000 0x00000000 0x40822adc 0x4203bc84 0x00000000 0x00000000 0x40824d60 0x4201588e
40824a90: 0x00000000 0x00000000 0x00000000 0x42010c32 0x00000000 0x00000000 0x00000000 0x420616c8
40824ab0: 0x00000000 0x00000000 0x00000000 0x40822ac8 0x00000000 0x00000000 0x40822adc 0x42000628
40824ad0: 0x00000000 0x00000000 0x00000000 0x4080af20 0x00000000 0x00000000 0x00000000 0x00000000
40824af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40824b10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x408248c0 0x00046716 0x40812b10 0x40812b10
40824b30: 0x40824b20 0x40812b08 0x00000014 0x4081c1d0 0x4081c1d0 0x40824b20 0x00000000 0x00000005
40824b50: 0x40822b1c 0x6267697a 0x6d5f6565 0x006e6961 0x00e23aed 0x00000000 0x40824b10 0x00000004
40824b70: 0x00000000 0x00000005 0x00000000 0x00000000 0x00000000 0x021105cd 0x00000009 0x4081a2c8
40824b90: 0x4081a330 0x4081a398 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
40824bb0: 0x42067b24 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824bd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824bf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824c70: 0x00000000 0x00000000 0x00000000 0x40000000 0x0000000d 0x40819904 0x40819904 0x40824c7c
40824c90: 0x00000022 0x00000000 0x00000abe 0x0029f630 0x00000000 0x0000ff00 0x000000fa 0x0003d090
40824cb0: 0x00000000 0x00000019 0x40819904 0x40819904 0x00000000 0x40824e04 0x64643a32 0x40824cb0
40824cd0: 0x00000036 0x00000000 0x4087d5f0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40824cf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000080 0x4087d28c
40824d10: 0x00000000 0x00000003 0x88161263 0xe0878364 0x197ff665 0x000000ff 0x000000ff 0x000000ff
40824d30: 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff
40824d50: 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff
40824d70: 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x000000ff 0x00000095
40824d90: 0x40819904 0x40819904 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... 255 lines cut ...]
.iram0.bss 0x4080f1d0 0x0 RW
.dram0.data 0x4080f1d0 0x3728 RW A
.flash.text 0x42000020 0x6a8fa R XA
.flash.appdesc 0x42070020 0x100 R  A
.flash.rodata 0x42070120 0x12c40 RW A
.eh_frame 0x42082d60 0xe98 R  A
.flash.rodata_noload 0x42083bf8 0x0 RW
.dram0.heap_start 0x408198f0 0x0 RW
.coredump.tasks.data 0x40824b20 0x160 RW
.coredump.tasks.data 0x40824950 0x1c0 RW
.coredump.tasks.data 0x4081db10 0x160 RW
.coredump.tasks.data 0x4081da30 0xd0 RW
.coredump.tasks.data 0x40825e80 0x160 RW
.coredump.tasks.data 0x40825d20 0x150 RW
.coredump.tasks.data 0x4081a6cc 0x160 RW
.coredump.tasks.data 0x4081bf70 0xf0 RW
.coredump.tasks.data 0x4082703c 0x160 RW
.coredump.tasks.data 0x40826ee0 0x150 RW
.coredump.tasks.data 0x408289f8 0x160 RW
.coredump.tasks.data 0x40828820 0x1d0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
```
```

nomis, 2023-08-30:
The crash happens about two minutes later:
```
2023-08-30 22:55:43.860388 D (768029) nutt.Light: Light 4 report temporary enable 1
2023-08-30 22:57:56.123116 Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

1. Disable endpoints 71 to 74
2. Disable endpoints 81 to 84
3. Enable endpoints 81 to 84

nomis, 2023-08-31:
It's also crashing in the same place on startup without doing anything other than connecting to the network:
```
42033e8a <zb_zcl_send_report_attr_command>:
42033e8a:       7139                    addi    sp,sp,-64
42033e8c:       de06                    sw      ra,60(sp)
42033e8e:       dc22                    sw      s0,56(sp)
42033e90:       da26                    sw      s1,52(sp)
42033e92:       d84a                    sw      s2,48(sp)
42033e94:       d64e                    sw      s3,44(sp)
42033e96:       d452                    sw      s4,40(sp)
42033e98:       d256                    sw      s5,36(sp)
42033e9a:       d05a                    sw      s6,32(sp)
42033e9c:       ce5e                    sw      s7,28(sp)
42033e9e:       8a2a                    mv      s4,a0
42033ea0:       8aae                    mv      s5,a1
42033ea2:       02e55703                lhu     a4,46(a0)
42033ea6:       00655683                lhu     a3,6(a0)
42033eaa:       00454603                lbu     a2,4(a0)
42033eae:       00255583                lhu     a1,2(a0)
42033eb2:       00154503                lbu     a0,1(a0)
42033eb6:       d22ff0ef                jal     ra,420333d8 <zb_zcl_get_attr_desc_manuf_a>
42033eba:       00354403                lbu     s0,3(a0) <-- here
```

I've previously had crashes in `zb_zcl_get_attr_desc_manuf_a()` but it looks like it now gets further and crashes after calling that function.

nomis, 2023-09-01:
This is caused by the zboss library calling `zb_zcl_send_report_attr_command()` with `manuf_code=0` which then means `zb_zcl_get_attr_desc_manuf_a()` returns `NULL` and there is no checking of this value before it's dereferenced to get the `access` value. For some reason this only happens on the 4 highest numbered endpoints.

You need to:
#. Fix all the instances where `zb_zcl_get_attr_desc_manuf_a()` returning `NULL` is not handled properly.
#. Determine why `zb_zcl_send_report_attr_command()` is being called with `manuf_code=0` and fix it.

```
2023-09-01 10:44:34.027754 I (8619) nutt.ZigbeeDevice: Joined network successfully (6f:06:fc:4a:5b:92:dd:9c/0x8813@26 as 0x6de5)
2023-09-01 10:44:34.138897 W (8729) nutt.ZigbeeDevice: NLME status indication: 09 0xfa82 00

2023-09-01 10:59:25.706256 D (900289) nutt: zb_zcl_send_report_attr_command(0x4087f2f4 { direction=0, ep=11, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=65535 }, 23)
2023-09-01 10:59:25.725572 D (900289) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=11, dst.profile_id=260 }
2023-09-01 10:59:25.730244 D (900309) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:25.735968 D (900309) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x4087f066
2023-09-01 10:59:25.744161 D (900319) nutt: ... { id=0, type=16, access=0x15, manuf_code=65535 data_p=0x4087f020 }

2023-09-01 10:59:25.797244 D (900379) nutt: zb_zcl_send_report_attr_command(0x40820ad0 { direction=0, ep=21, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=65535 }, 23)
2023-09-01 10:59:25.816490 D (900379) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=21, dst.profile_id=260 }
2023-09-01 10:59:25.821298 D (900399) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:25.827109 D (900399) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x4087f3e2
2023-09-01 10:59:25.835233 D (900409) nutt: ... { id=0, type=16, access=0x15, manuf_code=65535 data_p=0x4087f39c }

2023-09-01 10:59:25.907198 D (900489) nutt: zb_zcl_send_report_attr_command(0x40820e4c { direction=0, ep=31, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=65535 }, 23)
2023-09-01 10:59:25.926461 D (900489) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=31, dst.profile_id=260 }
2023-09-01 10:59:25.931203 D (900509) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:25.936989 D (900509) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x40820bbe
2023-09-01 10:59:25.945157 D (900519) nutt: ... { id=0, type=16, access=0x15, manuf_code=65535 data_p=0x40820b78 }

2023-09-01 10:59:26.006996 D (900589) nutt: zb_zcl_send_report_attr_command(0x40821798 { direction=0, ep=81, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=0 }, 23)
2023-09-01 10:59:26.026091 D (900589) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=81, dst.profile_id=260 }
2023-09-01 10:59:26.030881 D (900609) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:26.035978 D (900609) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x0
2023-09-01 10:59:26.040026 W (900619) nutt: fixing manuf_code
2023-09-01 10:59:26.044943 D (900619) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:26.050700 D (900629) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x4082150a
2023-09-01 10:59:26.058897 D (900629) nutt: ... { id=0, type=16, access=0x15, manuf_code=65535 data_p=0x408214c4 }

2023-09-01 10:59:26.073853 D (900639) nutt: zb_zcl_send_report_attr_command(0x40822d6c { direction=0, ep=82, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=0 }, 23)
2023-09-01 10:59:26.093102 D (900659) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=82, dst.profile_id=260 }
2023-09-01 10:59:26.097876 D (900679) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:26.103049 D (900679) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x0
2023-09-01 10:59:26.107058 W (900689) nutt: fixing manuf_code
2023-09-01 10:59:26.112013 D (900689) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:26.117737 D (900699) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x40822ade
2023-09-01 10:59:26.125812 D (900699) nutt: ... { id=0, type=16, access=0x15, manuf_code=65535 data_p=0x40822a98 }

2023-09-01 10:59:26.141207 D (900709) nutt: zb_zcl_send_report_attr_command(0x40823f7c { direction=0, ep=73, cluster_id=6, cluster_role=1, attr_id=0, flags=71, run_time=58612, manuf_code=65535 }, 23)
2023-09-01 10:59:26.160459 D (900719) nutt: ... { min_interval=0, max_interval=900, delta=0x00 0x00 0x00 0x00, reported_value=0xff 0xff 0xff 0xff, def_min_interval=0, def_max_interval=0, dst.short_attr=0, dst.endpoint=73, dst.profile_id=260 }
2023-09-01 10:59:26.165245 D (900739) nutt: zb_zcl_get_attr_desc_manuf_a()...
2023-09-01 10:59:26.171016 D (900749) nutt: zb_zcl_get_attr_desc_manuf_a() = 0x40823cee
[... cut ...]
	zb_zcl_attr_t *attr = zb_zcl_get_attr_desc_manuf_a(rep_info->ep, rep_info->cluster_id, rep_info->cluster_role, rep_info->attr_id, rep_info->manuf_code);
	ESP_LOGD(TAG, "zb_zcl_get_attr_desc_manuf_a() = %p", attr);
	if (attr) {
		ESP_LOGD(TAG, "... { id=%u, type=%u, access=0x%02x, manuf_code=%u data_p=%p }", attr->id, attr->type, attr->access, attr->manuf_code, attr->data_p);
	} else if (rep_info->manuf_code == 0) {
		ESP_LOGW(TAG, "fixing manuf_code");
		rep_info->manuf_code = ZB_ZCL_MANUFACTURER_WILDCARD_ID;
		goto retry;
	} else {
		ESP_LOGE(TAG, "ignoring call");
		return;
	}

	__real_zb_zcl_send_report_attr_command(rep_info, param);
}
```

nomis, 2023-09-02:
Successfully reconfiguring reporting on the co-ordinator for all of the endpoints (at least in terms of communication; #65 still means I don't get all reported changes) causes it to stop using `manuf_code == 0`, so there's something wrong in the stored reporting configuration that causes this, possibly related to adding more endpoints after previously joining a network.

chshu, 2024-12-19:
The subsequent issues all have been resolved, closing this as well. Feel free to reopen if any other pending issues.