espressif/esp-zigbee-sdk #88: ESP_ZB_ON_OFF_LIGHT: Failed to initialize Zigbee stack (status: ESP_FAIL)

tianrongqin, 2023-08-31:
I can't understand why a development board burns esp zigbee sdk examples esp_ Zigbee_ HA_ Sample HA_ On_ Off_ Switch and another development board burn esp zigbee sdk examples esp_ Zigbee_ HA_ Sample HA_ On_ Off_ Why can't light connect, HA_ On_ Off_ Light displays content as ESP_ ZB_ ON_ OFF_ LIGHT: Failed to initialize Zigbee stack (status: ESP-FAIL)，Is this function already implemented? We still need to modify the code ourselves？

tianrongqin, 2023-08-31:
I know the problem how to solve.When such problems arise，we should carry the order "idy.py -p （COM） erase-flash".

xieqinan, 2023-08-31:
The root cause is possibly due to flashing the development board with esp-zb-light, which could have retained previous network information. This prevents it from joining older networks and results in the issue you're experiencing.

xieqinan, 2023-09-01:
Hello @tianrongqin ,
Do you have other questions for this issue? If the issue has been fixed, please close it, thanks.

closed 2023-09-01 by tianrongqin

xieqinan, 2024-11-25:
Hi @Sisylam ,

Based on your logs, the dimmable light successfully joined the network. I believe the `Failed to initialize Zigbee stack (status: ESP_FAIL)` log should not appear in the `ESP_ZB_COLOR_DIMM_LIGHT` example. Could you clarify the issue you are encountering?  

By the way, if the message `Device started up in non-factory-reset mode` is printed, it indicates that the device is already part of the network from a previous configuration.

krambriw, 2025-11-21:
To recover, this way also worked fine for me. Anyone know why this has to be like this?

> Walters-MacBook-Air-2:~ walterkrambringpython3 -m esptool --chip esp32-C6 erase-flash
> esptool v5.1.0
> Connected to ESP32-C6 on /dev/cu.usbmodem11201:
> Chip type:          ESP32-C6FH4 (QFN32) (revision v0.2)
> Features:           Wi-Fi 6, BT 5 (LE), IEEE802.15.4, Single Core + LP Core, 160MHz
> Crystal frequency:  40MHz
> USB mode:           USB-Serial/JTAG
> MAC:                a0:85:e3:ff:fe:af:60:e4
> BASE MAC:           a0:85:e3:af:60:e4
> MAC_EXT:            ff:fe
> 
> Stub flasher running.
> 
> Flash memory erased successfully in 1.3 seconds.
> 
> Hard resetting via RTS pin...
>