espressif/esp-zigbee-sdk #89: Cannot connect when the policy of the coordinator is set to enable

tianrongqin, 2023-08-31:
There is a situation where the content burned on one of my development boards is esp-zigbee-sdk\examples\esp_zigbee_cli and the content burned on another development board is esp-zigbee-sdk\examples\esp_zigbee_HA_sample\HA_on_off_light.When I set  the policy of development board which is as a Coordinator as enable that burn esp-zigbee-sdk\examples\esp_zigbee_cli ，I will find the other device which is as a endpoint can't  connect the coordinator.I don't know what to cause this problem and how to solve this problem.Here are two separate printed contents：
I (178201) ESP_ZB_CLI: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (178201) ESP_ZB_CLI: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (252691) ESP_ZB_CLI: ZDO signal: NWK Permit Join (0x36), status: ESP_OK

I (202320) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (202320) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)

xieqinan, 2023-08-31:
If the `bdb -i policy enable` is called, the end device requires to use the install code mode to join a network.

lingzhichen, 2023-08-31:
@xieqinan Yes,I have already operated the one you mentioned, but I still cannot connect.Before I set the policy to enable,I have already executed "bdb -i set...."

xieqinan, 2023-08-31:
We have detailed the usage of the install code in our documentation. Please refer to the [README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_cli/README.md#ic) for more information.

closed 2023-09-01 by tianrongqin