espressif/esp-zigbee-sdk #90: Handling of handling raw commands

simonborje, 2023-08-31:
Hi, I'm trying to capture custom commands sent from a 3rd party remote control to my gateway application running on a ESP32-C6.

I am currently using the following library versions:
    espressif/esp-zboss-lib: "~0.6.2"
    espressif/esp-zigbee-lib: "~0.9.3"

My current attempt involves using `esp_zb_raw_command_handler_register` to register a raw command handler callback.
I saw something similar being suggested in https://github.com/espressif/esp-zigbee-sdk/issues/78#issuecomment-1695533471 which led me in this direction.
I can see that the callback is being called for every button press on the remote which looks promising.
However, I have not been able to figure out how to properly inspect the raw command that triggered the callback.

So far I have tried using `ZB_ZCL_DEVICE_CMD_PARAM_CMD_INFO(bufid)` and similar calls but that results in linker errors:
`c:/temp/main/esp_zigbee_gateway.cpp:665: undefined reference to 'zb_buf_get_tail_func(unsigned char, unsigned int)'`

Is there any additional configuration that I need to do to resolve the linker issues or should I perhaps take a different approach to handle the raw commands?

xieqinan, 2023-09-01:
Hello,
Do you add the header `#include "zboss_api.h"` to `esp_zigbee_gateway.cpp` file?

simonborje, 2023-09-01:
Yes, the header is included. Without it the compilation step failed instead.

xieqinan, 2023-09-01:
Please attach a part code of raw command handler for me to address this issue, thanks.

simonborje, 2023-09-03:
The linker issue seems to be caused by the main source file being a .cpp file instead of a .c file. I managed to resolve the link issue by wrapping the `#include "zboss_api.h"` line in `extern "C"`.

I'm now not sure how to get hold of and parse the data based on the `bufid` argument provided by the callback.
I have tried `ZB_ZCL_DEVICE_CMD_PARAM_CMD_INFO` and some other functions that take a bufid as an argument but none of the have seemed to return anything I could make sense of.

Are there any examples of how to correctly handle the callback?

simonborje, 2023-09-09:
The reply in https://github.com/espressif/esp-zigbee-sdk/issues/94#issuecomment-1706280284 solved the issue for me, thanks!

closed 2023-09-09 by simonborje

xieqinan, 2023-09-11:
> The reply in [#94 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/94#issuecomment-1706280284) solved the issue for me, thanks!

No problem.