espressif/esp-zigbee-sdk #93: Always running as " ESP_OTA_CLIENT: OTA receiving ..."

tianrongqin, 2023-09-04:
When I burn the esp_ota_client and esp_ota_server and then run them,I find they can not run like what the readme say.The esp_ota_client is running as " ESP_OTA_CLIENT: OTA receiving ..." and should not complete upgrade.

tianrongqin, 2023-09-04:
E (1128935) boot_comm: mismatch chip ID, expected 16, found 14
E (1128935) ESP_ZIGBEE_CORE: Error: Image validation failed, image is corrupted! err=0x1503
E (1128935) boot_comm: mismatch chip ID, expected 16, found 14
E (1128945) ESP_ZIGBEE_CORE: esp_ota_set_boot_partition failed (ESP_ERR_OTA_VALIDATE_FAILED)!

xieqinan, 2023-09-05:
Hello @tianrongqin ,

The default `ota_file.bin` in the `esp_ota_client` example is meant for the ESP32H4, not the ESP32H2. Please replace it with the correct firmware you expected, which can be found in the `/build` folder for each ESP compiled example.

closed 2023-09-05 by tianrongqin