espressif/esp-zigbee-sdk #94: Handling not all responses

BogatovT, 2023-09-04:
Hi,
I`m using Zigbee end devices based on Z-stack and ESP as coordinator device. Z-stack has a feature of automatic responding changed attributes. For example when Im requesting to set current level attribute, I will get response with value of current level.
ESP is not handling it in callbacks. Im setting callback with esp_zb_core_action_handler_register() but it never handling these responses. 

But if I set another callback with esp_zb_raw_command_handler_register() it always triggered on there responses.
I have no idea how I can provide more details about it. If someone answer in the [nearby topic](https://github.com/espressif/esp-zigbee-sdk/issues/90#issuecomment-1704331672) I can try it. Thanks!

mallaprashant, 2023-09-04:
I have a similar problem. After using esp_zb_raw_command_handler_register(), esp_zb_core_action_handler_register() become useless. Every callback is triggered in esp_zb_raw_command_handler_register() and as can see below that  esp_zb_raw_command_handler_register(device_cb) function only able to stop error message "Device callback id(0x5f) has not been handled" and 0x5f ever triggers.

>     
        bool device_cb(uint8_t bufid)
         {
               if (ZB_ZCL_DEVICE_CMD_PARAM_CB_ID(bufid) == 0x5f) {
                  zb_zcl_door_lock_read_lock_door_res_payload_t *payload =
                  ZB_ZCL_DEVICE_CMD_PARAM_OUT_GET(bufid, zb_zcl_door_lock_read_lock_door_res_payload_t);
                  // indicate the status of door lock for the stack to response. 
                  payload->status = ZB_ZCL_STATUS_SUCCESS;
                  ZB_ZCL_DEVICE_CMD_PARAM_STATUS(bufid) = RET_OK;
              }
          return true;
        }

BogatovT, 2023-09-05:
@xieqinan 
I was implemented simple callback to look into the buffer when I should get responses. But it always has only 2 bytes with zero values. 
```c
bool rawCmdHandlerCb(uint8_t bufid) {
    uint8_t buf[zb_buf_len(bufid)];
    memcpy(buf, zb_buf_begin(bufid), sizeof(buf));
    ESP_LOGI("RAW", "bufid: %d size: %d", bufid, sizeof(buf));
    for (int i = 0; i < sizeof(buf); ++i) {
        printf("0x%02X ", buf[i]);
    }
    printf("\n");
    return true;
}
``` 
> I (31183) RAW: bufid: 18 size: 2
0x00 0x00 

I cant find any other Zboss functions to read more data from the buffer. Do they exist?

xieqinan, 2023-09-05:
@BogatovT ,
Please use the following code to get the information of response command. 
```
bool rawCmdHandlerCb(uint8_t bufid) {
    uint8_t buf[zb_buf_len(bufid)];
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
    printf("cluster id: 0x%x, command id: %d\n", cmd_info->cluster_id, cmd_info->cmd_id);
    memcpy(buf, zb_buf_begin(bufid), sizeof(buf));
    ESP_LOGI("RAW", "bufid: %d size: %d", bufid, sizeof(buf));
    for (int i = 0; i < sizeof(buf); ++i) {
        printf("0x%02X ", buf[i]);
    }
    printf("\n");
    return true;
}
```

BogatovT, 2023-09-05:
Im sending current level attribute and then receiving `cluster id: 0x08, command id: 11`

mallaprashant, 2023-09-07:
@xieqinan, May I kindly request the update of the following commands, specifically "ZB_ZCL_DOOR_LOCK_LOCK_DOOR_CB_ID," "ZB_ZCL_DOOR_LOCK_UNLOCK_DOOR_CB_ID," "ZB_ZCL_DOOR_LOCK_LOCK_DOOR_RESP_CB_ID," and "ZB_ZCL_DOOR_LOCK_UNLOCK_DOOR_RESP_CB_ID," within the Zigbee stack?

mallaprashant, 2023-09-07:
@xieqinan, May I kindly request the update or implementation of the following commands, specifically "Door lock command, Door unlock command, Door lock response command and door unlock response command " within the Zigbee stack?

chshu, 2023-11-22:
> @xieqinan, May I kindly request the update or implementation of the following commands, specifically "Door lock command, Door unlock command, Door lock response command and door unlock response command " within the Zigbee stack?

@mallaprashant The door lock callbacks have been supported already: https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L86

chshu, 2023-11-22:
@BogatovT I'm closing the issue since the origin question has been resloved, feel free to file new issues if any.

closed 2023-11-22 by chshu