espressif/esp-zigbee-sdk #96: Add Support for Carbon Dioxide Cluster 0x040d

lmahmutov, 2023-09-06:
I'm working on first and easy project Air Sensor
Fully working example placed on github. https://github.com/lmahmutov/esp32_c6_co2_sensor
but for complete i need support for Carbon Dioxide Cluster. At this time Co2 reported to custom cluster.

lmahmutov, 2023-09-06:
I try to add custom cluster with cluster ID 0x040d, but firmware says "Cluster ID not in avaliable range"

RadekDymak, 2023-09-19:
Whem the Carbon Dioxide Cluster 0x040d will be added?

u236, 2023-09-19:
and we :)

lmahmutov, 2023-09-19:
it can be realised using native ZBOSS SDK, without ESP-ZigBee-Sdk. But binding and configureation report for cluster not working. Only manual reporting.
Fully working native CO2 cluster example.
https://github.com/lmahmutov/esp32_c6_co2_sensor/blob/esp32_c6_co2_ZBOSS-WIP/main/esp_co2_sensor.c#L43

closed 2023-10-11 by espressif-bot

xieqinan, 2023-10-11:
Hello @lmahmutov @u236 @vke @RadekDymak ,

The [Carbon Dioxide cluster](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_carbon_dioxide_measurement.h) has been supported in esp-zigbee-sdk v1.0.0. Could you please try using and testing it.