espressif/esp-zigbee-sdk #98: Crash after receiving ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID

simonborje, 2023-09-09:
After using `esp_zb_zcl_config_report_cmd_req` to request attribute reporting from a device the `zb_action_handler` function gets called with `callback_id` is set to `ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID`.

```c++
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    switch (callback_id)
    {
    case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
        ret = zb_configure_report_resp_handler((esp_zb_zcl_cmd_config_report_resp_message_t *)message);
        break;
    ...
```
Examining the data of the `esp_zb_zcl_cmd_config_report_resp_message_t message` suggests that the `message->attribute_id` field contains random garbage data with a different value every time. Much of the `message->info` part does look ok, with status being `ESP_ZB_ZCL_STATUS_SUCCESS` and many other fields having reasonable values, e.g.
```gdb
(gdb) p *message
$2 = {info = {status = ESP_ZB_ZCL_STATUS_SUCCESS, src_address = {addr_type = 0 '\000', u = {short_addr = 49444, src_id = 49444, ieee_addr = "$Á\000\000\000\000\000"}},
    dst_address = 0, src_endpoint = 1 '\001', dst_endpoint = 1 '\001', cluster = 6, profile = 260, command = {id = 7 '\a', direction = 1 '\001', is_common = 1 '\001'}},
  attribute_id = 56811}
```

Additionally, the program often crashes right after receiving this message with the following backtrace:
```gdb
#0  panic_abort (details=details@entry=0x4087e11c "assert failed: stop_current_operation esp_ieee802154_dev.c:235 (false)")
    at C:/Users/simon/esp/esp-idf/components/esp_system/panic.c:452
#1  0x40807698 in esp_system_abort (details=details@entry=0x4087e11c "assert failed: stop_current_operation esp_ieee802154_dev.c:235 (false)")
    at C:/Users/simon/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4080c008 in __assert_func (file=file@entry=0x420af693 "esp_ieee802154_dev.c", line=line@entry=235, func=<optimized out>,    
    func@entry=0x420afc64 <__func__.0> "stop_current_operation", expr=expr@entry=0x420abc0c "false") at C:/Users/simon/esp/esp-idf/components/newlib/assert.c:81
#3  0x420a5d4a in stop_current_operation () at C:/Users/simon/esp/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:235
#4  0x420a5d60 in tx_init (frame=frame@entry=0x4081403c <tx_buf> "§÷»\377\001rjØ") at C:/Users/simon/esp/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:618
#5  0x420a5ee0 in ieee802154_transmit (frame=0x4081403c <tx_buf> "§÷»\377\001rjØ", cca=<optimized out>)
    at C:/Users/simon/esp/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:633
#6  0x4205fb9a in esp_ieee802154_transmit (frame=<optimized out>, cca=<optimized out>) at C:/Users/simon/esp/esp-idf/components/ieee802154/esp_ieee802154.c:243
#7  0x4205ec2a in mac_esp_transmit_frame ()
#8  0x4205ef66 in mac_esp_send_packet ()
#9  0x4201fe7a in zb_mac_send_frame ()
#10 0x42020748 in zb_mac_logic_iteration ()
#11 0x4201e92e in zb_multimac_mac_logic_iteration_proxy ()
#12 0x4201ec5c in zb_sched_loop_iteration ()
#13 0x4204914e in zboss_main_loop_iteration ()
#14 0x42011166 in esp_zb_main_loop_iteration ()
#15 0x42006a7e in esp_zb_task (pvParameters=<error reading variable: value has been optimized out>) at C:/temp/esp32-c6-c++/main/esp_zigbee_gateway.cpp:713
#16 0x40809a2c in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at C:/Users/simon/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:202
```

xieqinan, 2023-09-12:
Hello @simonborje 

> Examining the data of the `esp_zb_zcl_cmd_config_report_resp_message_t message` suggests that the `message attribute_id` field contains random garbage data with a different value every time. Much of [...]

Only the attribute id will be reported in the case of an error. According to your description, when the status is equal to `ESP_ZB_ZCL_STATUS_SUCCESS` the `attribute id` field should not be represented. We will add a coment for this `attribute id` field to explain the case.

xieqinan, 2023-09-12:
> Additionally, the program often crashes right after receiving this message with the following backtrace:

To address this issue, we have developed a log tool specifically for IEEE802.15.4 to assist in debugging crashes. You can follow these steps to obtain the crash log:

Run 'idf.py menuconfig' to access the configuration menu.
Enable the 'ieee802154 debug' option under 'research.'
Also, make sure to enable the 'record state information' and 'record command information' options.
Reproduce the phenomenon that led to the crash.
Provide us with the generated log for further analysis and feedback.

simonborje, 2023-09-12:
This is the output I got after enabling the debug options:

```
...
I (4429) ESP_ZB_GATEWAY: Configure report response: status(0), cluster(0x6), attribute(0x40b5)
W (1913258645) ieee802154_debug: Print the record state, current state index: 0
W (1913258645) ieee802154_debug: index  0: line:     , state:   DISABLE, timestamp: 8531335223256833416
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42061cf4  RA      : 0x42061cca  SP      : 0x4087e1f0  GP      : 0x4080e150
0x42061cf4: ieee802154_assert_print at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/driver/esp_ieee802154_debug.c:200 (discriminator 1)

0x42061cca: ieee802154_assert_print at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/driver/esp_ieee802154_debug.c:200 (discriminator 1)

TP      : 0x408686e0  T0      : 0x4002289a  T1      : 0xffffffe0  T2      : 0x00000000
0x4002289a: usb_serial_device_tx_one_char in ROM

S0/FP   : 0x00000001  S1      : 0x40817000  A0      : 0x720a0295  A1      : 0x720a0295
A2      : 0x00000000  A3      : 0x96969400  A4      : 0x40815a98  A5      : 0xd8a217f0
A6      : 0x00000000  A7      : 0x00000001  S2      : 0x40815000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x40817000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0xd8a217f0
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087e1f0: 0x408197b8 0x40819938 0x408197b8 0x00000000 0x40815000 0x40817000 0x40814cc4 0x420a8632
0x420a8632: stop_current_operation at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:241 (discriminator 1)

4087e210: 0x0001000f 0x00000018 0x408109b0 0x420a86c8 0x00000098 0x40814cc4 0x00000001 0x420a891a
0x420a86c8: tx_init at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:658

0x420a891a: ieee802154_transmit at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:674

4087e230: 0x408108f4 0x00000000 0x0000616d 0x42061af2 0x00000000 0x00000007 0x00000001 0x42060b80
0x42061af2: esp_ieee802154_transmit at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/ieee802154/esp_ieee802154.c:244

0x42060b80: mac_esp_transmit_frame at zb_esp_mac.c.obj:?

4087e250: 0x00000009 0x00000001 0x00000000 0x42060ebe 0x00000000 0x00000000 0x00000000 0x420209fa
0x42060ebe: mac_esp_send_packet at ??:?

0x420209fa: zb_mac_send_frame at ??:?

4087e270: 0x00000000 0xc1240000 0x00000000 0x00010000 0x00000000 0x00000000 0x00000000 0x0f4087e3
4087e290: 0x00000000 0x4087e830 0x0000003a 0x00000121 0x00000000 0x40817000 0x00000000 0x420212c2
0x420212c2: zb_mac_logic_iteration at ??:?

4087e2b0: 0x00000000 0x00000001 0x4081052e 0x4201f44a 0x00000000 0x00000001 0x4081052e 0x4201f714
0x4201f44a: zb_multimac_mac_logic_iteration_proxy at ??:?

0x4201f714: zb_sched_loop_iteration at ??:?

4087e2d0: 0x00000000 0x4202e00a 0x00070000 0x423b3a00 0x00000078 0x00000008 0x4087e87c 0x4087e8c8
0x4202e00a: zb_mcps_data_confirm at ??:?

4087e2f0: 0x4087ec78 0x4087ee28 0x4087e9f0 0x4204ab60 0x00000000 0x00000000 0x4087ef80 0x420116a2
0x4204ab60: zboss_main_loop_iteration at ??:?

0x420116a2: esp_zb_main_loop_iteration at ??:?

4087e310: 0x4087ec78 0x4087ee28 0x4087e9f0 0x42006e10 0x00030000 0x00000000 0x00140000 0x00000000
0x42006e10: esp_zb_task(void*) at C:/temp/esp32-c6-c++/main/esp_zigbee_gateway.cpp:714 (discriminator 2)

4087e330: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4080a19e
0x4080a19e: vPortTaskWrapper at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205
[... 12 lines cut ...]
4087e470: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e490: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e4b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e4d0: 0x00000000 0x00000000 0x00000000 0x40000000 0x00000054 0x00000000 0x4087e4e4 0x00000000
0x40000000: _start in ROM

4087e4f0: 0x00000000 0x00000000 0x4087e4fc 0xffffffff 0x4087e4fc 0x4087e4fc 0x00000000 0x4087e510
4087e510: 0xffffffff 0x4087e510 0x4087e510 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000
4087e530: 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4087e53c 0x00000000 0x00000000 0x00000000
4087e550: 0x4087e554 0xffffffff 0x4087e554 0x4087e554 0x00000000 0x4087e568 0xffffffff 0x4087e568
4087e570: 0x4087e568 0x00000001 0x00000001 0x00000000 0x1d00ffff 0x00000000 0xb33fffff 0x00000000
4087e590: 0x0000000c 0x031a0008 0x00000000 0x4087f450 0x00000040 0x00000000 0x00000001 0x00000000
4087e5b0: 0x00000000 0x00000000 0x4087e5e8 0xb33fffff 0x00000000 0xffffffff 0x00000000 0x00000000
4087e5d0: 0x00000000 0x00000000 0x4087e640 0x00000000 0x00000000 0x00000054 0x00000000 0x4087e5e8

ELF file SHA256: 0000000000000000
Warning: checksum mismatch between flashed and built applications. Checksum of built application is e5fe41967fd1c7316dedc8f3b526f202f3840f414e18b7d63a3577e2a07d8036

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM
```

xieqinan, 2023-09-13:
Hi @simonborje ,
our log is valuable, but we require additional debugging information. Please use the following configuration settings in sdkconfig to enable more extensive debug logging.
[image]

Furthermore, I attempted to reproduce the crash by pressing the button to `esp_zb_zcl_config_report_cmd_req()` command, but it worked fine. If possible, could you please provide the steps to reproduce it.

simonborje, 2023-09-13:
Hi @xieqinan,

I created a smaller project that can recreate the crash by combining the two examples `customized_client` and `esp_zigbee_gateway` to create a project where the ESP32-C6 acts as a coordinator that I join a 3rd party zigbee lamp to. This was the easiest way I could find since I don't have access to a second ESP32-C6 at the moment.

[customized_client.zip](https://github.com/espressif/esp-zigbee-sdk/files/12600961/customized_client.zip)

By running this project on the ESP32-C6 and joining the network with my 3rd party lamp I can get the application to crash by pressing the button to send a new `esp_zb_zcl_config_report_cmd_req`. It doesn't crash every time, sometimes it works for 10+ presses and other times the first one triggers a crash.

Strangely enough, the crashes in these cases are different from the ones I got with my full project. Here there are usually no `ieee802154` functions in the call stack as can be seen below:

```
I (55247) ESP_HA_ON_OFF_SWITCH: Reveived report from address(0x1212) src endpoint(1) to dst endpoint(1) cluster(0x6)
I (55247) ESP_HA_ON_OFF_SWITCH: Received report information: attribute(0x0), type(0x10), value(1)

I (58847) ESP_HA_ON_OFF_SWITCH: Button pressed
I (58927) ESP_HA_ON_OFF_SWITCH: Configure report response: status(0), cluster(0x6), attribute(0xcb3e)

assert failed: zb_osif_serial_put_bytes zb_esp_osif_serial.c:103 (false)
Core  0 register dump:
MEPC    : 0x4080062e  RA      : 0x40807f42  SP      : 0x4087eb30  GP      : 0x4080e540
0x4080062e: panic_abort at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/esp_system/panic.c:452

0x40807f42: __ubsan_include at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x40870134  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x0000007f  S1      : 0x00000001  A0      : 0x4087eb6c  A1      : 0x4080e799
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40816000
A6      : 0x00000014  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4087ec88
S4      : 0x4080e798  S5      : 0x0000000e  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4087eb30: 0x16e1a3b8 0xf9445950 0x42076cf4 0x4080cd7a 0x4080edb4 0x42076cf4 0x4080f09c 0x42076be8
0x4080cd7a: newlib_include_assert_impl at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/newlib/assert.c:93

4087eb50: 0x4080edc4 0x4087eb64 0x4080edc8 0x42076ba8 0x4080e798 0x00333031 0xac195adf 0x65737361
4087eb70: 0x66207472 0x656c6961 0x7a203a64 0x736f5f62 0x735f6669 0x61697265 0x75705f6c 0x79625f74
4087eb90: 0x20736574 0x655f627a 0x6f5f7073 0x5f666973 0x69726573 0x632e6c61 0x3330313a 0x61662820
4087ebb0: 0x2965736c 0x00000000 0x00000000 0x00000000 0x40815000 0x40817000 0x40815c68 0x00000000
4087ebd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087ebf0: 0x40815000 0x00000006 0x40815c68 0x40031756 0x00000000 0xffffffff 0x4087f078 0x420770b0
0x40031756: sprintf in ROM

4087ec10: 0x00000004 0x4087ec98 0x40816543 0x4201a50e 0x7ffffffc 0xffff0208 0xffffffff 0x4080a336
0x4201a50e: zb_trace_put_bytes at ??:?

0x4080a336: __getreent at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/freertos/esp_additions/freertos_tasks_c_additions.h:40

4087ec30: 0x4087f048 0x00000000 0x00000000 0x42000000 0x00000004 0x00000018 0x00000018 0x00000098
4087ec50: 0x0001000f 0x00000018 0x00000006 0x420551d8 0x40880000 0x00000018 0x00000014 0x4205237a
0x420551d8: zb_osif_serial_init at ??:?

0x4205237a: zb_osif_mserial_put_bytes at ??:?

4087ec70: 0x600a3000 0x00000003 0x00000014 0x4201a50e 0x00000000 0x0000000e 0x40814eac 0x4201a9c6
0x4201a50e: zb_trace_put_bytes at ??:?

0x4201a9c6: zb_trace_put_hdr at ??:?

4087ec90: 0x0001000f 0x00000018 0x0212adde 0x40800f0a 0x00000005 0x00000018 0x0000049b 0x4201a5fc
0x40800f0a: esp_intr_enable at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/esp_hw_support/intr_alloc.c:798

0x4201a5fc: zb_trace_msg_port_vl at ??:?

4087ecb0: 0x408108f0 0x00000000 0x4087ecf4 0x00000008 0x00000000 0x00000018 0x40811014 0x4201a646
[... 22 lines cut ...]
4087ed90: 0x4087f338 0x4087f4a8 0x4087f360 0x42007ad4 0x00000000 0x00000000 0x000a0000 0x00000000
0x42007ad4: esp_zb_task at C:/temp/esp-zigbee-sdk/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c:334 (discriminator 2)

4087edb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4080a440
0x4080a440: vPortTaskWrapper at C:/Users/simon/esp/esp-idf-5.1.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

4087edd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087edf0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x4087ebe0
4087ee10: 0x000016f3 0x4080f7e0 0x4080f7e0 0x4087ee0c 0x4080f7d8 0x00000014 0x40818864 0x40818864
4087ee30: 0x4087ee0c 0x00000000 0x00000005 0x4087de08 0x6267695a 0x6d5f6565 0x006e6961 0x00000000
4087ee50: 0x00000000 0x4087ee00 0x00000005 0x00000000 0x00000000 0x00000000 0x00000009 0x408172c8
[... cut ...]
4087eed0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087eef0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087ef10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 4210c38140000000
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 2bb8ffc6bcdc198947a191f55038deee1363974eb7a1abe0bd26ba45faf7e9ff

Rebooting...
```

xieqinan, 2023-09-21:
Hi @simonborje 
I have followed the steps below to attempt to reproduce the crash:

1. I used the code from `customized_client` to modify the `esp-zigbee-gateway` example, turning it into a switch.
2. I configured the `esp_zb_buttons_handler()` to send the `esp_zb_zcl_config_report_cmd_req()` command.
3. I flashed the modified `esp-zigbee-gateway` onto the ESP32C6.
4. I used `esp-zigbee-HA-sample/HA_on_off_light` as the Zigbee end device to join the Zigbee gateway.
5. I pressed the button on the ESP32C6 to continuously send the `config_report_cmd` to the Zigbee light.
6. Continuously, I observed the log message: `ESP_ZB_GATEWAY: Configure report response: status(0), cluster(0x6), attribute(0xffff)`.

However, it worked well without any crashes. Do these steps meet the conditions required to reproduce the issue you are facing?

simonborje, 2023-09-26:
Hi @xieqinan 
Since updating to the latest esp-zboss-lib and esp-zigbee-lib versions I have not seen much of this issue, perhaps not at all. At most I have had a single crash out of several hundred attempts, but I'm not even sure if that crash was really triggered by this issue.

As it stands now I consider the problem resolved and will close the issue.

Thank you for your help!

closed 2023-09-26 by simonborje