espressif/esp-zigbee-sdk #99: esp_zb_sleep_now() does not wake up from gpio

blastrock, 2023-09-10:
Hello,

I'm trying to make my device sleep, and wake it up when a button is pressed. I am not sure what `esp_zb_sleep_now()` does. My signal handler looks like this:

```c
  case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP:
    ESP_ERROR_CHECK(
        gpio_wakeup_enable(GPIO_INPUT_IO_TOGGLE_SWITCH, GPIO_INTR_LOW_LEVEL));
    ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup());
    esp_zb_sleep_now();
    ESP_LOGI(TAG, "wake up cause %d", esp_sleep_get_wakeup_cause());
    break;
```

But when I push my button, the controller will not wake up from sleep. And when it wakes up, the cause is always 4 (the timer).

Is this supposed to work, and my issue is somewhere else? What does the `esp_zb_sleep_now()` function do? Is there a way to just call `esp_light_sleep_start()` instead?

kelin6, 2023-09-18:
@blastrock  `ESP_ZB_COMMON_SIGNAL_CAN_SLEEP` is the zigbee stack notify user stack can sleep, `esp_zb_sleep_now` will sleep the time that stack give, and it will put device into sleep mode(close RF/ Peripheral power down).

About wake up device  by GPIO, you need call `ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup());` before `esp_zb_init()`, it should not be called here every time.

agustinebrecht, 2023-09-18:
Hello @kelin6  how are you?, is it possible to modify the time spent by the zigbee stack? In the Nordic zigbee sdk, it is possible to modify this time that is passed to the zb_osif_sleep function, in order to force more or less sleep time.

kelin6, 2023-09-19:
@agustinebrecht  In the Nordic zigbee stack, there's a function call `zb_osif_sleep(sleep_tmo)` where sleep_tmo represents the amount of time the ZBOSS stack determines as suitable for sleeping, based on the time until the next scheduled task. This sleep duration cannot be directly modified.

If you wish to extend or shorten the sleep duration, you can adjust the value of [ED_KEEP_ALIVE](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/sleepy_end_device/main/esp_zb_sleepy_end_device.h#L20)

agustinebrecht, 2023-09-19:
Thank you very much for your reply @kelin6 . I take the opportunity to ask her about the gpio wake up that she mentioned to @blastrock . How do I indicate which pins have to wake me up due to interruption and which ones don't? since I have seen that the esp32H2 does not have configurations enabled in the esp-idf to establish that gpio raises the micro

blastrock, 2023-09-21:
Thanks for the answer, I experimented a bit more. Unfortunately, moving the esp_sleep_enable_gpio_wakeup call does not seem to change anything.

I tried changing the value of `esp_pm_config::light_sleep_enable`, and calling `esp_light_sleep_start` instead of `esp_zb_sleep_now`. Of course, this breaks the zigbee stack, but this is just to check if the gpio wake up works. Here are the results:

- no pm light sleep + esp_light_sleep_start : ok
- pm light sleep + esp_light_sleep_start : no
- no pm light sleep + esp_zb_sleep_now : no (but zigbee works nonetheless, the wake up cause is always 0 though)
- pm light sleep + esp_zb_sleep_now (as documented) : no

I am a bit surprised by this zigbee sleep API. Can't we ask the zigbee stack for how long we need to sleep (or let it set up the rtc wake up) and do the `esp_light_sleep_start` call ourselves?

In case it helps, I pushed my code [here](https://github.com/blastrock/HA_on_off_light/blob/master/main/esp_zb_light.c)

kelin6, 2023-10-07:
@blastrock  The GPIO power domain is on TOP. In the Zigbee sleep example, the TOP is powered down, so GPIO cannot be used for wake-up. Therefore, in the Zigbee sleep example, RTC_IO can be used as a wake-up source, as follows:

```
#define BOOT_BUTTON_NUM         GPIO_NUM_9     
#define GPIO_WAKEUP_NUM         BOOT_BUTTON_NUM  

esp_sleep_enable_ext1_wakeup(1ULL << GPIO_WAKEUP_NUM, ESP_EXT1_WAKEUP_ANY_HIGH);  
```
[image]

blastrock, 2023-10-11:
> In the Zigbee sleep example, the TOP is powered down

How is it disabled? Is it the `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` config? Is it possible to just enable it back?

> RTC_IO can be used as a wake-up source, as follows

I tried using RTC as you suggested. I completely disabled switch_driver.c's code so that the GPIO usage doesn't interfere with RTC, and I added this code to my main function:

```
  ESP_ERROR_CHECK(
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON));
  ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(
      1ULL << GPIO_INPUT_IO_TOGGLE_SWITCH, ESP_EXT1_WAKEUP_ANY_HIGH));
  rtc_gpio_pullup_dis(GPIO_INPUT_IO_TOGGLE_SWITCH);
  rtc_gpio_pulldown_en(GPIO_INPUT_IO_TOGGLE_SWITCH);
```

I am using GPIO0 here. This code works when I test it with a `while(true) esp_light_sleep_start();`, but I still can't make it work with the zigbee sleep. I do have some wake ups with cause `3`, but they seem to occur a bit randomly, definitely not right when I connect the pin. It usually takes a multiple seconds, almost as if it woke up because of the clock, despite the cause showing as `3`

kelin6, 2023-10-24:
@blastrock 
```
#define BOOT_BUTTON_NUM         GPIO_NUM_9     
#define GPIO_WAKEUP_NUM         BOOT_BUTTON_NUM  
esp_sleep_enable_ext1_wakeup(1ULL << GPIO_WAKEUP_NUM, ESP_EXT1_WAKEUP_ANY_HIGH);  
```

Using the code above is indeed feasible, and I have tested it. The occasional `3` printing occurs because the `esp_zb_sleep_now();` is followed by the print statement. This causes the `reason` to be printed only when `esp_timer` wakes up. I recommend adding a GPIO interrupt service function to print the wake-up `reason` within it. This way, when GPIO wakes up and triggers the GPIO interrupt, the `wake-up reason` can be obtained immediately. You can refer to the following example [GPIO wakeup](https://github.com/espressif/esp-idf/pull/12126/files#diff-5bf8d8f498b92995184f728cd795935d755bd10ad926bd1dfbddb27202661e76R83).

blastrock, 2023-10-29:
Thanks a lot for your help!

I realized that `esp_zb_sleep_now` does not actually make the device sleep, it makes the *task* sleep, and then the OS puts the device to sleep when there are no pending tasks. This, and your other advises helped me debug my code.

I have a couple bugs left. The first one is that when I press the button just before the zigbee task goes to sleep, the `esp_zb_zcl_on_off_cmd_req` function returns, but the zigbee message is not sent until the device wakes up again.

Then, I noticed that a new zigbee sdk was released, so I upgraded from 0.6.0 to 1.0.1 to see if the previous bug has been already fixed. Now, when I call `esp_zb_sleep_enable`, my device won't join anymore and crashes. When trying to join the network, it gets a LEAVE RESET signal. After a few seconds, it crashes with the following log:

```
I (634) ieee802154: ieee802154 mac sleep retention initialization
I (652) zigbee: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (653) zigbee: ZDO signal: ZDO Skip Start Up (0x1), status: ESP_OK
I (655) zigbee: Zigbee stack initialized
I (661) zigbee: ZDO signal: BDB Device Start (0x5), status: ESP_OK
I (667) zigbee: Start network steering
I (651) main_task: Returned from app_main()
I (9271) zigbee: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (9645) zigbee: zigbee wakeup
I (9647) zigbee: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (9788) zigbee: zigbee wakeup
I (9789) zigbee: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (9884) zigbee: zigbee wakeup
I (19292) zigbee: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (19293) zigbee: Reset device

abort() was called at PC 0x42017bb3 on core 0
0x42017bb3: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063a  RA      : 0x40809966  SP      : 0x4087e150  GP      : 0x4080fbf4  
0x4080063a: panic_abort at /home/blastrock/esp/esp-idf/components/esp_system/panic.c:452

0x40809966: __ubsan_include at /home/blastrock/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4086da5c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4087e1b4  A0      : 0x4087e17c  A1      : 0x4087e1b2  
A2      : 0x00000000  A3      : 0x4087e1a9  A4      : 0x00000001  A5      : 0x40817000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x0000077b  
S4      : 0x00000038  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
4087e150: 0x4081b1e0 0x40880000 0x4087e1b0 0x4080e2be 0x4206d0e8 0x00000225 0x00000001 0x408106a0
0x4080e2be: abort at /home/blastrock/esp/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4087e170: 0x4087e1b4 0x408106bc 0x4087e1b0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4087e190: 0x43502074 0x34783020 0x37313032 0x20336262 0x63206e6f 0x2065726f 0x00000030 0x42050000
0x42050000: esp_task_wdt_reset at /home/blastrock/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:783 (discriminator 3)

4087e1b0: 0x00000030 0x31303234 0x33626237 0x00000700 0x00000000 0x00000000 0x00000038 0x42017bb6
0x42017bb6: zb_globals_init at ??:?

4087e1d0: 0x4087ee0c 0x4087ee34 0x40812070 0x42026274 0x00000000 0x00000000 0x4087ce74 0x4201c7f4
0x42026274: zb_multimac_mlme_reset_request_proxy_send_req at nwk_multimac_proxy.c.obj:?

0x4201c7f4: zb_sched_loop_iteration at ??:?

4087e1f0: 0x00000000 0x42026260 0x0000077b 0x423b391e 0x0000003c 0x00000004 0x4087ed98 0x4087ede4
0x42026260: zb_multimac_debug_transaction_stuck_handler at nwk_multimac_proxy.c.obj:?

4087e210: 0x4087ee0c 0x4087ee34 0x4087ce74 0x42037cbe 0x00000000 0x00000000 0x4087f544 0x420103ae
0x42037cbe: zboss_main_loop_iteration at ??:?

0x420103ae: esp_zb_main_loop_iteration at ??:?

[... 19 lines cut ...]
4087e3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
0x40000000: _start in ROM

4087e410: 0x00000020 0x00000000 0x00000000 0x00000003 0x42071920 0x4081b0b0 0x00000001 0xb33fffff
4087e430: 0x00000000 0x00000054 0x00000000 0x4087e438 0x00000000 0x00000000 0x00000000 0x4087e450
4087e450: 0xffffffff 0x4087e450 0x4087e450 0x00000000 0x4087e464 0xffffffff 0x4087e464 0x4087e464
4087e470: 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0x00000054
4087e490: 0x00000000 0x4087e490 0x00000000 0x00000000 0x00000000 0x4087e4a8 0xffffffff 0x4087e4a8
4087e4b0: 0x4087e4a8 0x00000000 0x4087e4bc 0xffffffff 0x4087e4bc 0x4087e4bc 0x00000001 0x00000001
4087e4d0: 0x00000000 0x4000ffff 0x00000000 0xb33fffff 0x00000000 0x00000054 0x00000000 0x4087e4e8
0x4000ffff: ppRxFragmentProc in ROM

4087e4f0: 0x00000000 0x00000000 0x00000000 0x4087e500 0xffffffff 0x4087e500 0x4087e500 0x00000000
4087e510: 0x4087e514 0xffffffff 0x4087e514 0x4087e514 0x00000001 0x00000001 0x00000000 0x1d00ffff
4087e530: 0x00000000 0xb33fffff 0x00000000 0x00000028 0xffff0000 0x00002000 0x70030000 0x4081a100

ELF file SHA256: 496866645743efa3

Rebooting...
```

Sometimes it happens in another function like map_vram_to_pram, but I can't reproduce it anymore. I pushed my code here: https://github.com/blastrock/HA_on_off_light/blob/wip/main/esp_zb_light.c

Should I open another issue for this?

Bosemani, 2023-11-10:
Hai @blastrock 
Did you solved this issue?
I'm also facing same!

blastrock, 2023-11-10:
I haven't, I have no idea how I can investigate this bug.

blastrock, 2023-11-15:
I have reported this as a proper issue in https://github.com/espressif/esp-zigbee-sdk/issues/148 to give it more visibility. It includes more detailed steps to reproduce the issue.

kelin6, 2023-11-21:
@blastrock 
* Regarding the first issue: When the Zigbee task goes to sleep and a button is pressed before that, the Zigbee on/off command will be sent only after the device wakes up again. I have reviewed your code snippet:
```
case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP:
    ESP_ERROR_CHECK(
        gpio_wakeup_enable(GPIO_INPUT_IO_TOGGLE_SWITCH, GPIO_INTR_LOW_LEVEL));
    esp_zb_sleep_now();
    // esp_light_sleep_start();
    ESP_LOGI(TAG, "wake up cause %d", esp_sleep_get_wakeup_cause());

    if (esp_sleep_get_wakeup_cause() & ESP_SLEEP_WAKEUP_GPIO)
        check_gpio(button_func_pair, PAIR_SIZE(button_func_pair));
    break;
```
The issue appears to be that `check_gpio(button_func_pair, PAIR_SIZE(button_func_pair)) ` is called after `esp_zb_sleep_now()`. It should be called within the GPIO interrupt [IRAM_ATTR gpio_isr_handler](https://github.com/Laiwen-Li/esp-idf/blob/Add_PM_Example/examples/power_management/external_power_control/main/gpio_control/gpio_control.c#L83). 

According to your current code, When the button is pressed before the Zigbee stack can sleep, the Zigbee stack sends a sleep signal (`ESP_ZB_COMMON_SIGNAL_CAN_SLEEP`) to the user after it becomes possible for Zigbee to sleep. After that, `esp_zb_sleep_now` is called, causing the device to sleep. The device wakes up when `esp_zb_sleep_now ` exits, and `check_gpio(button_func_pair, PAIR_SIZE(button_func_pair));`  is then executed, leading to the delayed sending of the Zigbee on/off command until the device wakes up again.

*  Regarding the second issue, updating to IDF version v5.1.2 should resolve it, as mentioned in another one of your issues #148.

blastrock, 2023-11-21:
> It should be called within the GPIO interrupt [IRAM_ATTR gpio_isr_handler](https://github.com/Laiwen-Li/esp-idf/blob/Add_PM_Example/examples/power_management/external_power_control/main/gpio_control/gpio_control.c#L83).

Yes, I changed that after your previous message, but didn't push my code yet, I was still experimenting.

> Regarding the second issue, updating to IDF version v5.1.2 should resolve it

According to this comment https://github.com/espressif/esp-zigbee-sdk/issues/148#issuecomment-1813875241 ESP32-C6 is not supported at all...?

I have just tested upgrading to 5.1.2, I can confirm that I'm using the right version:

```
I (17) boot: ESP-IDF v5.1.2 2nd stage bootloader
```

However, I still have the exact same crash with the sleepy_end_device example.

kelin6, 2023-11-22:
@blastrock  Yes, as of `ESP-IDF V5.1 (commit: 482a8fb2d7)`, Zigbee sleep for ESP32-C6 is currently not supported. However, Zigbee sleep is supported for ESP32-H2, and I have confirmed and tested this. 

The relevant support code for ESP32-C6 Zigbee sleep has been internally reviewed and merged. It may take a few weeks for these changes to be visible on GitHub. I will notify you promptly once the support is available on GitHub.

blastrock, 2023-11-22:
I only have a C6, so I can't test on an H2. I just noticed I didn't specify I was using a C6 on this issue, sorry for that.

I'll just wait for the next release then. Thanks for your help!

chshu, 2023-12-01:
@blastrock The C6 light sleep issue has been fixed in IDF master, could you update both esp-zigbee-sdk and esp-idf to the latest master? Let us know if you see any other issues.

kelin6, 2023-12-20:
@blastrock  Based on your code, I have reproduced the same phenomenon. The issue lies in the fact that the task stack allocation for `switch_driver_button_detected` with a size of 2048 is too small, leading to a stack overflow and system crash. After increasing the `switch_driver_button_detected` task stack size to 4096, everything runs normally. The modifications based on your code are as follows:

[image]

Below is the log during runtime and the current situation of RTC IO wakeup:

```
I (23) boot: ESP-IDF v5.3-dev-867-gf0607f6d3b-dirty 2nd stage bootloader
I (24) boot: compile time Dec 19 2023 21:39:49
I (25) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 000a2800
I (81) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (89) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=12d98h ( 77208) map
I (125) esp_image: segment 1: paddr=00022dc0 vaddr=40800000 size=05258h ( 21080) load
I (130) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=652e0h (414432) map
I (215) esp_image: segment 3: paddr=0008d308 vaddr=40805258 size=0b730h ( 46896) load
I (227) esp_image: segment 4: paddr=00098a40 vaddr=40810990 size=01e6ch (  7788) load
I (229) esp_image: segment 5: paddr=0009a8b4 vaddr=50000000 size=00058h (    88) load
I (236) boot: Loaded app from partition at offset 0x10000
I (239) boot: Disabling RNG early entropy source...
I (255) cpu_start: Unicore app
W (264) clk: esp_perip_clk_init() has not been implemented yet
I (271) cpu_start: Pro cpu start user code
I (271) cpu_start: cpu freq: 160000000 Hz
I (271) cpu_start: Application information:
I (274) cpu_start: Project name:     light_bulb
I (279) cpu_start: App version:      6cb8617
I (284) cpu_start: Compile time:     Dec 20 2023 12:24:37
I (290) cpu_start: ELF file SHA256:  e63539e69...
I (296) cpu_start: ESP-IDF:          v5.3-dev-867-gf0607f6d3b
I (302) cpu_start: Min chip rev:     v0.0
I (307) cpu_start: Max chip rev:     v0.99 
I (312) cpu_start: Chip rev:         v0.0
I (316) heap_init: Initializing. RAM available for dynamic allocation:
I (323) heap_init: At 40819C50 len 000629C0 (394 KiB): RAM
I (330) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (336) heap_init: At 50000058 len 00003F90 (15 KiB): RTCRAM
I (343) spi_flash: detected chip: generic
I (347) spi_flash: flash io: dio
W (351) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (364) sleep: Enable automatic switching of GPIO sleep configuration
I (371) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (379) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (387) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (395) sleep_sys_periph: HP System sleep retention initialization
I (402) sleep_sys_periph: TEE/APM sleep retention initialization
I (408) sleep_sys_periph: UART sleep retention initialization
I (415) sleep_sys_periph: Timer Group sleep retention initialization
I (422) sleep_sys_periph: IO Matrix sleep retention initialization
I (429) sleep_sys_periph: SPI Mem sleep retention initialization
I (436) sleep_sys_periph: SysTimer sleep retention initialization
I (453) coexist: coex firmware version: e41c5cb
I (453) coexist: coexist rom version 5b8dcfa
I (454) main_task: Started on CPU0
I (456) main_task: Calling app_main()
I (460) zigbee: wake up early 0
I (470) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 160, APB_MIN: 160, Light sleep: ENABLED
I (474) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:5 
I (484) esp_zb_sleep: Enable ieee802154 zigbee light sleep, the wake up source is ESP timer
I (493) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
[... 214 lines cut ...]
I (40179) switch: GPIO interrupts: on
I (40183) switch: GPIO interrupts: off
I (40183) switch: GPIO ISR HANDLER cause 3
I (40193) switch: queue event
I (40196) switch: loop resume
I (40200) switch: state is 0, value is 1
I (40205) switch: state is 2, value is 1
I (40209) switch: state is 2, value is 1
I (40214) switch: state is 2, value is 1
I (40218) switch: state is 2, value is 1
I (40223) switch: state is 2, value is 1
I (40228) switch: state is 2, value is 1
I (40232) switch: state is 2, value is 0
I (40237) switch: state is 4, value is 0
I (40241) zigbee: Send 'on_off toggle' command
I (40246) zigbee: done!
I (40249) switch: back to sleep
I (40253) switch: GPIO interrupts: on
I (41825) zigbee: zigbee wakeup
I (41831) zigbee: ZDO signal: Can Sleep (0x16), status: ESP_OK
I (46898) zigbee: zigbee wakeup
I (46904) zigbee: ZDO signal: Can Sleep (0x16), status: ESP_OK

```
[image]

blastrock, 2023-12-23:
Thanks a lot for your help! So the short stack was indeed the cause of the crash. Are there tools to help debug that? Or was that just your intuition?

Anyway, I have one last bug. When I press the button, I call `esp_zb_zcl_on_off_cmd_req` to toggle a light (see https://github.com/blastrock/HA_on_off_light/blob/1bb427431ab81140c54f131d57b7eff554094398/main/esp_zb_light.c#L66 ). However, if I press the button during sleep, the device wakes up, registers the button press and goes back to sleep, but the zigbee message isn't sent yet. Later when the device wakes up again because of the zigbee task, it finally sends the message.

It looks like `esp_zb_zcl_on_off_cmd_req` should wake up the zigbee task to send the message, but it does not. Is there something else I did wrong in my code?

kelin6, 2023-12-25:
> Thanks a lot for your help! So the short stack was indeed the cause of the crash. Are there tools to help debug that? Or was that just your intuition?
>
> Anyway, I have one last bug. When I press the [...]

@blastrock 
There are two ways to debug:
1) Normally, esp-idf monitors stack overflow using esp_cpu_set_watchpoint. However, during sleep, the CPU is powered down, and the watchpoint-related functionality is not fully supported in sleep mode. Therefore, it won't prompt for a stack overflow after a crash. For example, based on your code, you can configure `CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=n` to keep the CPU powered, then compile your code. You should then be able to see the following information:
[image]

2) You can inspect the stack usage in the corresponding task by using the following code:
```
size_t stackHighWaterMark = uxTaskGetStackHighWaterMark(NULL);
esp_rom_printf("Task Stack High Water Mark: %u bytes\n", stackHighWaterMark);
```
  Print the stack consumption in the "button_detected" task to determine that ESP_LOGI requires the most stack, almost 1 kbytes, while the "button_detected" task is allocated only 2 kbytes in total. When executing subsequent code, this leads to insufficient stack resources.
  
  
  Regarding the issue of "esp_zb_zcl_on_off_cmd_req waking up the Zigbee task to send the message", this is a known problem, and we are currently discussing potential solutions.

blastrock, 2023-12-25:
>  There are two ways to debug:

Thanks for the tips, I'll check that next time I have a crash.

> Regarding the issue of "esp_zb_zcl_on_off_cmd_req waking up the Zigbee task to send the message", this is a known problem, and we are currently discussing potential solutions.

Ok, thanks for looking into it, I'll wait for your update then.

susch19, 2024-02-24:
> Regarding the issue of "esp_zb_zcl_on_off_cmd_req waking up the Zigbee task to send the message", this is a known problem, and we are currently discussing potential solutions.

Is there any update on this? When I try to send a button via light sleep, the behavior is still, that it sends it as soon as the task from Zigbee is woken up after the timer. I also tried to use vTaskResume with the handler of the Zigbee Main Task, but that didn't work either. 
As far as I understand the only solution is to use deep sleep? Because then the Zigbee task never sleeps and waking up the esp via ext1 from deep sleep always works.

closed 2024-04-01 by espressif-bot

xieqinan, 2024-04-01:
@blastrock @susch19 @agustinebrecht ,

The esp-zigbee-sdk v1.2.2 now supports immediate wake up from GPIO. Could you please try using the new version to test this issue again?

susch19, 2024-04-14:
> @blastrock @susch19 @agustinebrecht ,
> 
> The esp-zigbee-sdk v1.2.2 now supports immediate wake up from GPIO. Could you please try using the new version to test this issue again?

Just had the time to test it and it works wonderfully. I have an end device with 30s long poll interval, for "long" light sleeps and when I press a button the message is immediately send and received. The delay is nearly not existent / as good as with other zigbee end devices.