espressif/esp-zigbee-sdk #100: OTA Upgrade "Query Next Image Request Command" file version field is always zero

u236, 2023-09-11:
Hello! We're testing OTA Upgrade features with your client example (https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota/ota_client), and have found little bug:

After **Image Notify Command** sent from our "server", test device sends **Query Next Image Request Command**, but value of "File Version" field is always zero.

Some "client" code hunks:
https://github.com/espressif/esp-zigbee-sdk/blob/7f37d655d70dc2101b6901857bf5f7ecd6b27bb9/examples/esp_zigbee_ota/ota_client/main/esp_ota_client.h#L22-L24
https://github.com/espressif/esp-zigbee-sdk/blob/7f37d655d70dc2101b6901857bf5f7ecd6b27bb9/examples/esp_zigbee_ota/ota_client/main/esp_ota_client.c#L107-L110

**Query Next Image Request Command** dump and parsed data:
```
2023.09.11 11:34:09.540 (inf) zigbee:    request dump:        "01:01:10:11:10:00:00:00:00:01:01"
2023.09.11 11:34:09.540 (inf) zigbee:    request field ctrl:  "1"
2023.09.11 11:34:09.540 (inf) zigbee:    request manuf. code: "1001"
2023.09.11 11:34:09.540 (inf) zigbee:    request image type:  "1011"
2023.09.11 11:34:09.540 (inf) zigbee:    request file ver.:   "0"
```
This bug can cause "infinite" OTA Upgrades in some cases :)

xieqinan, 2023-09-13:
@u236 
Thank you for reminder, we will fix it.

xieqinan, 2023-10-16:
Hello @u236 ,

There is a workaround available, outlined below:
```
    esp_zb_ota_cluster_cfg_t ota_cluster_cfg = {
        .ota_upgrade_file_version = 0x01010111,
        .ota_upgrade_downloaded_file_ver = OTA_UPGRADE_FILE_VERSION,
        .ota_upgrade_manufacturer = OTA_UPGRADE_MANUFACTURER,
        .ota_upgrade_image_type = OTA_UPGRADE_IMAGE_TYPE,
    };
```

closed 2023-10-23 by espressif-bot

xieqinan, 2023-10-23:
@u236 ,

This issue has been resolved in the esp-zigbee-lib v1.0.1.