espressif/esp-zigbee-sdk #101: Joined network successfully but get no temperature log on sever side

vedatbotuk, 2023-09-13:
Hi everyone,

I try to send some temperature date from node(client) to coordinator(server). As coordinator I have use example "esp_zb_on_switch" as node and "esp_zb_light" as coordinator.
I edited this code on node side like this:
```
        // Here I register the device as temperature_sensor
       esp_zb_temperature_sensor_cfg_t temp_cfg = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();
       esp_zb_ep_list_t *esp_zb_temp_ep = esp_zb_temperature_sensor_ep_create(HA_ESP_TEMP_CLIENT_ENDPOINT, &temp_cfg); 
       esp_zb_device_register(esp_zb_temp_ep);

        // Here I try to send some data to cooardinator
        esp_zb_zcl_report_attr_cmd_t tmp_attr;
        tmp_attr.zcl_basic_cmd.src_endpoint = HA_ESP_TEMP_CLIENT_ENDPOINT;
        tmp_attr.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        tmp_attr.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        tmp_attr.cluster_role = ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE;
        tmp_attr.attributeID = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID;
        ESP_EARLY_LOGI(TAG, "Attribute: %d", ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID);
        esp_zb_zcl_report_attr_cmd_req(&tmp_attr);

        //Here find a match
        esp_zb_zdo_match_desc_req_param_t  find_req;
        find_req.addr_of_interest = 0x0000;
        find_req.dst_nwk_addr = 0x0000;
        esp_zb_zdo_match_cluster(&find_req, user_find_cb, NULL);   

        // Here bind_req      
        ESP_LOGI(TAG, "Found SERVER");
        esp_zb_zdo_bind_req_param_t temp_bind_req;
        light_bulb_device_params_t *temp_sensor = (light_bulb_device_params_t *)malloc(sizeof(light_bulb_device_params_t));
        temp_sensor->endpoint = endpoint;
        temp_sensor->short_addr = addr;
        esp_zb_ieee_address_by_short(temp_sensor->short_addr, temp_sensor->ieee_addr);
        esp_zb_get_long_address(temp_bind_req.src_address);
        temp_bind_req.src_endp = HA_ESP_TEMP_CLIENT_ENDPOINT;
        temp_bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        temp_bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(temp_bind_req.dst_address_u.addr_long, temp_sensor->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        temp_bind_req.dst_endp = endpoint;
        temp_bind_req.req_dst_addr = esp_zb_get_short_address(); /* TODO: Send bind request to self */
        ESP_LOGI(TAG, "Try to bind On/Off");
        esp_zb_zdo_device_bind_req(&temp_bind_req, bind_cb, (void *)temp_sensor);
         
```

On the server side:
```
        //Register
        esp_zb_temperature_sensor_cfg_t temp_cfg = ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG();
        esp_zb_ep_list_t *esp_zb_temp_ep = esp_zb_temperature_sensor_ep_create(HA_ESP_TEMP_SERVER_ENDPOINT, &temp_cfg);
        esp_zb_device_register(esp_zb_temp_ep);
```
I just want to ```esp_err_t zb_attribute_handler``` show me some log about the data from node.
Can someone help me why I did not get no data on server side?

Here also a log!
[image]

xieqinan, 2023-09-13:
@vedatbotuk ,
I believe that the failure is caused by the error in the `match_req()` function. Please refer to issue [#78](https://github.com/espressif/esp-zigbee-sdk/issues/78) to restructure your matching packet.

vedatbotuk, 2023-09-13:
Hi @xieqinan , thank you for your quick reply.

I think the problem is in another functions, which I tried to implement.
Which function I have to use for sending data to coordinator?
```esp_zb_zcl_report_attr_cmd_t tmp_attr;``` is the right one, or I have to use ```esp_zb_zcl_report_attr_message_t```?

Match function works with ligth example normal but when I want to use temperature cluster, then does not work. The issue maybe on the server side?

xieqinan, 2023-09-14:
@vedatbotuk 

For the temperature measurement cluster, the client has no dependencies, cluster-specific attributes, or specific commands generated or received. 

In contrast, the temperature measurement cluster server should use the 'Report Attributes' command, adhering to the minimum and maximum reporting intervals and considering reportable changes.

In the context mentioned above, there are some suggestions that can be helpful in achieving your goal.

Firstly, the `esp_zb_temperature_sensor_ep_create()` function is designed for creating a standard temperature sensor, specifically on the server side for temperature measurement. Instead, it is recommended to utilize the `esp_zb_cluster_list_add_temperature_meas_cluster()` function for the temperature measurement client, rather than using `esp_zb_temperature_sensor_ep_create()`.

Secondly, the `esp_zb_zdo_match_cluster()` function can be employed to match the corresponding clusters. However, it is essential to optimize its usage to avoid triggering error or warning logs.

Thirdly, the `esp_zb_zcl_report_attr_cmd_req(esp_zb_zcl_report_attr_cmd_t *req)` function can be utilized to transmit attribute reports from the server to the client. For obtaining the report information from the client, the cluster can make use of the `esp_zb_zcl_report_attr_message_t`.

Hope that these suggestions prove to be helpful for you.

vedatbotuk, 2023-09-15:
Thanks. I will try and report it.

vedatbotuk, 2023-09-15:
I've come a little further. 
On client side fails the function, where I want to receive temperature message. see next figure.
[image]

The code fails in this section of code:
```
static esp_err_t zb_configure_report_resp_handler(const esp_zb_zcl_cmd_config_report_resp_message_t *message)
{

    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);
    ESP_LOGI(TAG, "Configure report resp handler: status(%d)", message->info.status);
    ESP_LOGI(TAG, "Configure report response: status(%d), cluster(0x%x), attribute(0x%x)", message->info.status, message->info.cluster,
             message->attribute_id);
    return ESP_OK;
}

static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_REPORT_ATTR_CB_ID:
        ret = zb_attribute_reporting_handler((esp_zb_zcl_report_attr_message_t *)message);
        break;
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t *)message);
        break;
    case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
        ret = zb_configure_report_resp_handler((esp_zb_zcl_cmd_config_report_resp_message_t *)message);
        break;
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}
```
I think there is still something wrong in my code. @xieqinan , do you have some suggestion?

closed 2023-09-18 by vedatbotuk