espressif/esp-zigbee-sdk #103: Get source address of the message

vedatbotuk, 2023-09-18:
How can I get the address of the client, which sent data to coordinator. I want to get the log like following line.
Is there any method, which gives the source address of the message?
```
ESP_LOGI(TAG, "Receive temperature Value: %d from source short: 0x%04hx", temperature, XXX);
```
XXX should given the source address.

In following line you see  my attribute_handler.

```
static esp_err_t zb_attribute_handler(const esp_zb_zcl_set_attr_value_message_t *message)
{
    esp_err_t ret = ESP_OK;
    uint8_t temperature = 0;
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);
    ESP_LOGI(TAG, "Received message: endpoint(%d), cluster(0x%x), attribute(0x%x), data size(%d)", message->info.dst_endpoint, message->info.cluster,
             message->attribute.id, message->attribute.data.size);
    if (message->info.dst_endpoint == HA_ESP_LIGHT_ENDPOINT) {
        if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL) {
        if (message->attribute.id == ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_U8) {
            temperature = message->attribute.data.value ? *(uint8_t *)message->attribute.data.value : temperature;
            
            light_driver_set_power(1);
            // Blink during get data from client
            ESP_LOGI(TAG, "Receive temperature Value: ", temperature, message->info.dst_endpoint);
            vTaskDelay(20 / portTICK_PERIOD_MS);
            light_driver_set_power(0);
        } else {
            ESP_LOGW(TAG, "Level Control cluster data: attribute(0x%x), type(0x%x)", message->attribute.id, message->attribute.data.type);
        }
        }
    }
    return ret;
}
```

xieqinan, 2023-09-20:
Hello @vedatbotuk,

I'm afraid that the source address has not been provided for the `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` message. This is because this message only indicates that the stack informs the application layer about which attribute needs to be set in the actual driver, and the application doesn't need to be concerned about where the message originates. In other words, for the `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` message, the application only needs to execute the corresponding operation, as the correctness of the message has already been verified by the stack.

vedatbotuk, 2023-09-20:
Thanks, I will continue see if I can find another solution.

closed 2023-09-20 by vedatbotuk