espressif/esp-zigbee-sdk #104: ESP32-H2 sleep mode consumption

diazmanuel, 2023-09-18:
Hello!
I am trying to work with the H2 chips in low power mode and I am encountering the following consumption situations

[image]
When the chip wakes up due to the KEEP_ALIVE configured at 4000ms it presents a strange behavior. As seen in the image, the chip's current consumption skyrockets (which is understandable) but it remains too long at a consumption of 20ma. During this consumption, the signal ESP_ZB_COMMON_SIGNAL_CAN_SLEEP is generated multiple times.
the handler of the ESP_ZB_COMMON_SIGNAL_CAN_SLEEP signal is as follows
```
    case ESP_ZB_COMMON_SIGNAL_CAN_SLEEP:
            ESP_LOGI(TAG, "Zigbee can sleep");
            esp_zb_sleep_now();
    ESP_LOGI(TAG, "wake up cause %d", esp_sleep_get_wakeup_cause());
        break;
```

As seen in the following log, when the consumption is at 20ma it prints the following

[image]

On the other hand, when I read or write via Zigbee, the current consumption presents a replica. In the following image you can see the first consumption corresponding to a reading (which has a high consumption) and a short time later the consumption increases again as if it were a "replica" for some reason

[image]

Could you help me with these problems?

Thank you so much

kelin6, 2023-09-19:
@diazmanuel  I've looked at your log, and it seems that the sleep threshold is set very small, just a few milliseconds. Could you increase the threshold using `esp_zb_sleep_set_threshold` according to [README](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device#readme) ? Indeed, during the initial network setup phase of the device, there can be prolonged periods of 20mA current. However, once the network stabilizes, this phenomenon is rarely observed. I will continue to investigate the issue of prolonged 20mA current during the device's initial network entry.

There are recurring replica of 100mA current, which appear to be related to Zigbee packet retransmission. This might occur because the packets sent may not receive an acknowledgment from the other side, leading to retransmission. The 100mA current is primarily during the packet sending phase. 
Could you provide a detailed description of the reproduction process?

diazmanuel, 2023-09-22:
About the esp_zb_sleep_set_threshold threshold, this would only generate the CAN_SLEEP signal being called fewer times, but the problem is that immediately after falling asleep it wakes up. I understand that this threshold only works to guarantee that the chip stays awake for a minimum amount of time.

After analyzing the code you can identify the problem, this was occurring due to the log, the ESP_LOGI() function puts data into a task that is then responsible for printing it, this is why the chip woke up immediately after falling asleep since the afternoon of logeo required it, then again it went into the CAN_SLEEP handler and went into a loop

I suggest that you put a warning in the example since the same log of the example generates this effect of high consumption and looping

kelin6, 2023-10-24:
@diazmanuel Thank you very much for your advice, but at the moment, I haven't encountered the issue of excessive UART printing causing power consumption problems. So, this seems similar to the previous problem [sleep replica issue](https://github.com/espressif/esp-zigbee-sdk/issues/108).

Anyway, it appears that this issue has been resolved for now. If you come across any new findings, please do let us know. Thank you.

closed 2023-10-24 by diazmanuel