espressif/esp-zigbee-sdk #105: End device won't reconnect after using install code policy

mallaprashant, 2023-09-21:
Hi @xieqinan, I have diligently followed the specified steps mentioned in #82 using "HA_on_off_light" and "HA_on_off_switch" examples, successfully establishing connectivity between the end device and the coordinator through the implementation of the install code policy. However, when I manually restart the coordinator, it effortlessly reestablishes its connection with the end device. Unfortunately, the same cannot be said when I attempt to manually restart the end device. It fails to reconnect to the coordinator, and I encounter the following error message in the output log: "ESP_HA_ON_OFF_LIGHT: failed to initialize zigbee stack."

Additionally, I've encountered a similar issue when experimenting with the "esp_zigbee_customized_devices" examples, which results in the following log output: "ESP_HA_ON_OFF_SWITCH: stack BDB Device Reboot failure with ESP_FAIL status, steering."

I'm eager to gain insights into the root causes of these occurrences and would appreciate your guidance on this matter

xieqinan, 2023-09-21:
Hi @mallaprashant ,

Have you performed any erase operations on the End Device side? Please attempt to utilize `esp_zb_get_short_address()` to print the short address when receiving the `ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START` signal.

mallaprashant, 2023-09-21:
@xieqinan,
 
Yes, I performed an erase operation on the flash memory before updating the Install code policy. I want to clarify that I haven't made any modifications to the example code. Additionally, I followed your suggestion and used esp_zb_get_short_address() to print the short address upon receiving the ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START signal. However, I observed that the short address remains unchanged even after a device restart.

xieqinan, 2023-09-22:
hello,
Please indicate the channel mask when generating the manufacturer .bin file, for example:
```
python esp_zb_mfg_tool.py -i 966b9f3ef98ae6059708 -m CAFEBEEF50C0FFED -c 0x8000 -mn Espressif -mc 0x131B -c 0x07FFF800

```

mallaprashant, 2023-09-25:
it's working fine now.

closed 2023-09-25 by mallaprashant

JanneMantyharju, 2024-02-10:
@xieqinan Sorry to open up old topic, but I have much of the same problem. When I try to join the network, I'll get "Have not got nwk key" problem:
```
I (730) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (740) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (3110) ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0x42c9 on channel 11, dev 0x0, iface_id 0
I (3620) ZBOSS: mac/mac_associate.c:448   received association response
I (3630) ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0x55a6, iface_id 0, ch 11, rejoin 0
I (3630) ZBOSS: zdo/zdo_app_join.c:773   xpanid 29:d2:c5:99:8a:76:c1:bc
I (13670) ZBOSS: zdo/zdo_app.c:595   Have not got nwk key - authentication failed
I (13670) ZBOSS: nwk/nwk_main.c:4409   zb_nwk_do_leave param 5 rejoin 0
I (13670) ZBOSS: common/zb_address.c:1063   zb_address_delete ref 17
I (13690) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Leave (0x3), status: ESP_OK
I (13690) ZBOSS: nwk/nwk_join.c:251   No dev for join
I (13700) ZBOSS: nwk/nwk_join.c:251   No dev for join
I (13700) ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 0
I (13710) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (13720) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```

I used the above example to create mfg config and flashed it. Here's what esp_zb_get_short_address() returns:
`I (710) ESP_ZB_ON_OFF_LIGHT: SHORT CODE: 0xfffe`

Any ideas what to try?

xieqinan, 2024-02-29:
@JanneMantyharju ,

> I (730) ESP_ZB_ON_OFF_LIGHT: Start network steering
> I (740) ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
> I (3110) ZBOSS: [...]

The log indicates that the light device has a different global TC link key compared to the coordinator, which causes it to parse the network key from the coordinator during the "Transport key" process.

Have you used the `esp_zb_secur_ic_add()` function to add the install code of the device on the coordinator side?