espressif/esp-zigbee-sdk #106: [Feature request] Add write, read command request functions with multiple attributes

diazmanuel, 2023-09-22:
Hello, 
I was reading the zigbee documentation and it is assumed that it is possible to perform a read or write query for multiple attributes in the same message (this is especially beneficial in cases of low-power applications and optimizations) I leave you the images of the documentation

[image]

[image]

thanks

closed 2023-10-11 by espressif-bot

xieqinan, 2023-10-11:
Hello @diazmanuel,

The feature has been supported in esp-zigbee-lib v1.0.0 [ZCL general command](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L86)，Could you please try using and testing it?

xieqinan, 2023-10-25:
Hello @diazmanuel ,

Your suggestion is practical. We plan to fix this issue by ensuring that the `esp_zb_zcl_cmd_read_attr_resp_message_t` will carry the attribute list to the application layer in the upcoming version.

diazmanuel, 2023-10-25:
hello @xieqinan,
Thanks for taking this into account, it is also necessary to see the case of writing that generates an individual callback just like reading

xieqinan, 2023-11-03:
@diazmanuel ，

The issue has been fixed in esp-zigbee-sdk [v1.0.2](https://github.com/espressif/esp-zigbee-sdk), please refer to [read attribute message](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L1256).