espressif/esp-zigbee-sdk #107: Consumption reduction in low consumption application with ESP32-H2

diazmanuel, 2023-09-22:
Hello, 

I would like to ask about some aspects of reducing consumption.

- The first point I wanted to ask is if the function that manages the output power of the transmitter works correctly, I am referring to the esp_zb_set_tx_power function.  According to the documentation, this function takes as a parameter a int8_t that represents a gain (DB), shouldn't this parameter be an absolute power value (DBm)? Anyway, when I lowered the power to 0 (originally it had 20 according to its getter function) I did not notice a change in the energy consumption, could you help me with this, perhaps the call of this function should be made in some specific place as before zigbee stack initialization. Another aspect that I would like to know are the upper and lower limits that this function takes.

- On the other hand, when carrying out a project with multiple tasks, the consumption when the chip wakes up is very high as shown in the image, reaching values ​​of 100ma, is it possible to reduce this peak, what could be the cause of these current peaks?

[image]

- is there any topology to work with multiple tasks and maintain the minimum possible consumption? Currently, non-zigbee tasks wait in an xQueueRecive until the zigbee task adds data to the queue of said task. I understand that the xQueueRecive al being a blocker it sleeps the micro until it receives data but I don't know if it sleeps in the same way that the esp_zb_sleep_now() function does. Could you upload an example of how to create a low-consumption project with multiple tasks?

- When you want to put the micro to sleep there are several ways to do it, ruling out the case of deep_sleep, what is the difference between putting the micro to sleep with a vTaskDelay, light_sleep and esp_zb_sleep_now in terms of consumption, would it be the same?

kelin6, 2023-09-25:
@diazmanuel 
-  `esp_zb_set_tx_power` has a maximum value of 20dB, a minimum value of -24dB, and an adjustment step of 3dB, resulting in a total of 15 levels. The default setting is at the maximum level of 15 for transmit power, which corresponds to the highest transmission power of 100mA. To use esp_zb_set_tx_power correctly, please ensure that you call it after `esp_zb_init()`.(I have tested this)

- As mentioned earlier, 100mA represents the peak transmit power. If you want to adjust this peak, you can decrease the txpower using `esp_zb_set_tx_power`.

- It is possible to handle multiple tasks while keeping power consumption as low as possible. For example, you can create multiple tasks that utilize blocking system calls like xQueueReceive or vTaskDelay. This way, the CPU will only be awakened when an event occurs or when a specified time has passed.

- ESP32-H2 supports two main power-saving modes: Light-sleep and Deep-sleep. Light sleep is based on the FREETROS IDLE task. When the system enters the idle task (which can be achieved using vTaskDelay or xQueueReceive), ESP-IDF manages power-down operations within the idle task to minimize power consumption. The [`portSUPPRESS_TICKS_AND_SLEEP`](https://github.com/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h#L560) macro is crucial for this process, and in ESP-IDF, it is implemented through vApplicationSleep.
`esp_zb_sleep_now ` also utilizes light sleep. Its primary function is to respond when the Zigbee stack indicates that the system can enter sleep mode, along with the specified sleep duration. esp_zb_sleep_now then turns off the RF (Radio Frequency) and invokes vTaskDelay to put the system to sleep. After the designated sleep duration elapses, it restores the RF state.

diazmanuel, 2023-09-27:
Thanks for the answer, could you add to the documentation the characteristics of the set tx power function that you mentioned, I have noticed that if you use a value outside of the 3db steps this function ignores them, on the other hand I would like you to explain a little more about from the portSUPPRESS_TICKS_AND_SLEEP macro

This macro is implemented internally in the SDK, or is it something that should be implemented manually to improve chip consumption?. If it is the latter case, could you explain to me how it is used and give me an example? 

thank you

kelin6, 2023-10-07:
@diazmanuel  

- When FreeRTOS enables Tickless mode, if the idle task is the only runnable task (all other tasks are in a blocked or suspended state), and the system has been in low power mode for a duration greater than `configEXPECTED_IDLE_TIME_BEFORE_SLEEP` clock ticks, the FreeRTOS kernel will invoke the macro `portSUPPRESS_TICKS_AND_SLEEP` to handle low-power-related tasks.

- `portSUPPRESS_TICKS_AND_SLEEP`  has already been implemented in esp-idf, no additional implementation is needed.

xieqinan, 2023-10-11:
@diazmanuel ,

The usage of the `esp_zb_set_tx_power` API is explained in the [esp-zigbee-lib v1.0.0 documentation](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L408), and any errors setting for tx_power will be logged in the error messages. 

If this problem has been resolved, please consider closing the issue. Feel free to discuss any issues related to ESP-Zigbee-SDK here.

closed 2023-10-12 by chshu