espressif/esp-zigbee-sdk #108: ESP32-H2 reading or writing consumption is too high

diazmanuel, 2023-09-22:
Hello,

I am analyzing the consumption of the ESP32H2 when reading from an external device and I encounter the following consumption problem as seen in the image

[image]

So in this image you can see two problems: on the one hand, the type of replica that is observed and on the other hand, the excessive consumption that is maintained for a long time in 20ma. If we zoom in we see the following

[image]

In the zoom, 3 peaks are observed and then an almost constant consumption of approximately 20ma, but the response to the reading ends with those first 3 peaks. That is to say, the response to the data query is carried out with those 3 peaks and for some reason after that it remains at that high consumption for a long time.

Also perform a reading of several attributes with the same command, resulting in the following image.

[image]

In this image you can clearly see the 2 queries and you can also see the replica, but this replica is not a replica as was believed, because in this supposed replica you do not see the 2 data consumptions as in the other case.

kelin6, 2023-09-26:
@diazmanuel  May I inquire about the environment in which you are running the Zigbee sleep example, such as the ESP-IDF commit and ESP-Zigbee-SDK version?

We are currently making modifications to the light sleep code internally, based on our latest internal code, our test results are as follows:
[image]
[image]

- Regarding what you mentioned about the type of replica, it pertains to sending a read attribute command, receiving a response, and sending APS ack/ack packets, which align with a transmit current of 110mA. 

- The excessive power consumption issue within 20mA for long-term maintenance has now been resolved. Next, we will update ESP-IDF and ESP-Zigbee-SDK.

diazmanuel, 2023-09-27:
Hello!
Thanks for the response, I'm glad to know the problem was found.

I leave you the data of the environment
ESP-IDF: release v5.1
ESP-ZIGBEE-SDK: 0.9.5

May I know when they estimate that these updates will be uploaded to solve the problem and be able to continue testing?

kelin6, 2023-10-07:
@diazmanuel  The fixed code has already been internally reviewed and merged. It should be visible on GitHub in about 1 to 2 weeks.

kelin6, 2023-10-11:
@diazmanuel  
Please update the ESP-IDF and ESP-Zigbee-SDK using the following environment:

- ESP-IDF branch: release/v5.1 (commit:  [6b1f40b](https://github.com/espressif/esp-idf/commits/release/v5.1))
- ESP-Zigbee-SDK branch: main(commit: 41c7f14488d7d00cd975b6c2a74128d0840a0193)

Hope this helps you.

chshu, 2023-10-12:
@diazmanuel Please reopen if any follow up questions.

closed 2023-10-12 by chshu