espressif/esp-zigbee-sdk #109: ESP32-C6 unable to set model and manufacterer

baylf2000, 2023-09-23:
Espressif-IDE, SDK 5.1.1, esp-zboss 0.7.2 and esp-zigbee 0.9.5.

Using the "/esp_zigbee_customized_devices/customized_client/" example, modified to run as an end device, I'm unable to modify the code to allow zigbee2mqtt to receive the manufacturer name and model id when joining an existing network. I've tried many other examples and different versions of the idf and dependencies. Below is the relevant modifications I've made to the example.

```
char model_id[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufacturer_name[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

/* basic cluster - fully customized */
esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &model_id[0]);
esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufacturer_name[0]);
esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
````

Posts I've found suggest others have had success with this code, however I cannot get it to work. Has something changed in recent versions regarding how this code should work?

Any help would be appreciated.

baylf2000, 2023-09-23:
This is the log output of the device when joining the existing zigbee2mqtt network.
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd10
load:0x4086e610,len:0x2dc4
load:0x40875728,len:0x17d8
entry 0x4086c410
I (23) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (24) boot: compile time Sep 23 2023 17:51:53
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00165000
I (80) boot:  3 zb_storage       Unknown data     01 81 00175000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 00179000 00004000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=3a120h (237856) map
I (156) esp_image: segment 1: paddr=0004a148 vaddr=40800000 size=05ed0h ( 24272) load
I (162) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=ae540h (714048) map
I (308) esp_image: segment 3: paddr=000fe568 vaddr=40805ed0 size=080ech ( 33004) load
I (316) esp_image: segment 4: paddr=0010665c vaddr=4080dfc0 size=01930h (  6448) load
I (322) boot: Loaded app from partition at offset 0x10000
I (322) boot: Disabling RNG early entropy source...
I (337) cpu_start: Unicore app
I (337) cpu_start: Pro cpu up.
W (347) clk: esp_perip_clk_init() has not been implemented yet
I (354) cpu_start: Pro cpu start user code
I (354) cpu_start: cpu freq: 160000000 Hz
I (354) cpu_start: Application information:
I (357) cpu_start: Project name:     customized_light
I (363) cpu_start: App version:      1
I (367) cpu_start: Compile time:     Sep 23 2023 17:51:43
I (373) cpu_start: ELF file SHA256:  35cc23bb8a4a7abb...
I (379) cpu_start: ESP-IDF:          v5.1.1-dirty
I (384) cpu_start: Min chip rev:     v0.0
I (389) cpu_start: Max chip rev:     v0.99
I (394) cpu_start: Chip rev:         v0.0
I (399) heap_init: Initializing. RAM available for dynamic allocation:
I (406) heap_init: At 40816410 len 00066200 (408 KiB): D/IRAM
I (412) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (419) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (426) spi_flash: detected chip: generic
I (430) spi_flash: flash io: dio
I (434) sleep: Configure to isolate all GPIO pins in sleep state
I (441) sleep: Enable automatic switching of GPIO sleep configuration
I (448) coexist: coex firmware version: 80b0d89
I (453) coexist: coexist rom version 5b8dcfa
I (458) app_start: Starting scheduler on CPU0
I (463) main_task: Started on CPU0
I (463) main_task: Calling app_main()
I (473) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (473) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (523) phy: libbtbb version: b684fcb, May 17 2023, 20:14:35
I (523) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:197   FFD build
I (523) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:201   sizes: g_zb 17304 sched 1312 bpool 9856 nwk 1908 a                                                                                            ps 1408 addr 660 zdo 720
I (543) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:210   sec 2
I (553) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:212   zcl 600
I (553) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:215   zll 376
I (563) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:221   nvram 140
I (573) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:224   buttons 140
[... 3 lines cut ...]
I (613) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:245   ZB_IOBUF_POOL_SIZE 60 ZB_NWK_IN_Q_SIZE 30 ZB_MAC_P                                                                                            ENDING_QUEUE_SIZE 15 ZB_APS_DST_BINDING_TABLE_SIZE 32 ZB_APS_BIND_TRANS_TABLE_SIZE 16 ZB_N_APS_RETRANS_ENTRIES 20
I (633) ZBOSS: esp-zboss/components/zboss_stack/zboss/common/zb_init_default.c:275   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 32 ZB_IEEE_ADDR_TAB                                                                                            LE_SIZE 32 ZB_NEIGHBOR_TABLE_SIZE 32 ZB_NWK_ROUTING_TABLE_SIZE 5 ZB_APS_DUPS_TABLE_SIZE 32
I (663) ZBOSS: esp-zboss/components/zboss_stack/zboss/zdo/af_descriptor.c:357   ep_count 1
I (673) ZBOSS: esp-zboss/components/zboss_stack/zboss/zdo/zdo_app_prod_conf.c:144   no production config block found
I (683) ZBOSS: esp-zboss/components/zboss_stack/zboss/zdo/zdo_app_prod_conf.c:781   no valid production configuration found, signaling                                                                                             to application
I (693) ESP_HA_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (703) ESP_HA_ON_OFF_LIGHT: Zigbee stack initialized
I (703) ZBOSS: esp-zboss/components/zboss_stack/zboss/commissioning/bdb/zdo_commissioning_bdb.c:348   dev type 2, joined 0, ext_pan_id                                                                             [... cut ...]
I (1933) ZBOSS: esp-zboss/components/zboss_stack/zboss/nwk/nwk_main.c:3688   Put 16 into input queue (blocked by 15)
I (1933) ZBOSS: esp-zboss/components/zboss_stack/zboss/nwk/nwk_main.c:2301   set input_blocked_by and process 16 from in queue
I (2023) ZBOSS: esp-zboss/components/zboss_stack/zboss/commissioning/bdb/zdo_commissioning_bdb.c:548   COMMISSIONING_STOP: app signal 1                                                                                            0 comm status 0
I (2023) ESP_HA_ON_OFF_LIGHT: Network steering started
```

baylf2000, 2023-09-23:
I have tried modifying `esp_zigbee_type.c` as follows:

```
#define ESP_ZB_ZCL_BASIC_MANUFACTURER_NAME_DEFAULT_VALUE {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'}
```

And this still does not work. This is the response I get from zigbee2mqtt when attempting to read that attribute:
```
error 2023-09-23 21:32:06Publish 'set' 'read' to '0x6055f90000f6ecd8' failed: 'Error: Read 0x6055f90000f6ecd8/10 genBasic(["manufacturerName"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

nalexxx, 2023-09-23:
This works for me:

```
esp_zb_basic_cluster_cfg_t basic_cfg;
basic_cfg.zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
basic_cfg.power_source = 4; // DC power source
esp_zb_attribute_list_t* basic_cluster = esp_zb_basic_cluster_create(&basic_cfg);
esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, manufacturer_name));
esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, model_id);
```

baylf2000, 2023-09-24:
> This works for me:
>
> ```
> esp_zb_basic_cluster_cfg_t basic_cfg;
> basic_cfg.zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
> basic_cfg.power_source = 4; // DC power source [...]

Thanks for the info. Unfortunately the issue seems to lie elsewhere. However just out of interest, could you please show your declarations for manufacturer_name and model_id?

xieqinan, 2023-09-25:
@baylf2000 ,

Please terminate the ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID string with `\0`.

baylf2000, 2023-09-26:
@xieqinan 

I have managed to determine that there is something about your examples that assume the coordinator role that prevents those manufacturer name and model id attributes from working correctly, even after the role is changed to router or end device.

xieqinan, 2023-09-26:
@baylf2000 

I'm having trouble understanding the information provided above. Could you please provide a detailed description of the Zigbee network topology and device information in your test environment?

chshu, 2023-10-12:
@baylf2000 Do you still have problem with this issue, could you please close the issue if it's done?

chshu, 2023-11-22:
> This works for me:
>
> ```
> esp_zb_basic_cluster_cfg_t basic_cfg;
> basic_cfg.zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
> basic_cfg.power_source = 4; // DC power source [...]

@baylf2000 Let us know if you still have the issue with this code.

closed 2023-11-22 by chshu