espressif/esp-zigbee-sdk #115: Coordinator device won't connect to new end devices after some time

marcoacolla, 2023-09-28:
I'm using an end device to send a custom String to a coordinator device. Their communication works properly most of the time. Still, I noticed that if I leave only a factory reset coordinator on for around 10 minutes, no new end devices can connect to the Zigbee network, although those that were added before the 10 minutes are still able to rejoin.

I have this simple reconnection logic on my end device that tries to reconnect to a network every 10s:
void* zb_network_check(void* arg)
```
{
    while(1){
        if(!has_zb_network){
            ESP_LOGE(TAG, "End Device lost connection to Coordinator");
            esp_zb_nvram_erase_at_start(true);                                         
            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING); 
        }
        sleep(ZB_NETWORK_RECONECTION_WAIT);
    }
    
}
```
This is the log messages I get on my coordinator on the first 10 minutes:

```
I (385) app_start: Starting scheduler on CPU0
I (389) main_task: Started on CPU0
I (389) main_task: Calling app_main()
I (409) phy: phy_version: 101,0, 76d28a9, Mar 30 2023, 17:12:12
I (419) main_task: Returned from app_main()
I (429) COORDINATOR: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (429) COORDINATOR: Zigbee stack initialized
I (429) COORDINATOR: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (439) COORDINATOR: Start network formation
I (439) COORDINATOR: Formed network successfully (Extended PAN ID: ff:fe:3d:6f:f7:f9:55:60, PAN ID: 0x742c, Channel:13)
I (909) COORDINATOR: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (909) COORDINATOR: Network steering started
I (183389) COORDINATOR: ZDO signal: NWK Permit Join (0x36), status: ESP_OK

```
This is how it looks like after i try to connect a new end device:
```
I (909) COORDINATOR: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (909) COORDINATOR: Network steering started
I (183389) COORDINATOR: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (1011109) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1020959) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1030959) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1040949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1050949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1060949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1070949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1080969) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1090949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1100949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1110949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1120949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1130949) COORDINATOR: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
```
These NLME Status Indication messages happen every 10s as my end device keeps trying to reconnect.

kelin6, 2023-10-08:
@marcoacolla  After the coordinator forms the network, it is set to open the network by default and allows other devices to join for 180 seconds. So, when you try to join the network after 10 minutes, the coordinator's network is already closed, and you cannot join.

If you want to allow other devices to join the coordinator even after 180 seconds, you can use the following code in the coordinator to reopen the network and permit other devices to join.

```
 esp_zb_zdo_permit_joining_req_param_t cmd_req;
 cmd_req.dst_nwk_addr = 0x0000;
 cmd_req.permit_duration = 180;
 cmd_req.tc_significance = 1;
 esp_zb_zdo_permit_joining_req(&cmd_req, NULL, NULL);
```

xieqinan, 2023-10-10:
@marcoacolla ,
Please use the provided code after the network has been formed by the coordinator.

xieqinan, 2023-10-24:
@marcoacolla ,

I believe the following code will be helpful for your question. Please add it to the coordinator device to enable `permit_join` persistently. Please note that it should be added after the network has been formed.

```
 esp_zb_zdo_permit_joining_req_param_t cmd_req;
 cmd_req.dst_nwk_addr = 0x0000;
 cmd_req.permit_duration = 0xff;
 cmd_req.tc_significance = 1;
 esp_zb_zdo_permit_joining_req(&cmd_req, NULL, NULL);
 
```

Furthermore, if the issue has already been resolved, please consider closing it. Thank you.

marcoacolla, 2023-10-24:
Hey, I'm still having this problem I mentioned earlier.
> but I keep getting these two warnings whenever I call this method. What does this message mean?
The code works fine, I'm calling it only after the network has been formed as you suggested and now I can connect new devices at any time, but I still can't figure out what this message warns me about. What is this `tsn of callback`?

xieqinan, 2023-10-25:
Hello @marcoacolla 

> What is this `tsn of callback`?

The warnings occur when the ZDO (Zigbee Device Object) command does not receive a response within a specific timeframe. This could be due to a command error or the absence of the destination device.

For instance:
```
I (46330) ESP_ZIGBEE_API_ZDO: ZDO match descriptor request timeout! NO DEVICE FOUND!
W (46330) ESP_ZIGBEE_API_ZDO: The TSN of the callback is not found.
W (46330) ESP_ZIGBEE_API_ZDO: The TSN of the callback is not found and cannot be removed.
``` 

This log indicates a timeout in the match descriptor request. Please check your ZDO commands prefixed with `esp-zb-zdo-` to ensure their correctness.

marcoacolla, 2023-10-25:
> This could be due to a command error or the absence of the destination device.

The code works properly, so I assume there is no command error. 
My coordinator device doesn't have a specific destination device, and the only `esp_zb_zdo_ ` command I'm using is the `esp_zb_zdo_permit_joining_req` command, I assume the warnings are related to it.

Here's my coordinator code:
[coordinator.txt](https://github.com/espressif/esp-zigbee-sdk/files/13168740/coordinator.txt)

So, could you confirm if this behavior is expected in this case, considering the absence of a destination device for the coordinator?

xieqinan, 2023-10-26:
@marcoacolla,

Please refer to [coordinator.txt](https://github.com/espressif/esp-zigbee-sdk/files/13172346/coordinator.txt)

xieqinan, 2023-11-10:
@marcoacolla ,

If the issue has already been resolved, please consider closing it.

closed 2023-11-11 by marcoacolla