espressif/esp-zigbee-sdk #117: Device crashing after changing to C++

baylf2000, 2023-10-01:
I've tried to convert the custom_server project to C++ and the device is now crashing on startup as it runs the `esp_zb_init` function. It produces the following debug output.

Any help would be appreciated.

```
NVS FLASH INITIALISATION: OK
PLATFORM CONFIG: OK
esp_zb_task STARTED

assert failed: zb_osif_nvram_init zb_esp_nvram.c:70 (zb_partition != NULL)
Core  0 register dump:
MEPC    : 0x408011e6  RA      : 0x40808396  SP      : 0x4081eec0  GP      : 0x4080efc0
TP      : 0x407efc00  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000007d  S1      : 0x00000001  A0      : 0x4081eefc  A1      : 0x4080f165
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40817000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4081f01a
S4      : 0x4080f164  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000  

Stack memory:
4081eec0: 0x00000000 0x4087d1a4 0x420bac7c 0x4080d7a2 0x4080f77c 0x420bac7c 0x4080fa64 0x420bac40
4081eee0: 0x4080f78c 0x4081eef4 0x4080f790 0x420babf0 0x4080f164 0x00003037 0x4081806c 0x65737361
4081ef00: 0x66207472 0x656c6961 0x7a203a64 0x736f5f62 0x6e5f6669 0x6d617276 0x696e695f 0x627a2074
4081ef20: 0x7073655f 0x72766e5f 0x632e6d61 0x2030373a 0x5f627a28 0x74726170 0x6f697469 0x3d21206e
4081ef40: 0x4c554e20 0x0000294c 0x4087cda8 0x40808a20 0x00000000 0x420be000 0x40810d88 0x4080c51a
4081ef60: 0x00000000 0x00000000 0x00000000 0x40022498 0x40880000 0x4087ce74 0x4087c610 0x400225ca
4081ef80: 0x420babe4 0x00000000 0x4087ce74 0x40801b04 0x00000001 0x00000014 0x4087ce74 0x4080d7c8
4081efa0: 0x420babe4 0x00000000 0x4087ce74 0x4207e98a 0x00000000 0x00000004 0xffffffff 0x4207ea44
4081efc0: 0x420babe4 0x00000000 0x00000000 0x42000000 0x00000000 0x420be000 0x00000000 0x00000000
4081efe0: 0x00000000 0x420be000 0x40810d88 0x4207575e 0x407efc00 0x00000000 0x4081f034 0x4201a7a2
4081f000: 0x00000000 0x00000000 0x4081f034 0x4200ccee 0x00000000 0x00000000 0x00000000 0x00000000
4081f020: 0x00000000 0x4200040c 0x00000000 0x4200042e 0x00000000 0x00000001 0x00000000 0x00000000
4081f040: 0x00000000 0x00000000 0x00000000 0x4080a72e 0x00000000 0x00000000 0x00000000 0x00000000
4081f060: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081f080: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x0005d57d 0x40817704 0x40817704 0xdfc216a5 0x99ab1ef5
4081f0a0: 0xa7263a3b 0x043d725e 0x9701e6f8 0x3f987f10 0x25314f67 0xdd12bced 0x428bb48e 0xbc70209c
4081f0c0: 0x6e330cdb 0x8c6aeb46 0x152e5ef1 0x5ce377ca 0xf8c43be2 0x1c8152bb 0xe1a5b086 0x87fd1f44
4081f0e0: 0xa3fb4aa8 0x194cde7d 0x1f043c1c 0xe6bcf02b 0xee071057 0xe1208218 0x1135d5e4 0xa65c5b03
4081f100: 0x1842b8af 0x3fbf3cda 0xbdd042b2 0xc1b55417 0x22e6373d 0x2ba185d0 0x2c70ee58 0x35a0fa8d
4081f120: 0x44db9ead 0xe626f897 0xe148451b 0xc7f7a1a2 0x8720c23e 0x8f5a3ca1 0xfaf28f68 0x0227da08
4081f140: 0x7c841f09 0xeba483c3 0x18757555 0xefd1e9f6 0x777a67fa 0xe1906c3e 0x3abec564 0xc2298705
4081f160: 0x215913ae 0x479fdbb8 0x302d95ba 0xfdc08c54 0x7b65d73b 0x0475d209 0xdd14b359 0xc6bc2a66
4081f180: 0xe080a081 0x62dba128 0xb111b18d 0x26bb9b55 0xc035a904 0x90dd8b3b 0x06f25525 0x36d634bc
4081f1a0: 0x30d142f0 0x0faebad6 0x08f9f625 0x3b65a5ee 0xb42739b4 0x5aa5e160 0x26f4ca47 0xe0497849
4081f1c0: 0x3eb57429 0xdcba50ae 0x829c5504 0x6e0396ba 0xe65028d4 0xb18748be 0xf7155939 0xadd20d0a
4081f1e0: 0x210e74a5 0xe45aff0a 0x97e23fca 0xbb421a96 0x132ef4d8 0x8037aa23 0x52fe5991 0xac3ed1db
4081f200: 0x61d99563 0xd1b75fd9 0x86ce30f0 0x8ccf7e5d 0xd317983f 0x57376a47 0xad6e90bc 0x93122f03
4081f220: 0x3bb132ee 0x2656e417 0xca8c7f74 0x4c91ee22 0xbcca5933 0xcca0a6a4 0xe130b1eb 0xf056f94e
4081f240: 0x64967283 0xb7dbdbf7 0x522e37a2 0x0d9c4eab 0x6a2ca4e0 0xaaa24d25 0xbd88a56f 0xde20a558
4081f260: 0x15d90939 0x7a9fa7c5 0x87eb16d9 0xca668f2e 0x122ef7bf 0xd6f41466 0x9025a7ff 0xbdbc3368
4081f280: 0xd0d78e5a 0xfa06b0c9 0x963c3102 0x6c1a5e96 0x20dc0a97 0x3aa2112f 0x60e758e0 0x94000880
4081f2a0: 0x8041d9bd 0x42cf69d6 0x4d639675 0x985c2cac 0xc0b08a83 0x522a86a1 0x8b3ecf1a 0xf7258c28

ELF file SHA256: eb78889b95d7e01f

Rebooting in 10 seconds...```

kelin6, 2023-10-08:
@baylf2000 Based on the log, it seems that the Zigbee stack hasn't been allocated a partition in the example. Please check `partitions.csv` to see if `zb_storage` is allocated. You can refer to [zb_storage](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light/partitions.csv#L6).

xieqinan, 2023-10-24:
Hello @baylf2000 ,

Is the crash still occurring after creating the `zb_storage` partition? If not, please consider closing this issue.

Feel free to open new issues to discuss any other questions related to the esp-zigbee-sdk.

chshu, 2023-11-22:
@baylf2000 Please double check the partition table in your project, let us know if any further issues.

closed 2023-11-22 by chshu