espressif/esp-zigbee-sdk #118: Support for Multistate Value Cluster(0x0014) Zigbee

Bosemani, 2023-10-04:
Hai,
   I read this issue [link](https://github.com/espressif/esp-zigbee-sdk/issues/10) .
I was using espressif/esp-zigbee-lib: "~0.9.5". I think still there is no support for multistate value cluster.
We need this cluster, How can we use? Is there any alternate solution is available to use?

xieqinan, 2023-10-11:
The feature has been supported in [ZCL Multistate](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_multistate_value.h)， please try to test it.

xieqinan, 2023-10-11:
@Bosemani 
If the feature works fine, please close this issue.

Bosemani, 2023-10-11:
Hai @xieqinan 
Thank you for support.
It working well! Closing this issue.

closed 2023-10-11 by Bosemani