espressif/esp-zigbee-sdk #119: ESC32-C6-Devkit-C-1 don't run esp-zigbee-sdk examples esp_zigbee_HA_sample

RadekDymak, 2023-10-05:
Hello I'm quite new, to ESP32 product therefore I apologize myself for any eventual basic mistakes

I try to build and run examples of HA_on_off_light and HA_on_off_switch. Version of sdk is 0.9.5. I'm using Espressif IDE.
I copy this HA example and common folders to esp-idf-v5.1.1 example folder. Then I click File/new/Espressif IDF Project in ESP IDE . In wizard i select Create project using one of templates and then I select this HA templates from SDK.

Then i edit path to EXTRA_COMPONENT_DIRS in Cmakelists.txt.

Target was set to esp32c6 in ESP IDE. Then I build both HA examples without any error. Then I flash them to the dev kits.

First time i powered up HA_switch  (coordinator) wait 10s and then power up HA_light (End device), wait another 10s and try to press BOOT button on HA_Switch device, but nothing happens.

When I try HA switch/light example which is part of esp-idf everything works fine.

Please can you advice me what is wrong?

mallaprashant, 2023-10-06:
If you have copied the HA switch and light examples, have you tried running 'idf.py menuconfig' before building the project and selecting a custom partition table from the partition table options? Additionally, make sure to choose 'Coordinator' and 'End Device' for the respective switch and light configurations under 'Component Config -> ZigBee.' Don't forget to erase the flash using 'idf.py -p PORT erase-flash' before flashing it into the device. Hope this helps!"

RadekDymak, 2023-10-06:
> If you have copied the HA switch and light examples, have you tried running 'idf.py menuconfig' before building the project and selecting a custom partition table from the partition table options? [...]

Thank you a lot. settings in menuconfig was right, but I forget to erase flash by idf.py -p PORT erase-flash before I started to flash this example. Please can you tell me, why it is too much important to erase it before new flash? Is it because of changed to custom partition table?

closed 2023-10-06 by RadekDymak