espressif/esp-zigbee-sdk #121: should it actual be possible to join a network formed by an esp32-coordinator with a consumer device?

mw75, 2023-10-06:
I have two esp32-c6 boards and tried different scenarios. The examples/esp_zigbee_HA_sample/HA_on_off_light scenario works and that proofs my boards working.

I tried esp_zb_switch, esp_zigbee_cli as well as esp_zigbee_gateway/esp_zigbee_rcp to let https://www.zigbee2mqtt.io/devices/E1524_E1810.html and/or https://www.zigbee2mqtt.io/devices/ZTH02.html join the successfully formed network, but they do not join, or the leave the network after seconds and never send events.

I'm not an expert regarding zigbee, but i'm quite familiar with zigbee-herdsman and already got some unsupported devices integrated. Do i have a general misunderstanding? 

Would it be possible to publish an example that uses esp_zigbee_cli that just dumps raw interview and event data to the console? That would be much appreciated!

Bosemani, 2023-10-07:
Hai @mw75 
  I integrated [https://www.zigbee2mqtt.io/devices/E1524_E1810.html ](https://www.zigbee2mqtt.io/devices/E1524_E1810.html ) Ikia Button. But Not with the esp, I used Nordic chip.
  But One thing I Found in many zigbee devices, We need to bind the clusters to get data.
Your Ikia button have ON/OFF , Level control. You need to bind on/off of button with on/off of coordinator. (Server & Client)
You need to do immediately, once device got joined. Other vice Device will leave or It will go to sleep mode.

xieqinan, 2023-11-22:
Hello @mw75 ,

Have you found a resolution for your issue? If not, please feel free to discuss your problem further. Otherwise, consider closing this issue.

closed 2024-03-08 by chshu