espressif/esp-zigbee-sdk #122: End Device Join crashes Zigbee stack using Gateway Example

hobieboy, 2023-10-06:
Hello, I am trying to use the SDK to build a coordinator that can register end devices. Using the provided Zigbee Gateway example (with one ESP32-S2 as Gateway and one ESP32-C6 as RCP), whenever I try to join a Zigbee end device (an off-the-shelf Aqara Contact Sensor), I see 2 problems: 1) The device joined but would auto-leave after ~5 seconds, 2) The leave indication crashes the Zigbee stack as shown in the attached log file.

I have 2 questions:
1) What more can be enabled to know why the device leave the network? I am quite certain it is the Coordinator that initiated it since the device does not exhibit this with any other gateways/coordinators.

2) There is no mentioning in documentation nor relevant API I notice that indicates additional initialization needed to accept new device join. Am I missing something?

thanks!
[ESP_Zigbee_Join_Crash.pdf](https://github.com/espressif/esp-zigbee-sdk/files/12832489/ESP_Zigbee_Join_Crash.pdf)

xieqinan, 2023-10-10:
Hello @hobieboy ,

This issue is similar to the [device leave issue](https://github.com/espressif/esp-zigbee-sdk/issues/21). Could you please try using the method provided in the above issue to solve your problem?

Furthermore, if you can provide the Wireshark packet or a simple code snippet related to your problem, the issue can be resolved more quickly.

hobieboy, 2023-10-10:
Thank you for the quick response @xieqinan. Your suggestion above did solve the device leave problem so the main problem is solved now.

However, I would still like to know why would the device leave caused the stack to crash? Is there something I should enable so I can prevent future crash?

xieqinan, 2023-10-11:
@hobieboy ,

Since I am not fully aware of your experience saturation, I cannot pinpoint the specific reason for the crashes. However, based on the crash log, I suspect that the end device attempts to join the network created by the coordinator only once, increasing the children_num to 1. However, the coordinator receives leave indications twice, reducing children_num to -1. This asymmetric operation might be triggering the crash.

xieqinan, 2023-10-24:
Hello @hobieboy ,

If the issue has already been resolved, please close it. Please feel free to discuss any questions of esp-zigbee-sdk here.

hobieboy, 2023-10-24:
Thank you for the help @xieqinan  Since I can keep the device connected and I cannot find a reliable way to recreate the crash, I'll close this issue.

closed 2023-10-24 by hobieboy