espressif/esp-zigbee-sdk #123: Door/window switch implementation and example

Sandor-Hadas, 2023-10-12:
### Is your feature request related to a problem?

Yes. Altering example code does not make door switches working - basically network connection builds up but door open/close is not recognized. Adding dim lights is good but probably more devs wanna integrate doors/windows than fancy lights.

### Describe the solution you'd like.

Examples / door lock controller
Examples / door switch end device

### Describe alternatives you've considered.

TI CC1352P/CC2562P

xieqinan, 2024-04-02:
@Sandor-Hadas,

We have provided examples for temperature sensors and thermostats, which are similar to your request for a door and switch device. You can refer to [them](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor) to implement your device. If you encounter any obstacles, feel free to ask for assistance.

chshu, 2024-09-29:
We have provided a console based all device types app: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_all_device_types_app, you can test any supported device types with it.

Closing the issue, feel free to reopen if any further questions.

closed 2024-09-29 by chshu