espressif/esp-zigbee-sdk #124: ZBOSS NCP host

kklem0, 2023-10-13:
### Is your feature request related to a problem?

The current release only has ZBOSS RCP, but our old implementation uses ZBOSS NCP host.

### Describe the solution you'd like.

The ZBOSS libraries for the NCP host include the following stack layers:

    Zigbee cluster library (ZCL)
    Commissioning logic (part of ZDO)
    ZBOSS NCP serialization protocol
    ZBOSS platform support layer for Linux-based OS

Since you use ZBOSS already if it is possible that we can use our old implementation with ZBOSS NCP host or if we need to rewrite it against esp-zigbee-sdk.

Hedda, 2023-11-04:
+1 yes please consider adding support for ZBOSS NCP host for Zigbee NCP (Network Co-Processor) mode?

Currently is seems like the ESP ZBOSS can only be used Zigbee via a RCP (Radio Co-Processor) mode and no support for ZBOSS NCP host?

See example:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/zboss/3.11.2.0/zboss_ncp_host_intro.html

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/zboss/3.11.2.0/zboss_ncp_host.html

ZBOSS NCP host as Zigbee NCP mode is required to implement support in the zigpy-zboss radio library for zigpy as a Zigbee Coordinator adapter interface which is used by the ”ZHA” Zigbee gateway integration that is built-into Home Assistant:

https://github.com/kardia-as/zigpy-zboss/issues/19

https://github.com/kardia-as/zigpy-zboss

https://github.com/zigpy/zigpy

https://www.home-assistant.io/integrations/zha

Hedda, 2023-11-17:
ZBOSS NCP support would probably also be prefered for [Zigbee2Tasmota (Z2T)](https://tasmota.github.io/docs/Zigbee/) compatible in Tasmota:

https://github.com/arendst/Tasmota/discussions/20045

For reference, Tasmotas's Zigbee2Tasmota (Z2T) already supports Texas Instruments ZNP and Silicon Labs EZSP Zigbee stacks/interfaces with respective Zigbee Coordinator adapter running Zigbee firmware in NCP (Network Co-Processor) mode:

https://tasmota.github.io/docs/Zigbee/

chshu, 2023-11-30:
@kklem0 @Hedda We are working on a ESP Zigbee NCP solution, which will be based on the [ESP Zigbee APIs](https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib/include), and provide the serial interface to communicate with host.

We will also prepare a corresponding zigpy python library to work with Home Assistant.

The draft version will be ready in 2024 January.

kklem0, 2023-11-30:
🎉

Hedda, 2023-11-30:
> We are working on a ESP Zigbee NCP solution, which will be based on the [ESP Zigbee APIs](https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib/include), and provide the serial interface to communicate with host.

@chshu Nice! Will that API be compatible with the standard ZBOSS NCP serial protocol that and CLI the zigpy-zboss library already supports?

https://github.com/kardia-as/zigpy-zboss

Please see related feature request discussion here -> https://github.com/kardia-as/zigpy-zboss/issues/19

Hedda, 2023-11-30:
@chshu also see related upstream discussion in the zigpy repository here -> https://github.com/zigpy/zigpy/issues/1052

chshu, 2023-12-01:
> Will that API be compatible with the standard ZBOSS NCP serial protocol that and CLI the zigpy-zboss library already supports?

@Hedda  The ESP Zigbee NCP will not be compatible with ZBOSS NCP.

With ZBOSS NCP, users need to run ZCL and part of ZDO on host. We want to integrate ZCL and ZDO in the ESP Zigbee NCP as well, so host can handle the application logic only. If some hosts already have ZCL implementation, it's also flexible to access the NCP network layer directly.

Hedda, 2023-12-01:
> The ESP Zigbee NCP will not be compatible with ZBOSS NCP.
>
> With ZBOSS NCP, users need to run ZCL and part of ZDO on host. We want to integrate ZCL and ZDO in the ESP Zigbee NCP as well, so host can [...]

Ok, I can see that might be wanted for complete onboard solutions like Zigbee for Tasmota, though popular open-source Zigbee frameworks like zigpy (for Home Assistant’s ZHA + Zigbee Plugin for Domoticz) and zigbee-herdsman (for Zigbee2MQTT and IoBroker) have their own ZCL and ZDO implementations.

Hedda, 2023-12-31:
> We are working on a ESP Zigbee NCP solution, which will be based on the [ESP Zigbee APIs](https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-lib/include), and provide the [...]

FYI, assume this related pull request for zigpy-cli  -> https://github.com/zigpy/zigpy-cli/pull/42 is indication that making progress on -> https://github.com/zigpy/zigpy/issues/1052

Hedda, 2024-01-22:
FYI, looking at recent commits it looks like @lhespress has begun working on implementing `esp-zigbee-ncp` and API in this SDK:

https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ncp

https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_ncp/README.md

https://github.com/espressif/esp-zigbee-sdk/tree/main/components/esp-zigbee-ncp

https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/application.rst

https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/api-reference/esp_zigbee_ncp.rst