espressif/esp-zigbee-sdk #125: How to add a manufacturer specific attribute to a Standard cluster

jkano, 2023-10-13:
### Question

Hi, 

Im using the SDK to add a standard cluster `Electrical Measurement` which have everything we need but we need a custom attribute (manufacturer specific) for our application. 

My question is how to add a manufacturer specific attribute, if I use the API `esp_zb_electrical_meas_cluster_add_attr()` with the custom attribute id I get an error:

```
E (1028) ESP_ZIGBEE_ATTRIBUTE: incorrect/unsupported attribute_id!
```

And I cant use the API `esp_zb_custom_cluster_add_custom_attr()` because it complains the cluster is not a custom cluster.

So, how can I add a custom attribute (manufacturer specific) to a standard cluster like the electrical measurement?

xieqinan, 2023-10-13:
Hello @jkano ,

The SDK provides the [get attribute](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_attribute.h#L42C1-L42C18) function `esp_zb_zcl_get_attribute()`, which allows you to retrieve the attribute you have registered. You can utilize this function to modify the manufacturer-specific attribute.

jkano, 2023-10-13:
> Hello @jkano ,
>
> The SDK provides the [get attribute](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_attribute.h#L42C1-L42C18) function [...]

That api is to get the attribute value when its registered, but how do I register a manufacturer specific attribute to an standard cluster like the electrical measurement? 

I want to add an attribute with id 0x0518, S64 type, read only access and manuf code (our code)

As I said, if I use the function `esp_zb_electrical_meas_cluster_add_attr()` I get an error and if I use the function `esp_zb_custom_cluster_add_custom_attr(esp_zb_attribute_list_t *attr_list, uint16_t attr_id, uint8_t attr_type, uint8_t attr_access, void *value_p); ` I get another error.

xieqinan, 2023-10-13:
@jkano ,

Alright, I understand what you mean.

The SDK does not provide an API to add a manufacturer-specific attribute. However, there is a workaround for this issue.

```
    esp_zb_attribute_list_t *electrical_means_attr_list = esp_zb_electrical_meas_cluster_create(&electrical_cfg);
    esp_zb_attribute_list_t *new_elem = (esp_zb_attribute_list_t *)malloc(sizeof(esp_zb_attribute_list_t)); // Create manufacturer node
    int16_t value = 0;
    new_elem->attribute.id = 0x0518;
    new_elem->attribute.access = ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY;
    new_elem->attribute.type = ESP_ZB_ZCL_ATTR_TYPE_S64;
    new_elem->attribute.manuf_code = 0x1234;
    new_elem->attribute.data_p = malloc(sizeof(int16_t));
    *(int16_t *)new_elem->attribute.data_p = value;
    new_elem->cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
    esp_zb_attribute_list_t *head = electrical_means_attr_list;
    head = head->next; // skip the head
    while (head) {
        if (head->next) {
            head = head->next;
        } else {
            break;
        }
    }
    head->next = new_elem; // add new element to list
```
By the way, the handling of manufacturer-specific attributes is required to be done by the user.

jkano, 2023-10-13:
I found the following comment about how to handle raw commands: #94, but I'm having an error when handling:

I added new attribute using your code:

```c
    // Creeate cluster list
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
   
    // Create basic cluster
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);

    // Create electrical measurement
    esp_zb_attribute_list_t *esp_zb_el_meas_cluster = esp_zb_electrical_meas_cluster_create(&electrical_cfg);
    esp_zb_attribute_list_t *new_elem = (esp_zb_attribute_list_t *)malloc(sizeof(esp_zb_attribute_list_t));
    int16_t value = 0;
    new_elem->attribute.id = 0x0518;
    new_elem->attribute.access = ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY;
    new_elem->attribute.type = ESP_ZB_ZCL_ATTR_TYPE_S64;
    new_elem->attribute.manuf_code = 0x1234;
    new_elem->attribute.data_p = malloc(sizeof(int16_t));
    *(int16_t *)new_elem->attribute.data_p = value;
    new_elem->cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
    new_elem->next = NULL;

    esp_zb_attribute_list_t *head = esp_zb_el_meas_cluster ;
    head = head->next;
    while (head) {
        if (head->next) {
            head = head->next;
        } else {
            break;
        }
    }
    head->next = new_elem; // add new element to list

    // Create cluster lists for this endpoint
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();

    // Add the clusters
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_electrical_meas_cluster(esp_zb_cluster_list, esp_zb_el_meas_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    // Create ep list
    struct esp_zb_ep_list_s* esp_zb_ep_list = esp_zb_ep_list_create();

    // Add cluster list to endpoint list
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_METER_INTERFACE_DEVICE_ID);

    // Register the endpoint and device
   esp_zb_device_register(esp_zb_ep_list);
```

Then, I registered the raw cmd handler callback using: `esp_zb_raw_command_handler_register(zb_raw_cmd_handler)`

And added the function:

```c
static bool zb_raw_cmd_handler(uint8_t bufid)
{
    ESP_LOGI(TAG, "zb_raw_cmd_handler(): %d", bufid);
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);

    uint8_t data[zb_buf_len(bufid)];
    memcpy(data, zb_buf_begin(bufid), sizeof(data));
    ESP_LOGW(TAG, "Got data, size: %d", sizeof(data));

    if ((cmd_info->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT) &&
        (cmd_info->is_common_command) &&
        (cmd_info->cmd_id == ZB_ZCL_CMD_READ_ATTRIB))
    {
        bool handle_read_attr_request = false;
        uint8_t dst_ep = cmd_info->addr_data.common_data.dst_endpoint;

        // Iterate on the number of requested attributes
        int req_len= sizeof(uint16_t);
        for (int i = 0; i < sizeof(data); i += req_len)
        {
            uint16_t attr_id = (data[i] << 8) | data[i+1];
            ESP_LOGI(TAG, "Read attr: 0x%04x", attr_id);

            // If request includes custom attribute, mark handle request
            if (attr_id == 0x0518)
            {
                handle_read_attr_request = true;
            }

            esp_zb_zcl_attr_t* pAttr = esp_zb_zcl_get_attribute(dst_ep, 
                                                                cmd_info->cluster_id,
                                                                ZB_ZCL_CLUSTER_SERVER_ROLE,
                                                                attr_id);
            ESP_LOGI(TAG, "pAttr: %p", pAttr);
            if (pAttr != NULL)
            {
                ESP_LOGI(TAG, "pAttr->Id  : %d", pAttr->id);
                ESP_LOGI(TAG, "pAttr->Data: %p", pAttr->data_p);
                // Add data to response request...
            }
        }

        // If the read attr request needs to be handled by us (not the stack)
        if (handle_read_attr_request)
        {
            // How to send the response?

            // return true;
            //zb_zcl_send_default_handler();
        }
    }

    return false;
}
```

When I send a read attr request for attribute 0x0000 from the coordinator I get:

```
I (9234) RAW_CMD_HANDLER: zb_raw_cmd_handler(): 10
I (9234) RAW_CMD_HANDLER: Got data, size: 2
I (9234) RAW_CMD_HANDLER: Read attr: 0x0000
I (9234) RAW_CMD_HANDLER: pAttr: 0x4087ef32
I (9234) RAW_CMD_HANDLER: pAttr->Id: 0
I (9234) RAW_CMD_HANDLER: pAttr->Data: 0x4087ec90
```

This is the log when I send an attribute request for 0x0518 from the coodinator:

```
I (9594) RAW_CMD_HANDLER: zb_raw_cmd_handler(): 12
I (9594) RAW_CMD_HANDLER: Got data, size: 2
I (9594) RAW_CMD_HANDLER: Read attr: 0x0518
I (9594) RAW_CMD_HANDLER: pAttr: 0x0
```

So, as you see, the `pAttr` variable is NULL, so the function `esp_zb_zcl_attr_t* pAttr = esp_zb_zcl_get_attribute(dst_ep, cmd_info->cluster_id, ZB_ZCL_CLUSTER_SERVER_ROLE, attr_id);` its not finding the attribute in the electrical measurement cluster.

Do you know what can be happening here?

Also, once I get the attr data, how should I send a response?

jkano, 2023-10-17:
Any update with this one? 

Thanks!

jkano, 2023-10-17:
Hi @xieqinan,

Any update with this one?

Thanks!

xieqinan, 2023-10-18:
> ```c
>             esp_zb_zcl_attr_t* pAttr = esp_zb_zcl_get_attribute(dst_ep, 
>                                                                 cmd_info->cluster_id,
>                                                                 ZB_ZCL_CLUSTER_SERVER_ROLE,
>                                                                 attr_id);
> ```
Regarding this code, I suggest using `zb_zcl_get_attr_desc_manuf_a()` instead. This is because the `manufacturer_code` has been specified.

> ```c
>         if (handle_read_attr_request)
>         {
>             // How to send the response?
>             
>             //zb_zcl_send_default_handler();
>             // return true;
>         }
> ```

For this code, it is both correct and necessary. If the command has been handled correctly, it is essential to send a default response as shown below and return `true` to prevent the stack from handling the command again.
```
zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_SUCCESS);
```

jkano, 2023-10-18:
@xieqinan 

Thanks for the reply!

For the second part:

> If the command has been handled correctly, it is essential to send a default response as shown below and return true to prevent the stack from handling the command again

I'm handling the `read attribute request` and returning a `read attribute response` with the value of the read attribute, is it necessary to send a default response too? (so the stack does not handle it)

xieqinan, 2023-10-18:
> specific attributes and returning a `read attribute response` with the value of the read attribute, is it necessary to send a default response too?

If the `read attribute response` has already been sent, the default response is unnecessary. However, you still need to call `zb_buf_free()` to release the `bufid`.
Additionally, it's worth noting that `zb_zcl_send_default_handler()` has a dual effect: it sends the default response and frees the `bufid`.

jkano, 2023-10-18:
Ok thank you! 

Gonna test and close the issue if I dont have anything else to ask.

jkano, 2023-10-18:
Hi @xieqinan 

I have another question, I can't find an API to send a `read attribute response` do you know how can build and send a read attribute response? (noting that it need to be for a manuf. specific attribute)

I found the following one to send a cmd:

```
zb_zcl_send_cmd_tsn(zbuf_id, 
                    cmd_info->addr_data.common_data.source.u.short_addr, 
                    ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
                    cmd_info->addr_data.common_data.src_endpoint,       // dst ep is the cmd source ep
                    ZB_ZCL_FRAME_DIRECTION_TO_SRV,
                    cmd_info->addr_data.common_data.dst_endpoint,       // src ep is the cmd dst ep
                    payload,            // <----- What to put here?
                    payload_size,       // <----- What will be the size?
                    put_payload_cb,     // <----- What is this callback for?
                    cmd_info->cluster_id,
                    ZB_ZCL_DISABLE_DEFAULT_RESPONSE,
                    ZB_ZCL_CMD_READ_ATTRIB_RESP,
                    cmd_info->seq_number,
                    zb_cb);
```

But I dont know what to put on `payload` and `payload_size` (for a read attribute response for a manuf. specific attribute). Also what is the `put_payload_cb` used for?

xieqinan, 2023-11-22:
Hello,

You can attempt to utilize multiple `zb_zcl_read_attr_res_t` structs to fill the `payload`. The `payload_size` should be set equal to the number of `zb_zcl_read_attr_res_t` * `sizeof(zb_zcl_read_attr_res_t)`. The `put_payload_cb` can use `NULL` as a temporary placeholder. I am not sure it whether is valid, but you can try it, first.

chshu, 2024-01-08:
@jkano Closing the issues since the questions are all addressed.

closed 2024-01-08 by chshu