espressif/esp-zigbee-sdk #126: TimeOfDay type attributes are unsupported in custom clusters

juantxorena, 2023-10-14:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### esp-zigbee-lib verion.

1.0.0

### esp-zboss-lib verion.

1.0.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I'm creating a custom cluster with a time of the day type attribute, which is allowed as per the specs, but apparently it isn't supported.

### What is the actual behavior?

It fails with this error
`ESP_ZIGBEE_ATTRIBUTE: The data type of the value is not supported`

### Steps to reproduce.

This is the code

```
#define TIME_CONTROL_CLUSTER 0xFCCC    
const uint16_t attr_id = 0;
const uint8_t attr_type = ESP_ZB_ZCL_ATTR_TYPE_TIME_OF_DAY;
const uint8_t attr_access = ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING;
esp_zb_attribute_list_t *time_control_attr_list = esp_zb_zcl_attr_list_create(TIME_CONTROL_CLUSTER);
esp_zb_custom_cluster_add_custom_attr(time_control_attr_list, attr_id, attr_type, attr_access, &undefined_value_16);
```

xieqinan, 2023-10-16:
@juantxorena ,

The `time of day` type is temporarily unsupported in custom clusters. Could you please replace it with the `string` type before adding it to the esp-zigbee-sdk?
By the way, the first byte of `string` type indicates the length of string.

closed 2023-11-03 by espressif-bot

xieqinan, 2023-11-03:
Hello @juantxorena,

The issue has been fixed in esp-zigbee-sdk [v1.0.2](https://github.com/espressif/esp-zigbee-sdk), please refer to [custom command](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L1981).