espressif/esp-zigbee-sdk #127: esp_zb_zcl_set_attribute_val() should call the attribute change callback?

jkano, 2023-10-16:
### Question

Hi,

I'm testing the Customized Server and Client example, I understand that when both devices are binded, the button on the client works to turn on the light on the server.

I have another application where the server have a button and a relay to work as an outlet, when the button is pressed the relay is actioned, turning ON/OFF the appliance connected to it. Also, user can connect this device to a Zigbee network and control the relay using OnOff commands. 

So I modified the example and added a button using the button library and registered the callback for a single tap:

```
void button_single_click_cb(void *arg,void *usr_data)
{
    relay_state = !relay_state;  // Saving the last state for now, should get the current attr value using esp_zb_zcl_get_attribute()

    ESP_LOGI(TAG, "New relay state: %d", relay_state );
    esp_err_t err = esp_zb_zcl_set_attribute_val(HA_ESP_LIGHT_ENDPOINT,
                                 ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
                                 ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                                 ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
                                 &relay_state , false);  // False because onoff attr is not writable, so force the new value
}
```

The thing is that I expected that if I write an attribute using `esp_zb_zcl_set_attribute_val()` and the value changed, the callback `zb_action_handler()` should trigger  with `callback_id = ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` because the attribute value just changed. 

This way, the function on the example: `zb_attribute_handler()` will be triggered, the message parsed which calls:

```
light_state = message->attribute.data.value ? *(bool *)message->attribute.data.value : light_state;
ESP_LOGI(TAG, "Light sets to %s", light_state ? "On" : "Off");
light_driver_set_power(light_state);
```

I think the stack should not matter if this was changed on purpose from somewhere inside the application or from a radio message.

Am I wrong on this assumption or is there a way to change an attribute value which calls the `zb_action_handler()` callback?

xieqinan, 2023-10-16:
Hello @jkano ,

I believe there are some misunderstandings. The function `esp_zb_zcl_set_attribute_val()` is only meant to set the attribute of a cluster, and it does not trigger the `zb_action_handler()`. The callback ID `ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID` in `zb_action_handler()` will be triggered when the command comes from the remote device and expects to set a different attribute value to cluster.

jkano, 2023-10-16:
I have an update:

I sent the report config message using Zboss API like this:

```c
zb_bufid_t zbuf = zb_buf_get(ZB_FALSE, 0);
zb_uint8_t* cmd_ptr = NULL;
uint8_t rep_change = 1;

uint16_t short_addr = 0xd17f;   // Address of End device, saved when end device announce is received
uint8_t dst_ep = 0x01;

        
ZB_ZCL_GENERAL_INIT_CONFIGURE_REPORTING_SRV_REQ(zbuf, cmd_ptr, ZB_ZCL_ENABLE_DEFAULT_RESPONSE);
ZB_ZCL_GENERAL_ADD_SEND_REPORT_CONFIGURE_REPORTING_REQ(cmd_ptr, ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, 
                                                       ZB_ZCL_ATTR_TYPE_BOOL, 0, 15, &rep_change);
ZB_ZCL_GENERAL_SEND_CONFIGURE_REPORTING_REQ(zbuf, cmd_ptr, short_addr,
                                            ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT, dst_ep,
                                            HA_ONOFF_SWITCH_ENDPOINT, ZB_AF_HA_PROFILE_ID, 
                                            ZB_ZCL_CLUSTER_ID_ON_OFF, NULL);
```

And I get the response, then I send a read report config using Zboss' API like this:

```c
zb_bufid_t zbuf = zb_buf_get(ZB_FALSE, 0);
zb_uint8_t* cmd_ptr = NULL;

uint16_t short_addr = 0xd17f;   // Address of End device
uint8_t dst_ep = 0x01;

ZB_ZCL_GENERAL_INIT_READ_REPORTING_CONFIGURATION_SRV_REQ(zbuf, cmd_ptr, ZB_ZCL_ENABLE_DEFAULT_RESPONSE);
ZB_ZCL_GENERAL_ADD_SEND_READ_REPORTING_CONFIGURATION_REQ(cmd_ptr, ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID);
ZB_ZCL_GENERAL_SEND_READ_REPORTING_CONFIGURATION_REQ(zbuf, cmd_ptr, short_addr,
                                                     ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT, dst_ep,
                                                     HA_ONOFF_SWITCH_ENDPOINT, ZB_AF_HA_PROFILE_ID, 
                                                     ZB_ZCL_CLUSTER_ID_ON_OFF, NULL);
```

I receive both callbacks on `zb_action_handler()` for `ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID` and `ESP_ZB_CORE_CMD_READ_REPORT_CFG_RESP_CB_ID` with the responses info and I printed them.

This is the log of both operations:

```
I (132642) COORD: Received Config Report Response (0x1002)
I (132642) COORD:   status : ESP_OK
I (132642) COORD:   src add: 0xd17f
I (132642) COORD:   cluster: 6
I (132652) COORD:   attr id: 0xffff
I (132662) COORD:   cmd id : 0x07
I (132662) COORD:   cmd dir: 1
I (132732) COORD: Received Read Report Config Response (0x1003)
I (132732) COORD:   status   : ESP_OK
I (132732) COORD:   src add  : 0xd17f
I (132742) COORD:   cluster  : 0x0006
I (132742) COORD:   attr id  : 0xffff
I (132752) COORD:   cmd id   : 0x09
I (132752) COORD:   cmd dir  : 1
I (132762) COORD:     rep dir: 0
I (132762) COORD:     min_int: 0
I (132772) COORD:     max_int: 15
I (132772) COORD:     delta  : 0xf2 0x00
```

According to the read config response, the minimum and max intervals are set correctly (0 and 15) and rep change (delta) is weird (maybe because the attr is a discrete value). But at least the end device should be sending a report to the coordinator every 15 seconds as configured.

Do you know what can I be missing to configure attribute reporting?

Thanks!

jkano, 2023-10-17:
Closing as I already figured it out how reporting mechanism works (thanks to #76) and initial question was answered.

closed 2023-10-17 by jkano