espressif/esp-zigbee-sdk #128: Add more support for Door lock cluster.

mallaprashant, 2023-10-16:
### Is your feature request related to a problem?

"Yes, I am unable to handle door lock/unlock commands on the server device using **esp_zb_core_action_handler_register()**, and it's not straightforward to parse and implement them using **esp_zb_raw_command_handler_register()**.

I've also been wondering how the Lock State attribute receives a message indicating that the physical door lock is opened or closed to change its value. Could you please clarify how the Lock State attribute receives a message that the physical door lock is opened or closed and confirm if my understanding is correct?"

### Describe the solution you'd like.

It would be helpful if you could provide examples for both the door lock client and server devices, as mentioned in issue #123

mallaprashant, 2023-11-08:
Hi @chshu @xieqinan, I have been an active user of the ESP32 for an extended period, and I have recently embarked on my journey into the world of Zigbee. 

I would like to seek clarification on the level of support provided within this context. The reason for my query is the absence of a response to my previous inquiry, which was made in response to a comment indicating that questions related to the SDK could be addressed.

If my approach in raising the question is incorrect, I am open to correction. However, if this is indeed the avenue for seeking support, I kindly request that the support team provide information regarding the status of the issue. This could include whether it is currently under development, an estimated timeframe for resolution, or any relevant details.

Maintaining effective communication and support is crucial in any development environment. Neglecting or not addressing concerns may lead to reconsidering our choice of hardware platforms, favoring alternatives with a more robust support infrastructure. Your prompt response and assistance in this matter would be greatly appreciated.

xieqinan, 2023-11-08:
Hello @mallaprashant ,

Apologies for the delay and keeping you waiting. I have been busy with the some bugs recently and didn't pay attention to your issue. Moving forward, I will continue to focus on resolving the problem with the Door Lock cluster. Thank you for your support.

mallaprashant, 2023-11-10:
@xieqinan, Thanks for the update. I appreciate your efforts in addressing the Door Lock cluster issue. Hope this problem gets resolved soon.

xieqinan, 2023-11-10:
@mallaprashant ,
We have already added the "lock/unlock request and response" for the door lock cluster to the `esp-zigbee-sdk`, and it will be released this week. Please note that currently, we only support the handle of locking and unlocking commands for the door lock cluster. If you require additional support, kindly provide detailed requirements here.

closed 2023-11-10 by espressif-bot

xieqinan, 2023-11-13:
@mallaprashant ,
Please refer to the [door lock callback](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L85).

mallaprashant, 2023-11-20:
Hi @xieqinan,

Firstly, thanks for the recent update. Apologies for the delayed response due to some unexpected circumstances.

I've been experimenting with ESP_ZB_CORE_DOOR_LOCK_LOCK_DOOR_CB_ID and have successfully received door lock and unlock commands. However, I've encountered an issue where (message->info.endpoint = 0) is consistently returned while using ESP_ZB_CORE_DOOR_LOCK_LOCK_DOOR_CB_ID and ESP_ZB_CORE_DOOR_LOCK_LOCK_DOOR_RESP_CB_ID.

Also, I'm curious about how the Lock State attribute receives messages indicating the physical door lock's open or closed status. Can you shed some light on what happens when the door lock server receives door lock or unlock commands?

> I've also been wondering how the Lock State attribute receives a message indicating that the physical door lock is opened or closed to change its value. Could you please clarify how the Lock State [...]