espressif/esp-zigbee-sdk #129: Missing thermostat cluster or temperature sensor example

patryk-belka-wizzdev, 2023-10-16:
### Question

Hi, is there any documentation or example code that describes how to configure and use the thermostat cluster?
I see in the zb_zcl_thermostat header files that this cluster is done differently than other ones and has some init_server and init_client functions which I don't really get, and I did not find any sufficient description on how to use them.
Is there any reason why there are no "esp_zb_thermostat_cluster_create" and "esp_zb_thermostat_cluster_add" functions in the esp_zigbee_cluster.h file?
I would like to create a thermostat device, which includes a temperature sensor, humidity sensor, and a heating element. 
I also did not manage to run the temperature sensor endpoint with ESP_ZB_DEFAULT_TEMPERATURE_SENSOR_CONFIG from the HA standard. In my SmartThings app, it appears as a "Zigbee Thing" with no attributes.  
I tried to send some measurements from the device, it looks like a success but nothing appears in the app. 
This is my update function:
```
void zb_update_temp(int temperature)
{
    static esp_zb_zcl_report_attr_cmd_t temp_measurement_cmd_req = {};
        temp_measurement_cmd_req.zcl_basic_cmd.src_endpoint = 10;
        temp_measurement_cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        temp_measurement_cmd_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
        temp_measurement_cmd_req.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;

    /* Write new temp */
    esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(10, ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID, &temperature, false);
    
    /* Check for error */
    if(state != ESP_ZB_ZCL_STATUS_SUCCESS) {
        LOG_ERROR("Setting temp attribute failed!");
        return;
    }

    /* Request sending new phase voltage */
    esp_err_t state1 = esp_zb_zcl_report_attr_cmd_req(&temp_measurement_cmd_req);

    /* Check for error */
    if(state1 != ESP_OK) {
        LOG_ERROR("Sending temp attribute report command failed!");
        return;   
    }
    LOG_WARNING("Setting temp success");
    return;
}
```

Thank You in advance for any help 

### Additional context.

I'm using an ESP32-C6 devboard with esp-idf v5.1.1. 
 And an Aeotec Smart Home Hub with SmartThings mobile App

xieqinan, 2023-10-16:
Hello @patryk-belka-wizzdev 

The thermostat device has been supported in latest esp-zigbee-sdk, you can refer to [thermostat cluster](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/ha/esp_zigbee_ha_standard.h#L472) and [thermostat device](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/ha/esp_zigbee_ha_standard.h#L630).

vritzka, 2023-10-17:
I am also building a temperature measurement device.

If possible, I would also really appreciate an example of how to read temperature from a sensor.

patryk-belka-wizzdev, 2023-10-17:
Unfortunately, I'm still struggling with the thermostat cluster. I've followed the HA_color_dimmable_light example (which works fine) to create the thermostat endpoint but I still see it in my mobile app as a "Zigbee Thing" with no attributes to receive or set. 
This is the esp_zb_task I've created:

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = {};
    zb_nwk_cfg.esp_zb_role = ESP_ZB_DEVICE_TYPE_ROUTER;
    zb_nwk_cfg.install_code_policy = INSTALLCODE_POLICY_ENABLE;
    zb_nwk_cfg.nwk_cfg.zczr_cfg.max_children = 10;
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_thermostat_cfg_t thermostat_cfg = ESP_ZB_DEFAULT_THERMOSTAT_CONFIG();
    esp_zb_ep_list_t *esp_zb_thermostat_ep = esp_zb_thermostat_ep_create(10, &thermostat_cfg);
    esp_zb_device_register(esp_zb_thermostat_ep);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

xieqinan, 2023-10-20:
Hello @patryk-belka-wizzdev ,

I believe you have created the thermostat device from the above code. However, I am unsure whether this device is connected to your app or not. Could you please check the connection status between the device and your app? This information would be helpful in explaining how the mobile app can retrieve and set attributes for the thermostat device.

patryk-belka-wizzdev, 2023-10-20:
Sorry if I missed that information. The mobile app is correctly finding the device and pairing is successful. 
Logs similar to these appear:
[image]
The device is online in the mobile app. But in contrast to the HA_dimmable_light example, where I can set multiple things in the app and receive them on the device, with the thermostat I can't do anything in the mobile app. I only see that a Zigbee Thing is connected and whether it's online or offline.
So no problems with pairing and adding to the mobile app, works every time. The only thing I had to change here was the primary_network_channel to ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK. But I think it is not connected with my issue.

iamalek, 2023-10-23:
Hello colleagues!
Please share a complete code example for obtaining temperature values from the Tuya temperature sensor.
I tried all the combinations from all the examples, the sensor is successfully added to the zigbee network by the gateway on the ESP32-C6, but there are no events in the stack from it, so it is not clear how to see the temperature values. Other sensors, such as leaks or doors, work great.
Thank you!

lhespress, 2023-10-27:
@iamalek What do you mean how to get the temperature value on the Zigbee Coordinator role? I think you can reference the code: https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L172

vritzka, 2023-11-01:
@iamalek: My TUYA temperature sensor does not respond to requests by design.

To safe battery, it only wakes up every 300 seconds to transfer data. Or, when I push the small button.

teleger, 2023-11-02:
esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_client_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

en....

The code here will let the program crash, I run on esps3, run through rcp mode, after the program runs for a period of time, the program self-opened, I do not know what to do, the program code is consistent with the author, I am a novice, I am anxious to deal with this problem.

teleger, 2023-11-02:
Hello colleagues! one more question, how do I support multiple zigbee devices, I'm currently running a gateway demo. Now, I want to support multiple zigbee links to me at the same time, how can I let multiple sub-devices join the gateway at the same time? When I see the gateway code, I don't know how to deal with it, I can't find help in the zigbee gateway code, if you can provide an example of this, it will be good. I really appreciate it.

teleger, 2023-11-02:
I have trace enabled, but my debug log has the following print:

I (9060) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x33 received_bytes 51
I (9070) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x2d received_bytes 52
I (9080) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x31 received_bytes 53
I (9090) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x30 received_bytes 54
I (9100) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x2d received_bytes 55
I (9110) ZBOSS: macsplit/zb_macsplit_proto.c:361   rx body byte 0x33 received_bytes 56
I (9120) ZBOSS: macsplit/zb_macsplit_host.c:171   >radio boot indication
I (9120) ZBOSS: macsplit/zb_macsplit_host.c:173   boot indication
I (9130) ZBOSS: macsplit/zb_macsplit_host.c:213   Unexpected zb_mlme_dev_reset_conf - seems, device is asserted
I (9140) ZBOSS: common/zb_error_indication.c:93   >> zb_error_raise severity 254, err_code -2561
I (9150) ZBOSS: common/zb_error_indication.c:59   Error is not handled: severity 254, err_code -2561
I (9160) ZBOSS: common/zb_debug.c:64   Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/common/zb_error_indication.c:81

abort() was called at PC 0x4201a79f on core 0
0x4201a79f: zb_assert at ??:?

Backtrace: 0x403759da:0x3fca5a30 0x4037b22d:0x3fca5a50 0x40380e0e:0x3fca5a70 0x4201a79f:0x3fca5ae0 0x420713af:0x3fca5b10 0x42071403:0x3fca5b40 0x42068815:0x3fca5b70 0x42021a2a:0x3fca5c20 0x42054b5b:0x3fca5c70 0x4200f7d3:0x3fca5c90 0x4200b63a:0x3fca5cb0 0x4037dad1:0x3fca5cf0
0x403759da: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.1/components/esp_system/panic.c:452

....
emm...
....
I don't know what to do next, my log happened on esp32s3, using rcp mode to communicate with esph2, please give me some tips, so that I know how to deal with it, OK?

chshu, 2023-11-02:
@teleger Could you please file separate GitHub issue for each topic? It would be easier to follow up, thanks.

teleger, 2023-11-02:
> @teleger Could you please file separate GitHub issue for each topic? It would be easier to follow up, thanks.

ok，  I will submit a new question，Thank you for your quick reply

chshu, 2024-03-15:
The thermostat and temperature exampes are available now: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample.

Please file new issues if any follow up questions.

closed 2024-03-15 by chshu