espressif/esp-zigbee-sdk #130: Error in building Home Automation project

nalves23, 2023-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1.1

### esp-zigbee-lib verion.

0.5.3

### esp-zboss-lib verion.

0.3.6

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Compile with success the esp-zigbee-sdk project example "HA_on_off_light".

### What is the actual behavior?

At final compilation I get this error report, like I show on the log below. 

(...)
Successfully created esp32h2 image.
Generated C:/Users/nalves/ESP/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_light/build/bootloader/bootloader.bin
[104/104] cmd.exe /C "cd /D C:\Users\nalves\ESP\esp-zigbee-sdk\examples\esp_zigbee_HA_sample\HA_on_off_light\build\bootloader\esp-idf\esptool_py && c:\Users\nalves\ESP_IDF\Espressif\tools\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/nalves/ESP_IDF/Espressif/frameworks/esp-idf-v5.1.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/nalves/ESP/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_light/build/bootloader/bootloader.bin"
Bootloader binary size 0x5190 bytes. 0x2e70 bytes (36%) free.
[891/893] Linking CXX executable on_off_light_bulb.elf
FAILED: on_off_light_bulb.elf
cmd.exe /C "cd . && C:\Users\nalves\ESP_IDF\Espressif\tools\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32H2=0 -Wl,--Map=C:/Users/nalves/ESP/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_light/build/on_off_light_bulb.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32h2.peripherals.ld -T esp32h2.rom.ld -T esp32h2.rom.api.ld -T esp32h2.rom.libgcc.ld -T esp32h2.rom.newlib.ld -T esp32h2.rom.version.ld -T esp32h2.rom.wdt.ld -T esp32h2.rom.heap.ld -T memory.ld -T sections.ld @CMakeFiles\on_off_light_bulb.elf.rsp -o on_off_light_bulb.elf  && cd ."
c:/users/nalves/esp_idf/espressif/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/nalves/ESP/esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_light/managed_components/espressif__esp-zboss-lib/lib/esp32h2/libzboss_port.a(zb_esp_mac.c.obj): in function `.L0 ':
(.text.zb_transceiver_set_channel+0x4): undefined reference to `esp_ieee802154_set_channnel'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Steps to reproduce.

My folders organization:

ESP-IDF- folder:  c:\{$USERS_DIR}\ESP-IDF

ESP-Zigbee project folder:  c:\{$USERS_DIR}\ESP\esp-zigbee-sdk

Project that I want to compile: c:\{$USERS_DIR}\ESP\esp-zigbee-sdk\examples\esp_zigbee_HA_sample\HA_on_off_light

It seems the links was defined correctly.

Want I miss?

xieqinan, 2023-10-17:
@nalves23 ,
The esp-zigbee-lib and esp-zboss-lib had already update to v1.0.0, Could you please use the latest version to compile your project?

nalves23, 2023-10-17:
Thank you  @xieqinan !

Now is work perfectly.

Kind regards.