espressif/esp-zigbee-sdk #131: Assert after update zigbee v1.0.0

diazmanuel, 2023-10-17:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### esp-zigbee-lib verion.

1.0.0

### esp-zboss-lib verion.

1.0.0

### Espressif SoC revision.

ESP32_-H2

### What is the expected behavior?

I don't know what to complete here, the behavior I expect is for it to continue working as it did before the sdk update

### What is the actual behavior?

an assert is generated

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

Hello, after updating the sdk to version 1.0.0 I get the following assert, any idea what could be causing this?

```
assert failed: vTaskStepTick tasks.c:3061 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)
Core  0 register dump:
MEPC    : 0x40800616  RA      : 0x4080b99c  SP      : 0x40821140  GP      : 0x40811a74  
0x40800616: panic_abort at /home/manueldiaz/esp/esp-idf/components/esp_system/panic.c:452

0x4080b99c: __ubsan_include at /home/manueldiaz/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407fa500  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x408212c1  A0      : 0x40821198  A1      : 0x40812a09  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4081c000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000066  S3      : 0x40821198  
S4      : 0x40821198  S5      : 0x420e2cc2  S6      : 0x4081c000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
40821140: 0xa5a5a5a5 0xa5a5a5a5 0x420e2e2c 0x4081037c 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x40812a08
0x4081037c: esp_ptr_in_drom at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:301
 (inlined by) __assert_func at /home/manueldiaz/esp/esp-idf/components/newlib/assert.c:63

40821160: 0x31363033 0xa5a5a500 0x00000000 0x00000000 0xa5000000 0x40812c14 0x420e3294 0x40812bf8
40821180: 0x420e2cc2 0x40812c24 0x40821160 0x40812c28 0x420e2e2c 0x40812a08 0x65737361 0x66207472
408211a0: 0x656c6961 0x76203a64 0x6b736154 0x70657453 0x6b636954 0x73617420 0x632e736b 0x3630333a
408211c0: 0x28282031 0x69547820 0x6f436b63 0x20746e75 0x5478202b 0x736b6369 0x754a6f54 0x2920706d
408211e0: 0x203d3c20 0x78654e78 0x73615474 0x626e556b 0x6b636f6c 0x656d6954 0x00000029 0x00000000
40821200: 0x00000000 0x00000000 0x4081c000 0x42001ce6 0x00000000 0x00003c16 0x00000000 0x4080293c
0x42001ce6: clk_ll_rtc_slow_load_cal at /home/manueldiaz/esp/esp-idf/components/hal/esp32h2/include/hal/clk_tree_ll.h:742
 (inlined by) esp_clk_slowclk_cal_get at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/esp_clk.c:199
 (inlined by) esp_rtc_get_time_us at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/esp_clk.c:108

0x4080293c: esp_system_get_time at /home/manueldiaz/esp/esp-idf/components/esp_timer/src/system_time.c:55

40821220: 0x00000000 0x00000000 0x408113e8 0x420059b6 0x00000000 0x00000000 0x408113e8 0x42002916
0x420059b6: esp_set_time_from_rtc at /home/manueldiaz/esp/esp-idf/components/newlib/port/esp_time_impl.c:121

0x42002916: esp_light_sleep_start at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/sleep_modes.c:1141

40821240: 0x00000001 0x00000060 0x00000001 0x00021340 0x00000010 0x00000000 0x00000000 0x600b1c00
40821260: 0x40820b90 0x4081c000 0x00000004 0x4081c000 0x00000000 0x00000000 0x007d09b5 0x4080cf90
0x4080cf90: vTaskStepTick at /home/manueldiaz/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3062

40821280: 0x00000000 0x00000000 0x007d09b5 0x40807616 0x40820b90 0x40813364 0x4081c000 0x4080d342
0x40807616: vApplicationSleep at /home/manueldiaz/esp/esp-idf/components/esp_pm/pm_impl.c:665

0x4080d342: prvIdleTask at /home/manueldiaz/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4371

408212a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408212c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408212e0: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x40821210 0x4081da44 0x408133bc 0x408133bc 0x408212ec
40821300: 0x408133b4 0x00000019 0x1daabc1e 0x5573aff6 0x408212ec 0x00000000 0x00000000 0x40820ce8
40821320: 0x454c4449 0xf2228d00 0xf2e580b5 0x003ecb64 0x00000000 0x408212e0 0x00000000 0x00000000
40821340: 0x00000000 0x00000000 0x00000000 0x4081e3e4 0x4081e44c 0x4081e4b4 0x00000000 0x00000000
40821360: 0x00000001 0x00000000 0x00000000 0x00000000 0x420bf92a 0x00000000 0x00000000 0x00000000
0x420bf92a: _cleanup_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

40821380: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408213a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408213c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408213e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40821400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40821420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000200
0x40000200: __call_spi_flash_mmap_page_num_init in ROM

40821440: 0x0000000c 0x0b56000a 0x00000000 0x4084da84 0x00000054 0x00000000 0x40821454 0x00000000
40821460: 0x00000000 0x00000000 0x4082146c 0xffffffff 0x4082146c 0x4082146c 0x00000000 0x40821480
40821480: 0xffffffff 0x40821480 0x40821480 0x00000001 0x00000001 0x00000000 0x0b00ffff 0x00000000
408214a0: 0xb33fffff 0x00000000 0x00000054 0x00000000 0x408214ac 0x00000000 0x00000000 0x00000000
408214c0: 0x408214c4 0xffffffff 0x408214c4 0x408214c4 0x00000000 0x408214d8 0xffffffff 0x408214d8
408214e0: 0x408214d8 0x00000001 0x00000001 0x00000000 0x8000ffff 0x00000000 0xb33fffff 0x00000000
40821500: 0x000000a0 0x40808e9c 0x408018a6 0x40821110 0x40811a74 0x407fa500 0x400106da 0xbdd50000
0x40808e9c: esp_cpu_wait_for_intr at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/cpu.c:121

0x408018a6: do_cpu_retention at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/sleep_cpu.c:687
 (inlined by) esp_sleep_cpu_retention at /home/manueldiaz/esp/esp-idf/components/esp_hw_support/sleep_cpu.c:720

0x400106da: Cache_Suspend_ICache in ROM

40821520: 0xc0000000 0x408215a8 0x40811e34 0x40821504 0x600c8024 0x00000002 0x600c8024 0x00000000

ELF file SHA256: fee54b9ea48e34da

Rebooting...

```

thanks

xieqinan, 2023-10-18:
Hello @diazmanuel,

I checked out the branch `idf v5.1.1-588-gc1c843f5e2` and tested the [sleep end device](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/sleepy_end_device) example, which worked fine.

Have there been any changes in your test environment?

diazmanuel, 2023-10-18:
hello @xieqinan
 Yes, I checked that the example works correctly, but I had already started a project which tried to migrate and it failed, after reviewing it I found the problem, for some reason if the CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP parameter in the sdk config is commented or disabled, this assert happens, looking in The documentation shows this parameter as experimental.
Another thing that I can't understand is why this worked with the previous version, the migration was done in both sdk at the same time, the idf and the zigbee one.

xieqinan, 2023-10-19:
@diazmanuel 

> Another thing that I can't understand is why this worked with the previous version, the migration was done in both sdk at the same time, the idf and the zigbee one.

Because the esp-zigbee-sdk v1.0.0 optimizes the bottom current state for the sleep device, which needs the `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` support. As a result, if this MARCO is disabled that may trigger the panic.

diazmanuel, 2023-10-24:
The problem still exists because if you disable the CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP macro, it keeps crashing, for the moment I keep it enabled and it works correctly, it's up to you if you want to solve it or close the problem