espressif/esp-zigbee-sdk #132: Add the ability to remove the filter that ignores reports from clusters that were not declared

diazmanuel, 2023-10-19:
### Is your feature request related to a problem?

As mentioned in the issue #111  it is important to have the ability to do this when designing a gateway device, a clear example of this is when we have a gateway that supports third party devices, in which the commands through the gateway are form in the cloud.

### Describe the solution you'd like.

@xieqina mentions that there is a filter applied in which the report is discarded if the corresponding cluster was not declared 

A possible solution would be to add a parameter to the action handler register esp_zb_core_action_handler_register(zb_action_handler) that allows you to configure whether you want to ignore said reports.

### Describe alternatives you've considered.

In addition, @xieqina provides a solution using esp_zb_raw_command_handler_register(), this solution would be outside the api provided by esp and would work directly at the zboss api level, which becomes somewhat cumbersome and difficult to implement. I also tested the implementation as  @xieqina explain and it partially works, after a while the device crashes giving the following log infinitely
W (296453) ZB_ESP_MAC: MAC RX buffer is full!

xieqinan, 2023-10-20:
@diazmanuel ,

I believe you only need to return `false` when `cmd_info->cmd_id` is equal to `ZB_ZCL_CMD_REPORT_ATTRIB`. In this case, the Zigbee stack will handle the report command and filter reports from clusters that were not declared.

```
#include "zboss_api.h"
bool esp_zb_raw_command_handler(uint8_t bufid)
{
    zb_zcl_report_attr_req_t *req = NULL;
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);

    if(cmd_info && cmd_info->is_common_command && cmd_info->cmd_id == ZB_ZCL_CMD_REPORT_ATTRIB) {
        return false;
    }
}
```

diazmanuel, 2023-10-20:
hi @xieqinan 

- it is necessary to call the function zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_SUCCESS); At the end of the handler to free memory?

- That is to say, I have to obtain the information and process it within the if rigth? somthing like what you comment 
```
#include "zboss_api.h"
bool esp_zb_raw_command_handler(uint8_t bufid)
{
    zb_zcl_report_attr_req_t *req = NULL;
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);

    if(cmd_info && cmd_info->is_common_command && cmd_info->cmd_id == ZB_ZCL_CMD_REPORT_ATTRIB)
    {
        printf("dst_ep: %d, cluster id: 0x%x\n", cmd_info->addr_data.common_data.dst_endpoint, cmd_info->cluster_id);
        do {
            ZB_ZCL_GENERAL_GET_NEXT_REPORT_ATTR_REQ(bufid, req);
            if(req){
               //process
            }    
        }while(req);
    }
    zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_SUCCESS);
    return true;
}
```

- There is no way to force reporting of an undeclared cluster to go through the handler declared in esp_zb_core_action_handler_register()

xieqinan, 2023-10-23:
@diazmanuel ,

> * it is necessary to call the function zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_SUCCESS); At the end of the handler to free memory?

Yes, because the raw command requires the user to take responsibility for freeing the bufid of ZBOSS.

> #include "zboss_api.h"
> bool esp_zb_raw_command_handler(uint8_t bufid)
> {
>     zb_zcl_report_attr_req_t *req = NULL;
>     zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, [...]

If you want to handle the Report command, you can do as above.

> * There is no way to force reporting of an undeclared cluster to go through the handler declared in esp_zb_core_action_handler_register()

The `esp_zb_core_action_handler()` will handle standard and declared report attribute commands that are returned as `false` by the raw command handler.

diazmanuel, 2023-10-24:
Hello @xieqinan, thanks for the help, the handler I ended up using is the following

```
bool esp_zb_raw_command_handler(uint8_t bufid)
{
    uint32_t cast_value=0;
    zb_zcl_report_attr_req_t *req = NULL;
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
    esp_zb_ieee_addr_t ieee_addr;
    bool res=true;
    if(cmd_info && cmd_info->is_common_command && cmd_info->cmd_id == ZB_ZCL_CMD_REPORT_ATTRIB)
    {
        do {
            ZB_ZCL_GENERAL_GET_NEXT_REPORT_ATTR_REQ(bufid, req);
            if(req){
                if(esp_zb_ieee_address_by_short(cmd_info->addr_data.common_data.source.u.short_addr,ieee_addr)==ESP_OK){
                    if(ZBM_Is_Attr_Number(req->attr_type)){
                        if(ZBM_Get_Zigbee_Attr_Number(req->attr_type,req->attr_value,&cast_value)){
                            ESP_LOGI(TAG,"mac address: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, src_ep: %d, cluster id: 0x%x, attr id: %d, attr type: %d, val:%ld",
                            ieee_addr[7],ieee_addr[6],ieee_addr[5],ieee_addr[4],ieee_addr[3],ieee_addr[2],ieee_addr[1],ieee_addr[0],
                            cmd_info->addr_data.common_data.src_endpoint,
                            cmd_info->cluster_id,
                            req->attr_id,
                            req->attr_type,
                            cast_value
                            );
                        }
                    }
                }
            }    
        }while(req);
        zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_SUCCESS);
        return true;
    }
    return false;
}
```

closed 2023-10-24 by diazmanuel