espressif/esp-zigbee-sdk #133: doubts about client, server and the latest update of the library v1.0.0

diazmanuel, 2023-10-19:
### Question

I have some doubts about how this protocol works.

- What is the difference between client and server? I understand that if you want to have a cluster and generate reports or modify its values, you must declare yourself as a server, but I don't see in which cases you should declare yourself as a client.

- In the last update of the library v1.0.0, changes were made to the configuration of the reports, Why was the esp_zb_zcl_cmd_direction_t direction parameter added and how is it used?

xieqinan, 2023-10-20:
@diazmanuel ,

> What is the difference between client and server? I understand that if you want to have a cluster and generate reports or modify its values, you must declare yourself as a server, but I don't see in which cases you should declare yourself as a client.

1. **Server:**
   - A server in Zigbee refers to a device or application that provides services or resources. These services can include functionalities like temperature sensing, light control, or any other capabilities a device might have.
   - Servers respond to requests made by clients. For example, a temperature sensor (server) might provide temperature readings to a coordinator or a remote control (client) might send commands to a light bulb (server) to turn it on or off.
   - In Zigbee, devices often have both client and server functionalities. For instance, a light bulb might be a server for on/off commands and a client for network-related commands.

2. **Client:**
   - A client in Zigbee refers to a device or application that requests services or resources from servers.
   - Clients initiate requests to servers to access specific functionalities. For example, a remote control (client) sends a command to a light bulb (server) to change its brightness level.
   - Devices can act as clients for some interactions and servers for others, depending on the context of the communication.

In summary, servers provide services, and clients consume these services by making requests. Zigbee devices often have a mix of server and client functionalities to enable various interactions within a Zigbee network.

> In the last update of the library v1.0.0, changes were made to the configuration of the reports, Why was the esp_zb_zcl_cmd_direction_t direction parameter added and how is it used?

The question can refer to Zigbee cluster library spec 2.5.7.1.2 section.

[image]

diazmanuel, 2023-10-20:
thanks for the reply