espressif/esp-zigbee-sdk #134: Why is my zb_action_handler not running?

vritzka, 2023-10-20:
### Question

I have the Zigbee examples running. They all work well!

But I am struggling to read a temperature sensor.

I am running an  ESP32-s3 + ESP32-C6 (as RCP).

I am trying to read from a [Tuya Zigbee Mini Sensor](https://www.aliexpress.com/item/1005005454209285.html).

The sensor is discovered and successfully added to my network, I can see that in the console.

But I cannot read the temperature from it.

Could you please give me some advice on what I am doing wrong?

([My full code on Github](https://github.com/vritzka/TomatoTent/blob/v2/main/esp_zigbee_gateway.c))

`
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    ESP_LOGW(TAG, "Receive Zigbee action");
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID:
        ret = zb_attribute_handler((esp_zb_zcl_set_attr_value_message_t *)message);
        break;
   case ESP_ZB_CORE_REPORT_ATTR_CB_ID:
        ret = zb_attribute_reporting_handler((esp_zb_zcl_report_attr_message_t *)message);
        break;
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t *)message);
        break;
    case ESP_ZB_CORE_CMD_REPORT_CONFIG_RESP_CB_ID:
        ret = zb_configure_report_resp_handler((esp_zb_zcl_cmd_config_report_resp_message_t *)message);
        break;      
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
	esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    uint8_t test_attr;
    test_attr = 0;
    /* basic cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    /* identify cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    /* create client role of the cluster */
    esp_zb_attribute_list_t *esp_zb_identify_client_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    /* temp meas cluster */
    esp_zb_temperature_meas_cluster_cfg_t temperature_cfg;                                                                          
    temperature_cfg.measured_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_UNKNOWN;                   
    temperature_cfg.min_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MIN_VALUE_INVALID;                   
    temperature_cfg.max_value = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_MAX_VALUE_INVALID;                    
    esp_zb_attribute_list_t *esp_zb_temperature_cluster = esp_zb_temperature_meas_cluster_create(&temperature_cfg);      
    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_client_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    esp_zb_cluster_list_add_temperature_meas_cluster(esp_zb_cluster_list, esp_zb_temperature_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint (cluster_list) to endpoint list */
	esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, THERMOMETER_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID);
	esp_zb_device_register(esp_zb_ep_list);   
    
    esp_zb_core_action_handler_register(zb_action_handler);  
    
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    esp_zb_set_secondary_network_channel_set(ESP_ZB_SECONDARY_CHANNEL_MASK);  
    
    ESP_ERROR_CHECK(esp_zb_start(true));
	esp_zb_main_loop_iteration();
    
#if(CONFIG_ZB_RADIO_MACSPLIT_UART)
    esp_zb_add_rcp_failure_cb(rcp_error_handler);
#endif
    esp_zb_main_loop_iteration();
    esp_rcp_update_deinit();
    vTaskDelete(NULL);
}`

This code is run every 5 seconds to read the sensor:

`            esp_zb_zcl_read_attr_cmd_t cmd_req;
            cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = my_tent.thermometer_short_addr;
            cmd_req.zcl_basic_cmd.dst_endpoint = 1;
            cmd_req.zcl_basic_cmd.src_endpoint = THERMOMETER_ENDPOINT;
            cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_GROUP_ENDP_NOT_PRESENT;
            cmd_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
            uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID};
            cmd_req.attr_number = sizeof(attributes) / sizeof(uint16_t);;
            cmd_req.attr_field = attributes;
                                                                     
            esp_zb_zcl_read_attr_cmd_req( &cmd_req);  `

When I run  _esp_zb_zcl_read_attr_cmd_req_, I understand that  _zb_action_handler_ should run, because I registered it with  esp_zb_core_action_handler_register(zb_action_handler);  ?

But it doesn't, and I can't read the sensor.

vritzka, 2023-10-20:
It suddenly started working. Nevermind.

vritzka, 2023-10-20:
I am getting some output now, right after the sensor joined the network:

W (69731) ESP_ZB_GATEWAY: Receive Zigbee action
I (69731) ESP_ZB_GATEWAY: Reveived report from address(0x1df2) src endpoint(1) to dst endpoint(1) cluster(0x402)
I (69731) ESP_ZB_GATEWAY: Received report information: attribute(0x0), type(0x29), value(50)

It does that 3 or 4 times, then it stops.

This attempt at reading the sensor seems to have no effect:

```
             esp_zb_zcl_read_attr_cmd_t cmd_req;
            cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = my_tent.thermometer_short_addr;
            cmd_req.zcl_basic_cmd.dst_endpoint = 1;
            cmd_req.zcl_basic_cmd.src_endpoint = THERMOMETER_ENDPOINT;
            cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_GROUP_ENDP_NOT_PRESENT;
            cmd_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
            uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID};
            cmd_req.attr_number = sizeof(attributes) / sizeof(uint16_t);;
            cmd_req.attr_field = attributes;
                                                                     
            esp_zb_zcl_read_attr_cmd_req( &cmd_req); 
```

xieqinan, 2023-10-20:
Hello @vritzka 

I couldn't find any group clusters. Why use groupcast to send commands?
> cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_GROUP_ENDP_NOT_PRESENT;

Please attempt to use `ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT` instead.

xieqinan, 2023-10-20:
>It does that 3 or 4 times, then it stops.

Could you attempt to use the sniffer to check the frame the gateway received? it is possiable the report frame has not been sent by the  [Tuya Zigbee Mini Sensor](https://www.aliexpress.com/item/1005005454209285.html).

vritzka, 2023-10-20:
Thank you @xieqinan,

its partly working.

The device reports a value.

It does it about 10 times, then stops.

I don't have sniffer at hand, will get one.

closed 2023-10-20 by vritzka