espressif/esp-zigbee-sdk #137: add some zboss resources

diazmanuel, 2023-10-27:
### Is your feature request related to a problem?

Hello, I leave a list of resource requests present in zboss that I would like to see added if possible, I also leave the use case for them, most of them would be easy to implement in my opinion

1. add the zb_zdo_signal_device_authorized_params_t struct for the ESP_ZB_ZDO_SIGNAL_DEVICE_AUTHORIZED signal This structure would carry information about the new device, unlike the signal ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE which mixes new and old devices performing a rejoin, meaning that this signal is the only one that carries exclusive information about the new devices. I understand that they could use the signal ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE to obtain the mac address but it is more convenient and logical to make this structure available

2. add a function to close the network such as zb_bdb_close_network(0); from zboss, in my case this is very useful since I need to take individual control of the devices that are added, so I would use it to close the network after a new device is added. currently use the original function zb_bdb_close_network(0); and it worked correctly, but I think they should implement it in the sdk

3. add a function to modify the macros of the commissions as specified in section 2.5.2.1 of General Constants of the zigbee 3.0 documentation, in my case I need the possibility of modifying the time for which the steering is performed, that is, the macro BDBC_MIN_COMMISSIONING_TIME , currently I solve this with an alarm and the function of closing the network mentioned above.

4. Add sequence number for the read/write request command attribute ad mention in #135, At the moment I am using the mac addresses to match the read and write requests with their responses, but it would be very convenient to have this sequence number

5. add a function that allows you to configure the default local reporting parameters, currently I implement the following block of code to achieve this but it may be useful for others to allow you to edit these default values.
```
    esp_zb_device_register(esp_zb_ep_list);

    esp_zb_ep_list_t *aux= esp_zb_ep_list;
    while(aux!=NULL){
        if(aux->endpoint.reporting_info!=NULL){
            for(int i=0;i<aux->endpoint.rep_info_count;i++){
                if(aux->endpoint.reporting_info[i].cluster_role==ESP_ZB_ZCL_CLUSTER_SERVER_ROLE){
                    aux->endpoint.reporting_info[i].u.send_info.def_max_interval=REPORTABLE_MAX_INTERVAL_S;
                    aux->endpoint.reporting_info[i].u.send_info.def_min_interval=REPORTABLE_MIN_INTERVAL_S;
                    aux->endpoint.reporting_info[i].u.send_info.max_interval=REPORTABLE_MAX_INTERVAL_S;
                    aux->endpoint.reporting_info[i].u.send_info.min_interval=REPORTABLE_MIN_INTERVAL_S;
                    aux->endpoint.reporting_info[i].u.send_info.delta.u32=REPORTABLE_DELTA;
                }
            }
        }
        aux=aux->next;
    }
```

xieqinan, 2023-11-06:
Hello @diazmanuel ,

Thank you for your suggestions that is very practical and helpful. We will take a consider for them. 

> 1. add the zb_zdo_signal_device_authorized_params_t struct for the ESP_ZB_ZDO_SIGNAL_DEVICE_AUTHORIZED signal This structure would carry information about the new device, unlike the signal [...]

> 2. add a function to close the network such as zb_bdb_close_network(0); from zboss, in my case this is very useful since I need to take individual control of the devices that are added, so I would [...]

> 5. add a function that allows you to configure the default local reporting parameters, currently I implement the following block of code to achieve this but it may be useful for others to allow you to edit these default values.

These requests will be supported in recent version.

> 3. add a function to modify the macros of the commissions as specified in section 2.5.2.1 of General Constants of the zigbee 3.0 documentation, in my case I need the possibility of modifying the [...]

Regarding this issue, could you try utilizing the `esp_zb_zdo_permit_joining_req()` function to modify the time for steering?

> 4. Add sequence number for the read/write request command attribute ad mention in [Add sequence number for the read/write request command attribute (TZ-365) #135](https://github.com/espressif/esp-zigbee-sdk/issues/135), At the moment I am using the mac addresses to match the read and write requests with their responses, but it would be very convenient to have this sequence number

This issue will not be supported in the recent version temporarily. This is because it requires providing the same support for other commands, which necessitates time for testing and verification. Moreover, it is not deemed necessary or urgent at this time.

closed 2023-11-10 by espressif-bot

diazmanuel, 2023-11-10:
hello @xieqinan thanks for the reply, iam trying to implement the 5 point  (configure the reports), i read the comments of the new functions and notice that now there is a start and stop function for reporting, so after update the reporting parameters i need to call the start function to actually start reporting ? also in the struct esp_zb_zcl_attr_location_info_t the cluster_id variable type is a typedef enum with the supported clusters, i suppose that this will stop me to use custom clusters right ? could you change the var type to uint16_t? In the same struct there is a parameter called manuf_code i dont understand the use of it, iam planning to call esp_zb_zcl_find_reporting_info(filter) to get the reporting info and then call  esp_zb_zcl_update_reporting_info with the new values, but the i never use the manuf_code anywhere