espressif/esp-zigbee-sdk #138: When using the esp_HA_customized_light.c example, an error occurred, causing a restart，Hardware :esp32s3

teleger, 2023-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### esp-zigbee-lib version.

1.0.0

### esp-zboss-lib version.

1.0.0

### Espressif SoC revision.

ESP32-S3

### What is the expected behavior?

  esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

### What is the actual behavior?

The esp32s3 ran for less than a minute and then restarted
-----------------------------------------------------------
annotation： esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
will normalize

### Steps to reproduce.

1.  on esp32h2  , run  /examples/esp_zigbee_customized_devices/customized_client 
2.  on esp32s3  , run  /examples/esp_zigbee_customized_devices/customized_server
3.  esp32s3 The esp32s3 ran for less than a minute and then restarted

### More Information.

I have trace enabled, but my debug log has the following print:

I (9060) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x33 received_bytes 51
I (9070) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x2d received_bytes 52
I (9080) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x31 received_bytes 53
I (9090) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x30 received_bytes 54
I (9100) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x2d received_bytes 55
I (9110) ZBOSS: macsplit/zb_macsplit_proto.c:361 rx body byte 0x33 received_bytes 56
I (9120) ZBOSS: macsplit/zb_macsplit_host.c:171 >radio boot indication
I (9120) ZBOSS: macsplit/zb_macsplit_host.c:173 boot indication
I (9130) ZBOSS: macsplit/zb_macsplit_host.c:213 Unexpected zb_mlme_dev_reset_conf - seems, device is asserted
I (9140) ZBOSS: common/zb_error_indication.c:93 >> zb_error_raise severity 254, err_code -2561
I (9150) ZBOSS: common/zb_error_indication.c:59 Error is not handled: severity 254, err_code -2561
I (9160) ZBOSS: common/zb_debug.c:64 Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/common/zb_error_indication.c:81

abort() was called at PC 0x4201a79f on core 0
0x4201a79f: zb_assert at ??:?

Backtrace: 0x403759da:0x3fca5a30 0x4037b22d:0x3fca5a50 0x40380e0e:0x3fca5a70 0x4201a79f:0x3fca5ae0 0x420713af:0x3fca5b10 0x42071403:0x3fca5b40 0x42068815:0x3fca5b70 0x42021a2a:0x3fca5c20 0x42054b5b:0x3fca5c70 0x4200f7d3:0x3fca5c90 0x4200b63a:0x3fca5cb0 0x4037dad1:0x3fca5cf0
0x403759da: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.1/components/esp_system/panic.c:452

....
emm...
....
I don't know what to do next, my log happened on esp32s3, using rcp mode to communicate with esph2, please give me some tips, so that I know how to deal with it, OK?

xieqinan, 2023-11-03:
@teleger 
> 2\. on esp32s3  , run  /examples/esp_zigbee_customized_devices/customized_server

Please use another ESP32H2 to run customized_server example,  because the ESP32S3 does not support IEEE802154.

teleger, 2023-11-03:
> @teleger
> 
> > 2. on esp32s3  , run  /examples/esp_zigbee_customized_devices/customized_server
> 
> Please use another ESP32H2 to run customized_server example, because the ESP32S3 does not support IEEE802154.

i am use h2 + s3 board, run uart rsp mode , code run s3?

xieqinan, 2023-11-03:
@teleger，

You can use the ESP32S3 + ESP32H2 to support Zigbee by referring to the [gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) in the ESP-Zigbee SDK repository. Before doing so, please make sure to read the README.md file in the example.

teleger, 2023-11-06:
> @teleger，
> 
> You can use the ESP32S3 + ESP32H2 to support Zigbee by referring to the [gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) in the ESP-Zigbee SDK repository. Before doing so, please make sure to read the README.md file in the example.

on the esp32h2 board:
run  : /examples/esp_zigbee_HA_sample/HA_on_off_light
on the esp32s3 + esp32h2 board:
run : /examples/esp_zigbee_HA_sample/HA_on_off_switch
is  OK 
however,I want to use this example,
/examples/esp_zigbee_customized_devices   client and server folder;
The situation described above arises.

You can run the code on the hardware I'm talking about, and if you run it and it doesn't happen, then I'll close the problem. If there is a problem, can you take a look? Or is the current code not going to run on the hardware I'm talking about?

xieqinan, 2023-11-06:
@teleger ,

The `/examples/esp_zigbee_customized_devices/customized_client` is designed specifically for the Zigbee end device running on ESP32H2 and will work fine. Expecting it to run on ESP32S3+H2 is impractical and wasteful.

closed 2023-11-09 by teleger