espressif/esp-zigbee-sdk #139: Communicate with Digi Xbee module

nopnop2002, 2023-11-04:
### Question

I want to use esp_zigbee_cli to communicate with Digi Xbee module.

Is this possible?

xieqinan, 2023-11-07:
Hello @nopnop2002 ,

Do you have try to test them?

As far as I am aware, the XBee S2C is an RF module for IEEE 802.15.4. The `esp-zigbee-cli` is designed to provide a command-line interface for users to send IEEE 802.15.4 frames. It appears that there is an inclusive relationship between them. So I think they can not communicate with each other.

nopnop2002, 2023-11-07:
@xieqinan 

>As far as I am aware, the XBee S2C is an RF module for IEEE 802.15.4. 

The XBee module has three wireless methods: ZigBee/WiFi/IEEE 802.15.4.

XBee S2C uses ZigBee wireless technology.

xieqinan, 2023-11-08:
@nopnop2002 

> I'm using S2C+ZigBee firmware.

Could you please flash the `esp_zigbee_cli` example to the ESP32C6? After flashing, input the following commands into the console to configure `esp_zigbee_cli` and establish a Zigbee network. 

1. Set the role to Zigbee Coordinator:
```
bdb -r zc
```

2. Set the channel to 18:
```
bdb -c 18
```

3. Disable the install code policy:
```
bdb -i policy disable
```

4. Start the Zigbee network:
```
bdb -s
```
 Finally, join the S2C Xbee device to the Zigbee network.
  
Once the above steps are completed, you can test communication within this network.

 Note that the command can refer the [README](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_cli/README.md).

nopnop2002, 2023-11-08:
@xieqinan 
```
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (359) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (389) phy: libbtbb version: b684fcb, May 17 2023, 20:14:35
I (399) main_task: Returned from app_main()
> bdb -r zc
I (12579) : Coordinator role set
> bdb -c 18
I (16929) : Setting channel to 18
> bdb -i policy disable
I (21729) : Done
> bdb -s
I (25709) : Started coordinator
I (25789) : Done
I (25799) ESP_ZB_CLI: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
> I (26299) ESP_ZB_CLI: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (26299) ESP_ZB_CLI: Formed network successfully (Extended PAN ID: e0:47:41:fe:ff:ca:4c:40, PAN ID: 0x76a5, Channel:18)
I (26789) ESP_ZB_CLI: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (26789) ESP_ZB_CLI: Joined network successfully
```

PANID is displayed as 0x76a5.
Do I need to change my PANID?

xieqinan, 2023-11-10:
Hello @nopnop2002,

> ZDO Config Ready (0x17), status: ESP_FAIL

It is triggered by the absent pre-configurations manufacturer information, we provide a [tool](https://github.com/espressif/esp-zigbee-sdk/tree/main/tools/mfg_tool) to generate and flash it.

> Extended PAN ID: e0:47:41:fe:ff:ca:4c:40, PAN ID: 0x76a5, Channel:18

From the log you provided, I believe that the Xbee may not have joined the network formed by the coordinator (ESP32H2). I don't see any information exchange between Xbee and ESP32H2, so you need to ensure that the Xbee has the ability to join a network first.

> How to send [HelloWorld]?

It requires additional application layer support for this feature. The `esp-zigbee-cli` only provides basic commands for users to interact with Zigbee devices. It has not provide the transparent transmission feature directly which needs the user to implement.

nopnop2002, 2023-11-10:
@xieqinan 

Thank you.

xieqinan, 2023-11-22:
@nopnop2002 ,

Do you have any other questions or requests that require assistance? If not, please consider closing this issue.

closed 2023-11-22 by nopnop2002