espressif/esp-zigbee-sdk #140: Crash after after Macsplit uart rx buffer full

pieterjanbuntinx, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1 (release version)

### esp-zigbee-lib version.

1.0.2

### esp-zboss-lib version.

1.0.2

### Espressif SoC revision.

ESP32-S3 & ESP32-C6 (UART-RCP)

### What is the expected behavior?

The console just displays an error instead of crashing

### What is the actual behavior?

The console displays a few `ZB_ESP_MACSPLIT_UART: Macsplit uart rx buffer full !!` errors and afterwards crashes. I have tried experimenting with queues to limit the amount of requests that get sent (and therefore also received) but my requests seemed to always time out. 

### Steps to reproduce.

Make a device with this amount of endpoints and clusters join the network. This is a custom NRF device I developed. About 50% of the time the ESP does not crash. 

### More Information.

```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378cee
0x40378cee: esp_cpu_wait_for_intr at C:/Users/Pieter-JanBuntinx/esp-v5.1/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Nov  7 2023 12:58:14
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00138800
I (86) boot:  3 zb_storage       Unknown data     01 81 00149000 00004000
I (94) boot:  4 zb_fct           Unknown data     01 81 0014d000 00000400
I (101) boot:  5 rcp_fw           Unknown data     01 82 0014e000 000af000
[... 218 lines cut ...]
0x4201b6eb: zb_assert at ??:?

Backtrace: 0x403759be:0x3fca4d60 0x4037b075:0x3fca4d80 0x40380ba6:0x3fca4da0 0x4201b6eb:0x3fca4e10 0x42019d46:0x3fca4e40 0x42030c6c:0x3fca4e70 0x42022976:0x3fca4eb0 0x420544b7:0x3fca4f00 0x4200f4d7:0x3fca4f20 0x4200b130:0x3fca4f40 0x4037d919:0x3fca4fb0  
0x403759be: panic_abort at C:/Users/Pieter-JanBuntinx/esp-v5.1/esp-idf/components/esp_system/panic.c:452

0x4037b075: esp_system_abort at C:/Users/Pieter-JanBuntinx/esp-v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40380ba6: abort at C:/Users/Pieter-JanBuntinx/esp-v5.1/esp-idf/components/newlib/abort.c:38

0x4201b6eb: zb_assert at ??:?

0x42019d46: zb_buf_cut_left_func at ??:?

0x42030c6c: zb_nwk_handle_mcps_data_confirm at ??:?

0x42022976: zb_sched_loop_iteration at ??:?

0x420544b7: zboss_main_loop_iteration at ??:?

0x4200f4d7: esp_zb_main_loop_iteration at ??:?

0x4200b130: esp_zb_task at C:/GIT/esp-zigbee-sdk/examples/esp_zigbee_gateway/main/esp_zigbee_gateway.c:378 (discriminator 2)

0x4037d919: vPortTaskWrapper at C:/Users/Pieter-JanBuntinx/esp-v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

pieterjanbuntinx, 2023-11-07:
Oops, I forgot to tell you which firmware I am using. I am using a modified version of the gateway example. You can find the code in my repo: [](https://github.com/pieterjanbuntinx/esp-zigbee-gateway-test)

xieqinan, 2023-11-08:
@pieterjanbuntinx 
Ok, I will check this issue. As an alternative approach, please try using the ESP32C6 solely as the gateway for your example. This method does not require macsplit support.

teleger, 2023-11-16:
> 糟糕，我忘了告诉你我正在使用哪个固件。我正在使用网关示例的修改版本。您可以在我的存储库中找到代码：[链接](https://github.com/pieterjanbuntinx/esp-zigbee-gateway-test)

Hello, I have referred to your gateway code and transplanted most of your code. I think your gateway code is very correct and can adapt to multiple sub-device terminals, such as temperature, humidity, pressure and so on, because I am writing an application about the gateway.
I also encountered the same problem as you, when multiple sub-devices are connected to the gateway in turn, the gateway will crash, this is a probability problem, and the probability is very high, the error occurred is the same as yours. Could the crash have been caused by the buffer being full? I think this is not the main problem, more may be some reasons for conflict, I have not noticed, and now I have no way to solve

teleger, 2023-11-16:
> Oops, I forgot to tell you which firmware I am using. I am using a modified version of the gateway example. You can find the code in my repo: [Link](https://github.com/pieterjanbuntinx/esp-zigbee-gateway-test)

And the other thing I want to ask is, when this esp_zb_zdo_simple_desc_req function fails and crashes,
The output of log is  ：
Assertion failed /home/xieqinan/ESP/esp-zboss/components/zboss_stack/zboss/nwk/nwk_multimac_proxy.c:206

The stack of this function, I can't see, I don't know what I did, this is not a necessary bug, if you are in the same situation, you have a good way to deal with it

chshu, 2023-12-19:
The current Zboss macsplit based zigbee gateway+rcp solution is an experimental feature, we do see some stability issues recently.

We are working on a new zigbee gateway solution which will base on the [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp), some stress tests are on going, will push to github in 2024 Q1.

lmahmutov, 2024-01-12:
have similar issue 
[image]

chshu, 2024-02-04:
@pieterjanbuntinx The [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) based Zigbee Gateway example is available now: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway

Note:
- Please use the esp-idf master branch ([Commit ad3e1e3](https://github.com/espressif/esp-idf/commit/ad3e1e3daf7037e58c650c735e1f6218ee886651) or later). It will be supported in the coming IDF v5.1.3 and later release versions.
- Enable `OPENTHREAD_NCP_VENDOR_HOOK` option before building [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) example.

Feel free to try out the new feature, and let us know if any questions or suggestions.

NCP could be used for 3rd party host SoC. The RCP mode is recommended when using both Host and Zigbee SoCs from Espressif.

closed 2024-02-04 by chshu

pieterjanbuntinx, 2024-02-08:
Thanks @chshu . I've tested the new version and all seems to work fine.