espressif/esp-zigbee-sdk #141: Invalid response callback on write command request

diazmanuel, 2023-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-588-gc1c843f5e2

### esp-zigbee-lib version.

1.0.2

### esp-zboss-lib version.

1.0.2

### Espressif SoC revision.

ESP32_-H6

### What is the expected behavior?

the expected behavior is to work like the read callback

### What is the actual behavior?

after sending the write attribute command request (esp_zb_zcl_write_attr_cmd_req) with multiple or single attributes the ESP_ZB_CORE_CMD_WRITE_ATTR_RESP_CB_ID callback called with only 1 element in the attribute list, the status of the callback message is ESP_ZB_ZCL_STATUS_SUCCESS (message->info.status) also the only element of the array list posses the status in ESP_ZB_ZCL_STATUS_SUCCESS (message->variables->status) but the attribute id is always 0xffff. so the bug is that the attribute list is always of single element and the attribute id is always 0xffff. on the other hand the receiver receives the writer command without problem.
this update was referred for #106 so the read callback works perfectly as mentioned by @xieqinan but the same logic was no apply in the write callback

### Steps to reproduce.

1. Step
2. Step
3. Step
...

xieqinan, 2023-11-08:
@diazmanuel ,

Regarding this issue, it's not a bug and has been illustrated in the [comment](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L1264).

diazmanuel, 2023-11-09:
hi @xieqinan 
Sorry, I had misunderstood the comment. In any case, it is not possible to return the list of attributes written in that list when the writing is done successfully, as is done in the case of reading? I understand that this list is only returned in the cases in which there is an error, but I am using that same list to be able to match the message sent with its response, since there is no sequence number to be able to make the match, or if there is no way to identify who message corresponds to the callback, I am designing a gateway which receives high message traffic and I need to be able to identify its responses without errors

xieqinan, 2023-11-10:
@diazmanuel ,

> there is no sequence number to be able to make the match, or if there is no way to identify who message corresponds to the callback,

We have already taken it into consideration, and additional support for it will be added in subsequent versions.

diazmanuel, 2023-11-10:
hello @xieqinan, yes sure

```
void sendTest(){
    esp_zb_ieee_addr_t ieee_addr;
    ieee_addr[7]=0x60;
    ieee_addr[6]=0x55;
    ieee_addr[5]=0xf9;
    ieee_addr[4]=0xff;
    ieee_addr[3]=0xfe;
    ieee_addr[2]=0xf7;
    ieee_addr[1]=0x73;
    ieee_addr[0]=0x3a;
    uint16_t auxval=0;

    esp_zb_zcl_attribute_t attr_field[4];
    attr_field[0].id=16;
    attr_field[0].data.type=ESP_ZB_ZCL_ATTR_TYPE_S16;
    auxval=1;
    attr_field[0].data.value=malloc(sizeof(uint16_t));
    memcpy(attr_field[0].data.value,&auxval,sizeof(uint16_t));
    attr_field[0].data.size=1;

    attr_field[1].id=17;
    attr_field[1].data.type=ESP_ZB_ZCL_ATTR_TYPE_S16;
    auxval=15;
    attr_field[1].data.value=malloc(sizeof(uint16_t));
    memcpy(attr_field[1].data.value,&auxval,sizeof(uint16_t));
    attr_field[1].data.size=1;

    attr_field[2].id=18;
    attr_field[2].data.type=ESP_ZB_ZCL_ATTR_TYPE_S16;
    auxval=58;
    attr_field[2].data.value=malloc(sizeof(uint16_t));
    memcpy(attr_field[2].data.value,&auxval,sizeof(uint16_t));
    attr_field[2].data.size=1;
    zigbeeWriteAttrCmdReq(5,ieee_addr,1,0xFFE9,3,attr_field);
}

void zigbeeWriteAttrCmdReq(uint16_t rid,esp_zb_ieee_addr_t des_addr, uint8_t des_ep, uint16_t cluster_id,uint8_t attr_number,esp_zb_zcl_attribute_t *attr_field){
    struct queue_rid_manager_list_t element;
    esp_zb_zcl_write_attr_cmd_t res;

    element.next=NULL;
    element.rid=rid;
    element.rid_identify.id=ID_TASK_ZIGBEE_HANDLER_WRITE_ATTR_CMD_REQ;
    element.rid_identify.src_ep=des_ep;
    element.rid_identify.cluster_id=cluster_id;
    element.rid_identify.attr_num=attr_number;
    element.rid_identify.attr_id=malloc(sizeof(uint16_t)*attr_number);
    for(int i=0;i<attr_number;i++)element.rid_identify.attr_id[i]=attr_field[i].id;
    memcpy(element.rid_identify.src_addr,des_addr,sizeof(esp_zb_ieee_addr_t));
    addRidToList(element);

    res.address_mode=ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT;
    res.clusterID=cluster_id;
    res.attr_number=attr_number;
    res.attr_field=attr_field;
    res.zcl_basic_cmd.dst_endpoint=des_ep;
    res.zcl_basic_cmd.src_endpoint=ZIGBEE_HUB_ENDPOINT;
    memcpy(res.zcl_basic_cmd.dst_addr_u.addr_long,des_addr,sizeof(esp_zb_ieee_addr_t));
    esp_zb_zcl_write_attr_cmd_req(&res);
    ESP_LOGI(TAG,"SENDING");
}
```

the read and write callbacks are the followings
```

static esp_err_t zb_read_attr_resp_handler(esp_zb_zcl_cmd_read_attr_resp_message_t *message){
    int32_t rid;
    struct rid_identify_t filter;
    esp_zb_zcl_read_attr_resp_variable_t *nav=message->variables;
    esp_zb_ieee_address_by_short(message->info.src_address.u.short_addr,filter.src_addr);
    filter.id=ID_TASK_ZIGBEE_HANDLER_READ_ATTR_CMD_REQ;
    filter.cluster_id=message->info.cluster;
    filter.src_ep=message->info.src_endpoint;
    filter.attr_num=0;
    filter.attr_id=malloc(sizeof(uint16_t));
    while(nav!=NULL){
        filter.attr_num++;
        filter.attr_id=realloc(filter.attr_id,(sizeof(uint16_t))*filter.attr_num);
        filter.attr_id[filter.attr_num-1]=nav->attribute.id;
        nav=nav->next;
    }
    rid=removeRidFromList(filter);
    free(filter.attr_id);
    if(rid>=0){
        sendReadCallbackToMainTask((int16_t)rid,message->info.status,message->variables);
    }
    return ESP_OK;

}
static esp_err_t zb_write_attr_resp_handler(esp_zb_zcl_cmd_write_attr_resp_message_t *message){
    ESP_LOGW(TAG,"WRITE CALLBACK");
    int32_t rid;
    struct rid_identify_t filter;
    esp_zb_zcl_write_attr_resp_variable_t *nav=message->variables;
    esp_zb_ieee_address_by_short(message->info.src_address.u.short_addr,filter.src_addr);
    filter.id=ID_TASK_ZIGBEE_HANDLER_WRITE_ATTR_CMD_REQ;
    filter.cluster_id=message->info.cluster;
    filter.src_ep=message->info.src_endpoint;
    filter.attr_num=0;
    filter.attr_id=malloc(sizeof(uint16_t));
    while(nav!=NULL){
        filter.attr_num++;
        filter.attr_id=realloc(filter.attr_id,(sizeof(uint16_t))*filter.attr_num);
        filter.attr_id[filter.attr_num-1]=nav->attribute_id;
        ESP_LOGE(TAG,"%d",filter.attr_id[filter.attr_num-1]);
        ESP_LOGE(TAG,"STATUS %d %d",nav->status,message->info.status);
        nav=nav->next;
    }
    rid=removeRidFromList(filter);
    free(filter.attr_id);
    if(rid>=0){
        sendWriteCallbackToMainTask((int16_t)rid,message->info.status);
    }else{
        ESP_LOGI(TAG,"RID INVALIDO");
    }
    return ESP_OK;

}
static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id, const void *message)
{
    esp_err_t ret = ESP_OK;
    switch (callback_id) {
    case ESP_ZB_CORE_CMD_WRITE_ATTR_RESP_CB_ID:
        ret = zb_write_attr_resp_handler((esp_zb_zcl_cmd_write_attr_resp_message_t *)message);
        break;
    case ESP_ZB_CORE_CMD_READ_ATTR_RESP_CB_ID:
        ret = zb_read_attr_resp_handler((esp_zb_zcl_cmd_read_attr_resp_message_t*)message);
        break;
    default:
        ESP_LOGW(TAG, "Receive Zigbee action(0x%x) callback", callback_id);
        break;
    }
    return ret;
}
```
the var filter.attr_num contain the number of elements of the list (iam sure the logic is correct because i use the same logic on the read callback and works fine)

xieqinan, 2023-11-13:
@diazmanuel ,

The code you provided is correct. However, I would like to emphasize that if the write_attr command with multiple attributes is handled correctly, the stack will only respond with `nav->status == ESP_ZB_ZCL_STATUS_SUCCESS`, `message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS`, and `nav->attribute_id == 0xFFFF`, indicating that all writing has been successfully completed.

chshu, 2023-11-22:
@diazmanuel If no further request on this issue, please help to close it, thanks.

closed 2023-12-21 by espressif-bot