espressif/esp-zigbee-sdk #143: esp_zigbee_customized_devices example: multiple channel switch

muc-martin, 2023-11-09:
### Question

Could you help me with the required chanes to the esp_zigbee_customized_devices example code to achieve multiple on / off channels on the switch side as well as to handle each channel separately on the coordinator side.

What I would like to achieve:
- have 3 on off channels (possible via switches connected to the ESPs gpios)
- have these three channels communicate to the coordinator via zigbee and identify on the coordinator what channel was pressed (is it possible to just create multiple HA on off endpoints?)
- on the coordinator switch on / off the corresponding LED as connected to the ESPs gpios.

xieqinan, 2023-11-10:
@muc-martin ，

I am trying to understand your request first. Are you expecting a Zigbee coordinator with three switches can control three on/off lights through the Zigbee network? Is that correct?

muc-martin, 2023-11-13:
Yes. In terms of the example provided in esp_zigbee_customized_devices I would like to communicate between two ESP32H2.
The coordinator should be able to receive switch events on three channels from the switch ESP32H2. 

This is the conceptual setup:

LED1--|                                                                      | --- physical switch on / off: Channel 1
LED2--|-- ESP32 Coord <---------> ESP32 Switch --| --- physical switch on / off: Channel 2
LED3--|                                                                      | --- physical switch on / off: Channel 3

So I want to switch on / off LED1 with Channel 1, LED2 with channel 2 and so on.

Is this possible?

xieqinan, 2023-11-14:
@muc-martin,

Certainly, this is achievable.

Allow me to outline the straightforward steps:

1. Initially, create three endpoints for the on/off server clusters on the Coordinator side and establish three corresponding endpoints for the on/off clients on the switch side.

2. Next, bind the switch and LED individually by referring to the [bind_cb](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c#L72).

3. Lastly, ensure you specify the `src_endpoint` when using `esp_zb_zcl_on_off_cmd_req()` to dispatch the on/off command to the Coordinator side, thereby controlling the associated LED. For further details, consult the [on/off cmd](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c#L42).

muc-martin, 2023-11-15:
@xieqinan 

Thank you for your hints!

I try to outline the changes I made to the code since I get errors in the monitor. I used the [HA_on_off_light](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light) and [HA_on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) as base.

1. Creating three endpoints for switch on the client side, I adapted the esp_zp_task function and added two additional endpoints. 

``` 
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep1 = esp_zb_on_off_switch_ep_create(1, &switch_cfg);

    /* Martin */
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep2 = esp_zb_on_off_switch_ep_create(2, &switch_cfg);
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep3 = esp_zb_on_off_switch_ep_create(3, &switch_cfg);
    /* Martin */

    esp_zb_device_register(esp_zb_on_off_switch_ep1);

    /* Martin */
    esp_zb_device_register(esp_zb_on_off_switch_ep2);
    esp_zb_device_register(esp_zb_on_off_switch_ep3);
    /* Martin */

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

Create three endpoints for lights on the Coordinator side:

```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep1 = esp_zb_on_off_light_ep_create(10, &light_cfg);

    /* Martin */
    esp_zb_ep_list_t *esp_zb_on_off_light_ep2 = esp_zb_on_off_light_ep_create(11, &light_cfg);
    esp_zb_ep_list_t *esp_zb_on_off_light_ep3 = esp_zb_on_off_light_ep_create(12, &light_cfg);

    /* Martin */
    
    esp_zb_device_register(esp_zb_on_off_light_ep1);

    /* Martin */
    esp_zb_device_register(esp_zb_on_off_light_ep2);
    esp_zb_device_register(esp_zb_on_off_light_ep3);
    /* Martin */

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

However, calling esp_zb_device_register multiple times leads to a crash.

2. Bind the switch and LED 

[... 68 lines cut ...]
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* implemented light switch toggle functionality */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.src_endpoint = 2;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
    }
}

static void esp_zb_buttons_handler3(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* implemented light switch toggle functionality */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.src_endpoint = 3;
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
    }
}
/* Martin */
```

xieqinan, 2023-11-16:
> 1. Creating three endpoints for switch on the client side, I adapted the esp_zp_task function and added two additional endpoints.

   ```c
   esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
   esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_switch_clusters_create(&switch_cfg), 1, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
   esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_switch_clusters_create(&switch_cfg), 2, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
   esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_switch_clusters_create(&switch_cfg), 3, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID);
   esp_zb_device_register(ep_list);
   ```

> Create three endpoints for lights on the Coordinator side:

Note: Create three endpoints for lights on the Coordinator side, specifying the device type as `ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID`.

> Bind the switch and LED
Adapting the function user_bind_cb on the coordinator side

   ```c
   // on coordinator side
   static void user_find_cb(esp_zb_zdp_status_t zdo_status, uint16_t addr, uint8_t endpoint, void *user_ctx)
   {
       if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
           ESP_LOGI(TAG, "Found switch");
           bind_req.src_endp = endpoint;
           bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
           bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
           esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
           bind_req.dst_endp = "the lightx endpoint on coordinator side";   // Fill it accordingly; lightx will be bound with the switch endpoint.
           bind_req.req_dst_addr = "switch short address";  // Send binding command to switch
           esp_zb_zdo_device_bind_req(&bind_req, bind_cb, NULL);
       }
   }
   ```

 Note: Utilize `esp_zb_zdo_match_cluster()` instead of `esp_zb_zdo_find_on_off_light()` to find the switch on the coordinator side.

chshu, 2024-01-08:
@muc-martin closing the issue, let us know in case any follow up questions.

closed 2024-01-08 by chshu