espressif/esp-zigbee-sdk #145: receive custom cluster, abort

hzmnue, 2023-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) and ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### esp-zigbee-lib version.

1.0.2

### esp-zboss-lib version.

1.0.2

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

i can receive this msg in esp_zb_raw_command_handler_register call back

### What is the actual behavior?

if i send cluster Id( 0xff00) ,Program will abort

[image]

[image]

[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

xieqinan, 2023-11-16:
@hzmnue，

Could you elaborate on how you manage the custom cluster (0xff00) within the `zigbee_msg_handler()` callback? Although I assume that the `zigbee_msg_handler()` will not receive the custom message. 

We have introduced `ESP_ZB_CORE_CMD_CUSTOM_CLUSTER_REQ_CB_ID` and `ESP_ZB_CORE_CMD_CUSTOM_CLUSTER_RESP_CB_ID` specifically for the `zb_action_handler()` to address the custom cluster.

To better understand your issue, I would appreciate more details. On one hand, it would be beneficial to demonstrate how to send the custom cluster command and specify the values being sent. On the other hand, it would be helpful to illustrate how to handle the custom cluster command.

Additionally, I want to mention that the custom command has been optimized in the upcoming version.

xieqinan, 2023-11-21:
@hzmnue ,

We have already updated the esp-zigbee-lib to v1.0.4, optimizing the custom command and message. Could you please update the library and test the feature again?

chshu, 2024-01-08:
@hzmnue Please reopen if you see the issue with latest SDK.

closed 2024-01-08 by chshu