espressif/esp-zigbee-sdk #146: How to build an OTA file that abide to Zigbee Standard OTA file format?

g-chevalier, 2023-11-15:
### Question

Hi,

Do you have a tool that can generate a standard Zigbee OTA file giving it as input the .bin generated by the IDF?
By "standard" I mean a format that is compliant with the Zigbee OTA format as described in §11 "OVER-THE-AIR UPGRADE" of Zigbee Alliance ZCL spec (I have the version 07-5123 Revision 8 of this spec)?

This format is supported by Zigbee2MQTT and allow to manage OTA for a bunch of already supported devices. A list of manufacturers and devices can be found here: https://github.com/Koenkk/zigbee-OTA/tree/master/images

Thank you.

nomis, 2023-11-26:
It's possible to do this using [zigpy.ota.image.OTAImage](https://github.com/zigpy/zigpy/blob/0.60.0/zigpy/ota/image.py), although the API isn't great because it expects you to provide size details for `zigpy.ota.image.OTAImageHeader` that you don't know without generating the header first.

You will need to handle the 6 byte sub-element header in your OTA client because the whole image content (after the header) is sent to the device: https://github.com/nomis/candle-dribbler/blob/0.5.2/src/zigbee.cpp#L186-L255.

I'm able to create [zlib-compressed OTA images](https://github.com/nomis/candle-dribbler/blob/0.5.2/bin/create-ota.py), send them using Home Assistant and then [decompress them on the device](https://github.com/nomis/candle-dribbler/blob/0.5.2/src/ota.cpp).

g-chevalier, 2023-12-15:
Hi @nomis 

I wanted to create OTA file with your scripts (create-ota.py and esp32-app-set-desc.py), but cannot understand how to use them.

Can you please explain a little bit how to use them once the .bin has been generated by ESP-IDF?

The OTA file I managed to create with some assumption about the script usage triggers the following error when used with Z2M node scripts/add.js command (which works with other manufacturer OTA files…):
RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds

Thank you.

nomis, 2023-12-15:
I don't use Z2M but I added an image to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/test/stub/otaImages.js and it was able to parse it successfully.

You'd need to provide the full stack trace and version of Z2M that you're using to debug it further but your image file is probably truncated. All of the parameters are validated by zigpy so it shouldn't be possible to create an invalid image.

Could you attach your OTA file?

g-chevalier, 2023-12-15:
Hi,

Before going ahead and providing more details, can you please give me some instructions on how to use create-ota.py and esp32-app-set-desc.py after I have compiled and the binary is available into the build directory?

Because I'm afraid I misused those tools, and this could explain why it's not working.

nomis, 2023-12-15:
Usage: `create-ota.py -m 123 -i 456 -v 789 firmware.bin > firmware.ota`

You'll have to come up with your own Zigbee manufacturer ID (allocated by the CSA), file ID (allocated by you - identifies which device a firmware upgrade is for, so it needs to be different for incompatible revisions of the device or when you're building for different ESP32 models) and file version (allocated by you - increment for each new version - must match what's in the firmware or it's going to upgrade repeatedly).

The "header string" is optional. The `Makefile` in my project gets these values out of the `sdkconfig.h` file because I made them configurable.

(You don't need to use `esp32-app-set-desc.py`. It modifies the ESP32 app descriptor in the binary. I don't like the date/time format that Espressif use and it's more complex to parse when [generating the Zigbee date code automatically](https://github.com/nomis/candle-dribbler/blob/0.5.2/src/device.cpp#L329-L349).)

g-chevalier, 2023-12-16:
Hi @nomis 

Initially, I did what you wrote: create-ota.py -m 100 -i 200 -v 1 firmware.bin > firmware.ota
But I was confused with the other script esp32-app-set-desc.py, and thought I had to use it on the firmware to do some kind of preparation.
I used an arbitrary manufacturerCode for this first test.

Nevertheless, the OTA file just build is not working with the Z2M script "add.js" to prepare its usage.

I just redo it right now and got:

```node scripts/add.js ..\OTA-build\make_ota\flood_projector.ota
node:internal/errors:497
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:406:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Object.parseImage (E:\Devel\ZigBee\OTA_FileManagement\zigbee-OTA\lib\ota.js:17:34)
    at main (E:\Devel\ZigBee\OTA_FileManagement\zigbee-OTA\scripts\add.js:94:28)
    at Object.<anonymous> (E:\Devel\ZigBee\OTA_FileManagement\zigbee-OTA\scripts\add.js:166:1)
    at Module._compile (node:internal/modules/cjs/loader:1241:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
    at Module.load (node:internal/modules/cjs/loader:1091:32)
    at Module._load (node:internal/modules/cjs/loader:938:12) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}

Node.js v20.9.0
```

For nodejs tools, I use the last version of https://github.com/Koenkk/zigbee-OTA

nomis, 2023-12-16:
Your file is being truncated to the first zero byte because stdout on Windows is opened in text mode. I've updated [create-ota.py](https://github.com/nomis/candle-dribbler/blob/22c8292e565928d3d19594c7e089110d146caf9d/bin/create-ota.py) so that it will work on Windows.

The output filename is now an additional parameter instead of a redirect:
`create-ota.py -m 100 -i 200 -v 1 firmware.bin firmware.ota`

g-chevalier, 2023-12-16:
Bingo, your script is now working!

I first got
Error: 100 not in manufacturerNameLookup (please add it)

And then, using a code that is in the lookup table and which I'm sure I'll never use, it worked:
```
python create-ota.py -m 4420 -i 200 -v 1 flood_projector.bin flood_projector.ota
node scripts/add.js ..\OTA-build\make_ota\flood_projector.ota
Added new entry (bellow is the formatted result)
{
    "fileVersion": 1,
    "fileSize": 352373,
    "manufacturerCode": 4420,
    "imageType": 200,
    "sha512": "bd9fbd0160a2bbba00df8d876dd179ea203995a2cc9a7f769a021f074f6e590b2655f0605f2511f14eae11de0042f8f221105404473f51bf9236193616e945c6",
    "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Lutron/flood_projector.ota",
    "path": "images/Lutron/flood_projector.ota"
}
```

I just have now to adapt the other stuff to have a local url as explained here https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
And probably discover new problems :) ...

Thank you for your support.
I'll update this issue when I'll make new headways.

g-chevalier, 2023-12-19:
Hi,

I finally managed to upload the firmware on the ESP using Zigbee2MQTT after having formatted the .bin file to a valid OTA file using the script from Simon.

Unfortunately, after everything has been received and written into flash, I get a validation error upon call of esp_ota_end().
The error message is:
```
Failed to end OTA partition, status: ESP_ERR_OTA_VALIDATE_FAILED
```
And just before returning from esp_ota_end(), the following message is logged:
```
E (00:33:42.183) esp_image: Segment 0 load address 0x42018020, doesn't match data 0x00120020
0x42018020: zb_buf_copy_func at ??:?
```

Anyone have an idea of what can be the issue?

nomis, 2023-12-19:
Did you handle writing only the sub-image content and not its header which gets transmitted to the device too?
Did you decompress the content? (You could modify the script to not use zlib but it'll be about double the size.)

It shouldn't be possible to get this far without doing those steps because the first byte won't be `E9` and it'll fail an earlier check.

I suggest logging the first and last few bytes written to the OTA partition to check those match the original file.

g-chevalier, 2023-12-19:
Yes, I'm writing only the sub-image data without its header (I did get the error with the E9 magic byte on first try and fixed it).
Currently, I simplified your script to not use compression to have thing simpler (but 2 times slower!).

I will dump the end of the file to check (I already dumped the beginning to discover various issues…).
Up to now, downloaded bytes through OTA match .ota file size (minus the two headers).

g-chevalier, 2023-12-21:
Some news on the error "esp_image: Segment 0 load address 0x42018020, doesn't match data 0x00120020".

This was due to a difference between the Flash Partitions definition of the current firmware and the one downloaded. Normally, they are the same, but as OTA is so long, I took another firmware to test the process (which had different partitions).
After changing partition table of the other firmware, OTA worked fine.

Then I also tested compression, and I was able to make it working.

Thanks again to nomis for its help and the python script.

Perhaps this is now time that Espressif Zigbee SDK team take care of providing such a tool and document it?

Suxsem, 2024-01-13:
Just to leave a working code for the future... this allow receiveing OTA updates from zigbee2mqtt (ota image created using https://[raw.githubusercontent.com/nomis/candle-dribbler/0.5.2/bin/create-ota.py](https://raw.githubusercontent.com/nomis/candle-dribbler/0.5.2/bin/create-ota.py)), modified to NOT compress the image:

Still...it takes about 2 hours to complete the update so not really feasible for battery powered devices...

```
static const esp_partition_t *s_ota_partition = NULL;
static esp_ota_handle_t s_ota_handle = 0;
size_t ota_data_len_;
size_t ota_header_len_;
bool ota_upgrade_subelement_;
uint8_t ota_header_[6];

static esp_err_t zb_ota_upgrade_status_handler(esp_zb_zcl_ota_upgrade_value_message_t message)
{
    static uint32_t total_size = 0;
    static uint32_t offset = 0;
    static int64_t start_time = 0;
    esp_err_t ret = ESP_OK;
    if (message.info.status == ESP_ZB_ZCL_STATUS_SUCCESS) {
        switch (message.upgrade_status) {
        case ESP_ZB_ZCL_OTA_UPGRADE_STATUS_START:
            ESP_LOGI(TAG, "-- OTA upgrade start");
			ota_upgrade_subelement_ = false;
			ota_data_len_ = 0;
            ota_header_len_ = 0;
            start_time = esp_timer_get_time();
            s_ota_partition = esp_ota_get_next_update_partition(NULL);
            assert(s_ota_partition);
            ret = esp_ota_begin(s_ota_partition, OTA_WITH_SEQUENTIAL_WRITES, &s_ota_handle);
            ESP_RETURN_ON_ERROR(ret, TAG, "Failed to begin OTA partition, status: %s", esp_err_to_name(ret));
            break;
        case ESP_ZB_ZCL_OTA_UPGRADE_STATUS_RECEIVE:
        	size_t payload_size = message.payload_size;
            const uint8_t *payload = message.payload;

            total_size = message.ota_header.image_size;
            offset += payload_size;

            ESP_LOGI(TAG, "-- OTA Client receives data: progress [%ld/%ld]", offset, total_size);

            /* Read and process the first sub-element, ignoring everything else */
			while (ota_header_len_ < 6 && payload_size > 0) {
                ota_header_[ota_header_len_] = payload[0];
                ota_header_len_++;
				payload++;
				payload_size--;
			}

			if (!ota_upgrade_subelement_ && ota_header_len_ == 6) {
				if (ota_header_[0] == 0 && ota_header_[1] == 0) {
					ota_upgrade_subelement_ = true;
					ota_data_len_ =
						  (((int)ota_header_[5] & 0xFF) << 24)
						| (((int)ota_header_[4] & 0xFF) << 16)
						| (((int)ota_header_[3] & 0xFF) << 8 )
						|  ((int)ota_header_[2] & 0xFF);
					ESP_LOGD(TAG, "OTA sub-element size %zu", ota_data_len_);
				} else {
					ESP_LOGE(TAG, "OTA sub-element type %02x%02x not supported", ota_header_[0], ota_header_[1]);
					return ESP_FAIL;
				}
			}

            if (ota_data_len_) {
                payload_size = min(ota_data_len_, payload_size);
				ota_data_len_ -= payload_size;

                if (message.payload_size && message.payload) {
                    ret = esp_ota_write(s_ota_handle, payload, payload_size);
                    ESP_RETURN_ON_ERROR(ret, TAG, "Failed to write OTA data to partition, status: %s", esp_err_to_name(ret));
[... 45 lines cut ...]
static void esp_zb_task(void *pvParameters)
{
.....
    /* ota cluster */
    esp_zb_ota_cluster_cfg_t ota_cluster_cfg = {
        .ota_upgrade_file_version = OTA_UPGRADE_RUNNING_FILE_VERSION,
        .ota_upgrade_downloaded_file_ver = OTA_UPGRADE_DOWNLOADED_FILE_VERSION,
        .ota_upgrade_manufacturer = OTA_UPGRADE_MANUFACTURER,
        .ota_upgrade_image_type = OTA_UPGRADE_IMAGE_TYPE,
    };
    esp_zb_attribute_list_t *ota_client_attr_list = esp_zb_ota_cluster_create(&ota_cluster_cfg);
    /** add client parameters to ota client cluster */
    esp_zb_zcl_ota_upgrade_client_variable_t variable_config = {
        .timer_query = ESP_ZB_ZCL_OTA_UPGRADE_QUERY_TIMER_COUNT_DEF,
        .hw_version = OTA_UPGRADE_HW_VERSION,
        .max_data_size = OTA_UPGRADE_MAX_DATA_SIZE,
    };
    esp_zb_ota_cluster_add_attr(ota_client_attr_list, ESP_ZB_ZCL_ATTR_OTA_UPGRADE_CLIENT_DATA_ID, (void *)&variable_config);

    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_ota_cluster(esp_zb_cluster_list, ota_client_attr_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
...
}
```

Hedda, 2024-04-08:
Tip is that can/should also check/verify and test OTA files with zigpy-cli ( command line tool by/from the zigpy project):

https://github.com/zigpy/zigpy-cli

https://github.com/zigpy/zigpy-cli/blob/dev/README.md#ota

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

Note that the zigpy-cli is used by the zigpy project (and the ZHA integration in Home Assistant) but can also be used stand-alone.

https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

https://github.com/zigpy/zigpy

https://www.home-assistant.io/integrations/zha

chshu, 2025-02-08:
An official tool is provided to generate the standard OTA file format: [image_builder_tool](https://github.com/espressif/esp-zigbee-sdk/tree/main/tools/image_builder_tool), please follow the instructions here: [ota docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_ota_upgrade.html#generate-an-ota-upgrade-image-for-the-esp-platform).

Please file new issues if any problem when using the new tool.

closed 2025-02-08 by chshu